rwaldron/johnny-five #719: Servo Arduino Raspberry Pi serial port

goliatone, 2015-04-10:
Logging this here as it drove me nuts for a good while.

I have a servo connected to an Arduino Uno which is connected to a Raspberry Pi 2 B. I was running the sample servo script. It was able to connect, as soon as I would send a command from the `rpl` it would error out with a **Error: Serialport not open.**.

However, I installed the arduino IDE- `sudo apt-get install arduino`- and I was able to run a simple sketch from the Raspberry controlling the servo OK.

I was also able to control the pin 13 LED from the `rpl` with no issues.

The Raspberry Pi was powering the  Arduino so I ended up powering it with an external source and that fixed the issue.

johnny-five v 0.8.53
node v0.12.0
node-gyp v1.0.3
serialport v1.6.3

Incidentally I was having issues compiling `node-gyp` and `serialport` was not being installed. It took me a bit to fix that and get it running.

```
>> servo.center()
Error: Serialport not open.
    at SerialPortFactory.SerialPort.write (/home/pi/CODE/servo/node_modules/johnny-five/node_modules/serialport/serialport.js:254:17)
    at Board.analogWrite (/home/pi/CODE/servo/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:606:11)
    at Board.servoWrite (/home/pi/CODE/servo/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:651:20)
    at Servo.Controllers.Standard.servoWrite.value (/home/pi/CODE/servo/node_modules/johnny-five/lib/servo.js:109:17)
    at Servo.to (/home/pi/CODE/servo/node_modules/johnny-five/lib/servo.js:332:10)
    at Servo.center (/home/pi/CODE/servo/node_modules/johnny-five/lib/servo.js:422:15)
    at repl:1:8
    at REPLServer.self.eval (repl.js:112:21)
    at Interface.<anonymous> (repl.js:239:12)
    at Interface.emit (events.js:95:17)
```
 
```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var servo = new five.Servo(10);

  this.repl.inject({
    servo: servo
  });

  servo.min();
});
```

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var led = new five.Led(13);

  this.repl.inject({
    led: led
  });

  led.blink();
});
```

Resseguie, 2015-04-10:
cc @bryan-m-hughes

bryan-m-hughes, 2015-04-10:
It makes sense that trying to power the servo from the Arduino from the Pi wouldn't work. The Raspberry Pi's power supply is pretty notorious. On older models, powering and controlling a servo straight from the pi would cause the board to reset the instant you turned it on! The newer models are better, but powering servos from the pi still isn't really an option.

@goliatone can you tell us more about the compile issues you had?

goliatone, 2015-04-10:
Well, I was **able to control** the Arduino board from the Raspberry Pi without extra power by installing the arduino IDE and loading a basic sketch. 
I will dig around for the error logs, I know I did save them.

bryan-m-hughes, 2015-04-10:
Well, just because it worked under testing doesn't mean you should do it :-P. As soon as the servo stalls, it'll probably reset the RPi.

So just to clarify, do you only get that error when powering servos from the RPi? or do you get that error all the time?

goliatone, 2015-04-11:
You can install the arduino IDE on the RPi. I had a node script and an arduino sketch that were equivalent. Running the node script gave me a **Error: Serialport not open** but running the sketch did not- and the servo was behaving as expected so the serial port was actually open.

You actually **have** to enable serial port communication on the RPi. By default is used by a login shell that you can connect over serial cable. So getting a serial port not open makes sense in that context. 
But to load the sketch on the arduino from the RPi the IDE also uses the serial port, so that threw me off for a bit.

I guess the difference is that the IDE only uses the serial port to **load** the sketch, but when the servo is running there is no serial port communication. Node is talking over the serial port on real time and the RPi does not have enough power to feed the servo and keep up with the chatter. 

And just to clarify, I did not file this as a bug, my intention was to leave a comment in case someone else has the same issue in the future. So, nothing to fix :) Just give it a FYI label or whatever and close it.

>Well, just because it worked under testing doesn't mean you should do it :-P. As soon as the servo stalls, it'll probably reset the RPi.
Actually, the RPi was not getting reset. I kept working fine outside throwing that error.

rwaldron, 2015-10-08:
Thanks for documenting the issue here :)

closed 2015-10-08 by rwaldron