rwaldron/johnny-five #727: Servo: Event move:complete

goliatone, 2015-04-14:
I just observed that servo **will not** fire events if **time** is not provided:
```js
function handler(){
  console.log('DONE');
}
servo.on('move:complete', handler);
servo.to(20);
```

If you **do specify** a time argument, it does fire the event:
```js
function handler(){
  console.log('DONE');
}
servo.on('move:complete', handler);
servo.to(20, 200);
```

Is this behavior expected? If so, could we mention such detail in the wiki?

dtex, 2015-04-14:
Expected behavior. It's an exclamation on the wiki tonight.

Ha, so much for speech to text... Let's try that again...

That is expected behavior. I'll write up an explanation for the wiki tonight.

dtex, 2015-04-14:
So currently the wiki says:

## Events

It's recommended to use a rotary potentiometer as the mechanism for determining servo movement.

- **move:complete** this is emitted when a timed move is completed.

---------------------------------------

Here's my proposed change:

## Events

It's recommended to use a rotary potentiometer as the mechanism for determining servo movement.

- **move:complete** this is emitted when a timed move is completed. A timed move is any call to Servo.to() or Servo.step() that includes a duration parameter. Calling either of these methods without a duration is not an asynchronous task.

---------------------------------------

Is that better or is it still murky?

goliatone, 2015-04-14:
Yes, I think is better.
I actually had read that paragraph in the documentation, but apparently my
attention got stuck on the potentiometer part and I started thinking how
would that work making me not pay enough attention to the move:complete
part.

Traverso, 2015-04-14:
What about:
move:complete this is emitted when a timed move is completed.
The event won't fire unless a time argument has been passed to the servo.to method

Resseguie, 2015-04-14:
@dtex Could we just fire the event anyway, even for the synchronous task? That makes the event listening code simple and consistent as in the OP example.

dtex, 2015-04-14:
That makes me uncomfortable because the servo will not have actually finished moving when move:complete is fired.

rwaldron, 2015-04-14:
> That makes me uncomfortable because the servo will not have actually finished moving when move:complete is fired.

Agree.

Resseguie, 2015-04-14:
Misunderstood you then. I thought you were saying the whole thing was synchronous (including the servo movement) if no duration given.

I thought that was strange, but it's what your proposed doc update implied to me.

dtex, 2015-04-14:
Well in that case my wording is poor. @Traverso's is simpler and avoids that miscommunication

rwaldron, 2015-04-14:
Unless you explicitly specify the duration, the value angle is written to the servo, and it travels there at default factory speed. There is no way to know how long that takes or when it completed without closing the feedback loop with some kind of sensor. Therefore, it's wrong to emit confirmation of completion when you have no control over the time an operation takes.

Resseguie, 2015-04-14:
> @Traverso's is simpler and avoids that miscommunication

agreed

rwaldron, 2015-05-18:
Is this updated?

dtex, 2015-05-19:
I've updated the wiki

closed 2015-05-19 by dtex