rwaldron/johnny-five #728: Led animation callbacks when the animation is cut short

scottgonzalez, 2015-04-14:
Currently all the Led animation methods take an optional callback, but those callbacks aren't invoked if `.stop()` is called or if another animation is started in the middle of the current animation.

Is this the desired behavior or would you like a PR to invoke the callback?

rwaldron, 2015-04-15:
@Resseguie @dtex @boneskull wdyt?

Resseguie, 2015-04-15:
@dtex how would this work if we make the switch to your Animation()?

dtex, 2015-04-15:
On animation we have onStop and onComplete. onStop only fires when the user explicitly calls Animation.stop(), whereas onComplete is only called when the animation finishes naturally.

Resseguie, 2015-04-15:
I could convince myself that both behaviors is "correct" with a little thinking.

On one hand, if I specify a callback, I might expect it to be in a particular state when my callback function is called. (Probably more so for servos than leds, but...)

But I might just care that it's not running anymore, in cases where I want to fire off the "next thing," whatever that is.

Resseguie, 2015-04-15:
@dtex Oh, I like that concept. Answers both my cases above.

But do you always have to register a listener for both? Or could 'stop' also fire when it completes?

dtex, 2015-04-15:
My bad, stop is also called when the animation completes.

Resseguie, 2015-04-15:
So @scottgonzalez it might be a moot point if we make the switch to use the other external Animation functions.

Resseguie, 2015-04-15:
See #730

scottgonzalez, 2015-04-15:
Sure, let's hash out the details in #730. If we're going to use `onstop` for the Led animations, we'll need to document that the callback for all the methods like `.pulse()` might be invoked with the Led in a random state due to the animation being stopped. Once we have a decision, we can make sure we have tests to cover the desired scenarios.

dtex, 2015-12-17:
#730 landed a while back. Can I close this?

reconbot, 2016-05-25:
I'm going to say yes.

closed 2016-05-25 by reconbot