rwaldron/johnny-five #730: Reuse Animation for Led instead of it's own implementation

Resseguie, 2015-04-15:
For example. Led.prototype.pulse becomes something like:

``` js
Led.prototype.pulse = function(time) {
  this.animation = this.animation || new Five.Animation(this);
  this.animation.enqueue({
    duration: time, 
    keyFrames: [0, 0xff],
    metronomic: true,
    easing: "inQuad"
  });
};
```

scottgonzalez, 2015-04-15:
If I'm understanding this correctly, this would change the behavior of `.pulse()`. Is that desired, or would `.stop()` be added before the `.enqueue()`? This is also missing the callback, which I guess would use the `onloop` option.

dtex, 2015-04-15:
You're right. To behave the same stop() would have to be added before enqueue().

boneskull, 2015-04-15:
I'm coming here from #728.  I don't have a whole lot of input other than what I know from using [bluebird](https://github.com/petkaantonov/bluebird).

Bluebird has a "cancellable Promise" API, which works like this:

1.  Create a `Promise`, which are by default *not* cancellable
2.  Call `myPromise.cancellable()` which flips some bit
3.  If `myPromise` is not canceled, it is resolved or rejected like any other `Promise`. 
4.  If `myPromise.cancel()` *is* called, then the `Promise` is rejected--however, it's rejected with a particular `Error` "subclass" instance, which the consumer may detect

So if we applied this strategy to `Led.prototype.pulse()`, the callback would *always* be called.  If the callback currently gets passed an `err` parameter for some reason, then we may want to consider subclassing or at least tossing a property on it in the case of a "stopped" animation.  If not, then we could start passing it, and it would only be present if the animation was explicitly stopped.

*I would not expect a callback to be called with some sort of "success" message if the animation did not run as originally intended.*

A few questions:

1.  Can animations be "combined" somehow?  Can you call `pulse()` and `colorCycle()` (or whatever) at the same time?  Why or why not?
2.  When you call `stop()`, how do you know what you're stopping?  If they are in a queue, do you end up stopping all of them?
3.  Would it be more natural for an "animation" function to return an `EventEmitter` having a `stop()` method?  Should that object emit `complete` and/or `stopped` events?

dtex, 2015-04-15:
> A few questions:

> Can animations be "combined" somehow? Can you call pulse() and colorCycle() (or whatever) at the same time? Why or why not?

Oh, this is good foresight. Currently any device or group of devices that are being animated have a ```render``` method. The render method can be overridden with anything you want. I use this ability to define an IK solver for my robots' legs. You could combine ```colorCycle``` and ```pulse``` this way, but I think we can do better. I will look at allowing multiple properties of a device to be animated on a single keyframe.

> When you call stop(), how do you know what you're stopping? If they are in a queue, do you end up stopping all of them?

Yes, ```stop``` stops the queue and empties it. ```next``` halts the current animation segment and jumps to the next segment in the queue.
 
> Would it be more natural for an "animation" function to return an EventEmitter having a stop() method? Should that object emit complete and/or stopped events?

I thought the same thing when looking at ```Animation()``` yesterday and don't remember why it is the way it is. I went back to conversations about ```Animation()``` and didn't find anything there. Hmmmm....

scottgonzalez, 2015-04-15:
Although it's not currently documented, `Animation` has two events: `animation:pause` and `animation:stop`. However, in the case of `Led`, the animation methods (such as `.pulse()`) return the `Led` instance for chaining, so you don't have access to the events. However, the `Led` instance could forward the `Animation` events. We'd only need to do that setup once, something like (untested):

```js
Led.prototype._getAnimation() {
  if (!this.animation) {
    this.animation = new Five.Animation(this);
    this.animation.on("animation:pause", this.emit.bind(this));
    this.animation.on("animation:stop", this.emit.bind(this));
  }

  return this.animation;
};

Led.prototype.pulse = function(time, callback) {
  this.animation = this._getAnimation();
  ....
};
```

This does make you wonder why the `Animation` events are namespaced since they're emitted on the `Animation` instance.

Resseguie, 2015-04-15:
> If I'm understanding this correctly, this would change the behavior of .pulse().

Yep, sorry. That was pseudo-code-ish. I should have specified.

dtex, 2015-04-25:
Hey guys, if there is anything that is being held up by the Animation() implementation, please move ahead without it. The Animation() implementation should not break backwards compatibility on the API (if it does I'm doing it wrong). I'm struggling with implementing the mock process.nextTick/global.setImmediate functions. I'll refactor/rebase as needed.

rwaldron, 2015-04-25:
Don't worry about mocking them, we can put timing sensitive tests in the extended set—without mocks.

dtex, 2015-04-26:
I have proposed API additions for LED. Currently we handle:
````
led.pulse(); // default duration, no callback
led.pulse(2000); // two second pulse no callback
led.pulse(2000, function() { /* Do something here */ }); // two second pulse with callback
````

For completeness I would like to add:
````
led.pulse(function() { /* Do something here */ }); default duration, with callback
````

And for new functionality I would like to add:
````
led.pulse(opts); // opts apply to the animation segment
led.pulse(opts, callback); // Same, with a callback
````
The most obvious use case here would be to change the default easing function, but all the [segment options](https://github.com/rwaldron/johnny-five/wiki/Animation#segment-properties) would be available.

Thoughts?

dtex, 2015-04-26:
Adding an example:
````
led.pulse({
  duration: 500,
  easing: "inOutBounce"
}); 
````

Resseguie, 2015-04-27:
I like it.

dtex, 2015-04-27:
I've moved forward with it. Everything is working nicely, I just need to finish up the extended tests and rebase. The mocked setImmediate and nextTick felt so close and so much better than the status quo... We have got to try again (and by "we" I mean someone smarter than me).

A couple of notes:
* The signature for fade() is ugly. It already had 3 params and now with a possible opts object it's worse. There is a lot of conditional handling of those params.
* The pulse and fade tests for the PCA-9685 weren't actually testing anything that wasn't already tested in the default tests.

dtex, 2015-04-27:
I have a PR ready at #756

scottgonzalez, 2015-06-02:
This is done.

rwaldron, 2015-06-02:
Thanks everyone!

closed 2015-06-02 by rwaldron