rwaldron/johnny-five #732: Led.RGB stop() errors on 0.8.56

bartmichu, 2015-04-15:
Hi, after updating to 0.8.56 I'm getting errors when I try to ```led.stop()``` in repl
```
"use strict";

var five = require("johnny-five");
var uno = new five.Board();

uno.on("ready", function () {
  var led = new five.Led.RGB({pins: [9, 10, 11]});

  this.repl.inject({
    led: led
  });
});
```

```
1429124110134 Device(s) /dev/ttyUSB0,/dev/ttyACM0,/dev/ttyACM1,/dev/ttyACM2 
1429124110147 Connected /dev/ttyUSB0 
1429124113335 Repl Initialized
>> led.blink()
// it starts blinking

>> led.stop()
TypeError: Cannot set property 'isRunning' of undefined
    at Led.stop (/home/test/johnny-five/node_modules/johnny-five/lib/led/led.js:478:19)
    at repl:1:6
    at REPLServer.self.eval (repl.js:112:21)
    at Interface.<anonymous> (repl.js:239:12)
    at Interface.emit (events.js:95:17)
    at Interface._onLine (readline.js:203:10)
    at Interface._line (readline.js:532:8)
    at Interface._ttyWrite (readline.js:761:14)
    at ReadStream.onkeypress (readline.js:100:10)
    at ReadStream.emit (events.js:98:17)
>>
```
LED stops blinking anyway so it kind of works. Arduino Nano ATmega328 connected like in wiki examples. Code was working fine before ```npm update``` when I was on 0.8.53

Resseguie, 2015-04-15:
Thanks, I'll have a look!

rwaldron, 2015-04-15:
@Resseguie I see the issue: 

```js
Led.RGB.prototype.stop = Led.prototype.stop;
```

Should be: 

```js
Led.RGB.prototype.stop = function() {
  Led.RGB.colors.forEach(function(color) {
    this[color].stop();
  }, this);
  return this;
};
```

Can you write a patch, with tests?

Resseguie, 2015-04-15:
90% there already...

Resseguie, 2015-04-15:
It worked before because `this` and state/priv was shared, I guess.

It doesn't actually need to call stop on each of the individual R/G/B leds. It just needs to clear `this.interval` and set `isRunning = false` as it does in Led.prototype.stop.

I'm going to go ahead and duplicate the Led.prototype.toggle in Led.RGB too so we avoid this in the future for that function. 

I'm adding tests for stop and toggle.

PR coming soon.

Resseguie, 2015-04-15:
PR: #734

rwaldron, 2015-04-15:
@bartmichu please confirm that v0.8.58 addresses this bug (close when ready)

bartmichu, 2015-04-16:
Confirmed, no problems now. Thanks guys!

closed 2015-04-16 by bartmichu

Resseguie, 2015-04-16:
@bartmichu thanks for the report!