rwaldron/johnny-five #735: Move mock-firmata to standalone repo

Resseguie, 2015-04-15:
Make it available for use in other libraries, creating Johnny-Five apps without hardware, etc.

dtex, 2015-04-15:
mock-io?

rwaldron, 2015-04-15:
`mock-io` is good with me. Should I make this repo and module? I will make you both authors/owners

Resseguie, 2015-04-15:
Sounds good to me. 

Is io plugin model generic enough for testing other libraries, etc?

Resseguie, 2015-04-15:
How does or should this relate to board-io? I'm mobile so I can't look in detail until later.

rwaldron, 2015-04-16:
> Is io plugin model generic enough for testing other libraries, etc?

Minimal testing, yes.

rwaldron, 2015-04-16:
Could be that we just use Board-io. I have PR towards that end: https://github.com/achingbrain/board-io/pull/5

Resseguie, 2015-04-29:
@rwaldron I'd like to reopen this. I like the new use of board-io to stub things out instead of copying all that functionality to mock-firmata. But just that by itself doesn't give me all that I'm interested in.

Our concept of `mock-io` would still need to set `this._pins` etc (as in test/util/mock-firmata.js) to be useful for testing w/o hardware and the other benefits we talked about above. `mock-io` would also need to emit `connect` and `ready` itself instead of those having to be triggered manually as our mock-firmata requires (and as used in tests).

boneskull, 2015-11-10:
see referenced issue #949 for more information.  I closed that as a dupe of this

rwaldron, 2015-11-10:
https://github.com/rwaldron/mock-firmata

closed 2015-11-10 by rwaldron

reconbot, 2015-12-16:
+1 found this today and it's beautiful