rwaldron/johnny-five #737: Add Led.RGB.intensity()

Resseguie, 2015-04-19:
Similar to current `Led.brightness()` but has range 0-100 (percent).

Should consider deprecating `.brightness()` completely and add `.intensity(0-100)` to Led as well.

Reasoning: "brightness" can be misinterpreted. "intensity" is standard term in lighting industry.

rwaldron, 2015-04-19:
Thanks for filing this. I will finish up my patch and push a branch for review tomorrow

Resseguie, 2015-04-28:
Related to #370

scottgonzalez, 2015-06-02:
> Thanks for filing this. I will finish up my patch and push a branch for review tomorrow

@rwaldron Are you still going to do this?

What's the plan for changing brightness? Does it take the current color into account or is it always an equal mix of red, green, and blue?

rwaldron, 2015-06-02:
> @rwaldron Are you still going to do this?

If you want to do this, go for it.

> What's the plan for changing brightness? Does it take the current color into account or is it always an equal mix of red, green, and blue?

If I understand you correctly, it takes the color into account. So if it's currently red, `intensity(0-100)` scales red from off to full red, but always some shade of red.

PR #821 (2015-07-03): Led.RGB: Add intensity method
Fixes #737 

Alternative to #818, storing the intensity as separate state from color.
commit d230a181 (2015-07-03): Led.RGB: Add intensity method

Fixes #737
closed 2015-07-03 by rwaldron