rwaldron/johnny-five #738: how can i access max6675, max31855K

chedongzhe, 2015-04-19:
how can i access max6675, max31855K with johnny-five

dtex, 2015-04-23:
Hi @chedongzh 

Sorry nobody has gotten back to you. I suspect nobody has any first hand knowledge of these specific devices. It looks like they use SPI. How best to tackle that will depend on what MicroController you are using. Are you using an Arduino?

rwaldron, 2015-04-23:
(Somehow I just missed it! sorry :frowning:  )

ScrewedBrain, 2016-07-13:
It's been a year since this messages, but I'd love to see it as well.)

ScrewedBrain, 2016-07-13:
max6675

rwaldron, 2016-07-14:
@ScrewedBrain sorry, still no SPI in the protocol spec

dtex, 2018-01-11:
Hi @chedongzhe and @ScrewedBrain  ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet (and we still need SPI support in firmata). Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the MAX6675 and MAX31855K there.

closed 2018-01-11 by dtex