rwaldron/johnny-five #742: Refactor Led.RGB to device/controller model.

rwaldron, 2015-04-20:
- Add support for BlinkM controller.

scottgonzalez, 2015-04-22:
If you have some explanations or examples of this model, I could try to tackle this. Would probably give me some better understanding of how everything works.

Resseguie, 2015-04-22:
I need to learn more about the device/controller model too, so I'm happy to think it through with you @scottgonzalez 

Servo and LCD are two examples I was going to start with.

@dtex is a good resource for understanding the organization.

dtex, 2015-04-22:
Motor is also a good example. Mentioning @BrianGenisio because he's been in the position of having to make sense of it all.

dtex, 2015-04-22:
This helps: https://github.com/rwaldron/johnny-five/issues/665#issuecomment-85332468

And there was another, more detailed explanation but I'm having trouble finding it right now.

dtex, 2015-04-22:
Here's the other: https://github.com/rwaldron/johnny-five/issues/486#issuecomment-63515604

Resseguie, 2015-04-22:
Seems like we always have to hunt that stuff down. We need to put an "understanding devices/controllers" section on the wiki somewhere and make it easy to find.

scottgonzalez, 2015-04-22:
I'll take notes as I work on this tonight, then we can flesh it out into full API docs or a tutorial or something.

frxnz, 2015-04-22:
> We need to put an "understanding devices/controllers" section on the wiki somewhere

[We have this](https://github.com/rwaldron/johnny-five/wiki/Control-System-Hierarchy-Overview), but looking at the comments @dtex linked we could maybe expand on it a little.

Resseguie, 2015-04-22:
Thanks @frxnz. An example or two there to put them in context would probably also be helpful.
See #746

scottgonzalez, 2015-05-17:
Since `Led.RGB` is documented as having `red`, `green`, and `blue` properties which contain `Led` instances, how do you want to handle devices like BlinkM where you can't use a composite of multiple `Led` instances to construct the `Led.RGB` instance? Should we just drop those properties from the public API and update the documentation for the shape of the component?

rwaldron, 2015-05-17:
TBH...

> Should we just drop those properties from the public API and update the documentation for the shape of the component?

Sounds like what should've been done the first time. So yeah, do this.

commit 1e9c148d (2015-06-01): Led.RGB: Add BlinkM support

Fixes #742
closed 2015-06-01 by rwaldron