rwaldron/johnny-five #745: custom logging

boneskull, 2015-04-21:
afaik, if you don't want J5 dumping its logs to console, your recourse is to monkeypatch:

```js
    var boardProto = j5.Board.prototype;

    // redirect logs to server logs
    _.each(_.keys(boardProto.log.types), function (type) {
      boardProto[type] = function () {
        var args = _.toArray(arguments).join(' ');
        // this is a Hapi server
        server.log.apply(server, [type].concat(args));
      };
    });
```

Any ideas on how to make this easier?

rwaldron, 2015-04-21:
Doesn't `debug: false, repl: false` shut it up?

rwaldron, 2015-04-21:
(if not, then we got bugz)

boneskull, 2015-04-21:
Don't worry; that works. But I don't want do suppress the logging--I want to redirect it.

rwaldron, 2015-04-21:
Ah, I see. I completely misunderstood. That was before I ate lunch. 

Let's figure out a way to allow this

boneskull, 2015-04-21:
probably some cmd-line flag that allows you to specify an adapter module

scottgonzalez, 2015-04-22:
Perhaps the simplest solution would be to just emit log events from `Board.prototype.log`?

I did a search for uses of `console` and it looks like there are lots of warning/deprecation notices that go directly through `console.log()`/`console.warn()`. If you want to handle these as well, you could create a log module and have everything use that.

rwaldron, 2015-04-22:
Yeah, those need to be cleaned up :/

rwaldron, 2015-07-11:
https://github.com/rwaldron/johnny-five/commit/5381dc2b39166488a87e26b87875133a1c4b4cbe

closed 2015-07-11 by rwaldron