rwaldron/johnny-five #746: Expand wiki entry for Control System Hierarchy (component, controller, device)

Resseguie, 2015-04-22:
The basics of the Control System Hierarchy are described [here](https://github.com/rwaldron/johnny-five/wiki/Control-System-Hierarchy-Overview) but it would be helpful to expand this, and maybe add examples.

Two comments from @dtex that might provide additional content to include:
https://github.com/rwaldron/johnny-five/issues/665#issuecomment-85332468
https://github.com/rwaldron/johnny-five/issues/486#issuecomment-63515604

rwaldron, 2015-04-22:
Sure, sounds like a good idea.

scottgonzalez, 2015-04-30:
As I ask a million questions about controller initialization in gitter, @Resseguie says:

> A wiki page on I2C support would be helpful too. What it is, assumptions like the SDA/SCL pins, etc.

dtex, 2018-01-08:
Add serial to that list

dtex, 2018-01-08:
Oh, and a wiki page on back-packs. Some of the other links on the www are starting to disappear

closed 2019-01-10 by stale[bot]