rwaldron/johnny-five #749: Requests to create a digital Pin with mode 0 not working

mMerlin, 2015-04-24:
johnny-five@0.8.58
$ uname -srvpio
Linux 3.19.3-200.fc21.x86_64 #1 SMP Thu Mar 26 21:39:42 UTC 2015 x86_64 x86_64 GNU/Linux
$ node --version
v0.10.38

The Pin() constructor is not honoring requests to create a Pin with mode 0.  The problem is pin.js:86
```javascript
this.mode = this.as || opts.mode || (isAnalogInput ? 0x00 : 0x01);
```
Back at pin.js:32
```javascript
var isAnalogInput = Pin.isAnalog(opts);
```
sets isAnalogInput to true, only for an analog pin number (prefix "A" or "I").  Everything else is javascript 'falsey', so the right end of that logical expression is always 0x01.  When opts.mode === 0, that is 'falsey' again, so this.mode gets set to 1.

Dropping the apparently always undefined (and 'falsey') this.as, I think the correct line should be
```javascript
this.mode = typeof opts.mode === 'undefined' ? (isAnalogInput ? 0x00 : 0x01) : opts.mode;
```
Or using the defined constants
```javascript
this.mode = typeof opts.mode === 'undefined' ?
  (isAnalogInput ? Pin.INPUT : Pin.OUTPUT) : opts.mode;
```

Test Code
```Javascript
'use strict';

var five = require('johnny-five');
var myBoard = new five.Board();

function boardIsReady() {
  var tst, p = {};
  console.log('\nTest five.Pin(?)');

  tst = 2;
  p.p2 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p2.mode);

  tst = { pin: 4, mode: 0 };
  p.p4 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p4.mode);

  tst = { pin: 7, mode: 1 };
  p.p7 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p7.mode);

  tst = { pin: 14, mode: 2 };
  p.p14 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p14.mode);

  tst = { pin: 3, mode: 3 };
  p.p3 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p3.mode);

  tst = { pin: 5, mode: 4 };
  p.p5 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p5.mode);

  tst = { pin: 15, mode: 0 };
  p.p15 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p15.mode);

  tst = { pin: 16, mode: 1 };
  p.p16 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p16.mode);

  tst = { pin: 17, mode: 2 };
  p.p17 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p17.mode);

  tst = { pin: 18, mode: 3 };
  p.p18 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p18.mode);

  tst = { pin: 19, mode: 4 };
  p.p19 = new five.Pin(tst);
  console.log(JSON.stringify(tst), '==> mode:', p.p19.mode);

  process.exit();
}

myBoard.on('ready', boardIsReady);
```

Output Before changing pin.js
```
Test five.Pin(?)
2 ==> mode: 1
{"pin":4,"mode":0} ==> mode: 1
{"pin":7,"mode":1} ==> mode: 1
{"pin":14,"mode":2} ==> mode: 2
{"pin":3,"mode":3} ==> mode: 3
{"pin":5,"mode":4} ==> mode: 4
{"pin":15,"mode":0} ==> mode: 1
{"pin":16,"mode":1} ==> mode: 1
{"pin":17,"mode":2} ==> mode: 2
{"pin":18,"mode":3} ==> mode: 3
{"pin":19,"mode":4} ==> mode: 4
```
Output After
```
Test five.Pin(?)
2 ==> mode: 1
{"pin":4,"mode":0} ==> mode: 0
{"pin":7,"mode":1} ==> mode: 1
{"pin":14,"mode":2} ==> mode: 2
{"pin":3,"mode":3} ==> mode: 3
{"pin":5,"mode":4} ==> mode: 4
{"pin":15,"mode":0} ==> mode: 0
{"pin":16,"mode":1} ==> mode: 1
{"pin":17,"mode":2} ==> mode: 2
{"pin":18,"mode":3} ==> mode: 3
{"pin":19,"mode":4} ==> mode: 4
```

rwaldron, 2015-04-24:
Excellent bug! Thanks for the thorough report :)

rwaldron, 2015-04-24:
I can fix this now, _or_ if you'd like to convert your tests and add them to `test/pin.js` and make your suggested change, that works too. Let me know!

mMerlin, 2015-04-24:
I had a look at the tests, but do not understand how (at least) the mocking works.  I could easily do a pull request with just the one line fix to pin.js, but not so sure about the tests.  Is there an overview description somewhere about how the tests work?  Also is there a way to run a test from node instead of through grunt?  To learn how the test run, I could use node-debug, and trace a bit.  If I could figure out how to run it.

rwaldron, 2015-04-24:
> I had a look at the tests, but do not understand how (at least) the mocking works. I could easily do a pull request with just the one line fix to pin.js, but not so sure about the tests.

Do this and I will write up some tests. 

> Is there an overview description somewhere about how the tests work?

There might be, but I don't actually remember— @divanvisagie would know

mMerlin, 2015-04-25:
I see a variation of my fix was included in the latest set of commits.  I created some more unit test for pin.  A PR is queued for that, although it has failing asserts.  I need a few details about whether they are detecting a bug that needs fixing, or the tests should be changed.  See the note with the PR.

rwaldron, 2015-04-25:
> I see a variation of my fix was included in the latest set of commits.

Ugh, I had actually meant to manually enter you as the author, but I completely forgot by the time I was done with the patch. I'm really sorry about that :(

closed 2015-05-07 by mMerlin