rwaldron/johnny-five #750: No callbacks reached for digital input Pins

mMerlin, 2015-04-24:
context: johnny-five@0.8.58
$ uname -srvpio
Linux 3.19.3-200.fc21.x86_64 #1 SMP Thu Mar 26 21:39:42 UTC 2015 x86_64 x86_64 GNU/Linux
$ node --version
v0.10.38

This seems like it should have been reported long ago as a bug, but I can not see what I am doing wrong.  I started by trying to create a digital sensor and could not get the .on('data', callback) to actually execute the callback.  So I dropped down to the simpler Pin object and tried again.  Still nothing.  Below are 2 almost identical scripts.  One for an analog pin, the other for a digital pin.  All 3 callbacks for the analog pin are being executed, but none for the digital pin.  Same results before and after I patched the problem that was changing input pins to output.

I did some browsing of the source code, and tracing with Node Inspector.  I could not find where the digital values emit the callbacks.  I found where it looked like it should happen, but breakpoints say that the code is never executed.

```javascript
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Analog test',
  pin: 'A5',
  type: 'analog',
  range: [0, 1023],
  freq: 1000
};

function readCallback1(val) {
  console.log('read callback 1 with:', val);
}
function readCallback2(val) {
  console.log('read callback 2 with:', val);
}
function dataCallback() {
  /* jshint validthis:true */
  console.log('data callback for:', this.id, this.value);
}

function boardIsReady() {
  var tst = new five.Pin(cfg);
  tst.on('data', dataCallback);
  tst.io.analogRead(tst.pin, readCallback1);
  five.Pin.read(tst, readCallback2);
}

myBoard.on('ready', boardIsReady);
```

```javascript
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Digital test',
  pin: '4',
  type: 'digital',
  range: [0, 1],
  mode: five.Pin.INPUT,
  freq: 1000
};

function readCallback1(val) {
  console.log('read callback 1 with:', val);
}
function readCallback2(val) {
  console.log('read callback 2 with:', val);
}
function dataCallback() {
  /* jshint validthis:true */
  console.log('data callback for:', this.id, this.value);
}

function boardIsReady() {
  var tst = new five.Pin(cfg);
  tst.on('data', dataCallback);
  tst.io.digitalRead(tst.pin, readCallback1);
  five.Pin.read(tst, readCallback2);
}

myBoard.on('ready', boardIsReady);
```

rwaldron, 2015-04-24:
@mMerlin thanks again for such a great bug report :)

> This seems like it should have been reported long ago as a bug,

Maybe, but I'm not surprised when bugs in `Pin` class go unnoticed, because I suspect it's very rarely used in practice, as the most common components have a dedicated class. 

> I started by trying to create a digital sensor and could not get the .on('data', callback) to actually execute the callback. So I dropped down to the simpler Pin object and tried again. Still nothing. 

Can you give more detail about the sensor you're using?

>  Below are 2 almost identical scripts. 

A few things I've noticed about both: 

- Neither have a call to `pinMode`
  - This will work with `analogRead` because analog pins (in Arduino parlance: "A0" and friends) don't need the mode set explicitly, so StandardFirmata does not require it. 
  - Conversely, this won't work with digital pins because digital pins _do_ need the mode set explicitly, so StandardFirmata require it. 
- I wonder why you're using the `io` methods? Johnny-Five provides a `Sensor` class, which is an analog sensor component class by default, or can be set to "digital". The class does all the setup work internally.  

```js
var a = new five.Sensor("A0");
a.on("change", function() {
  // emits on change
});
a.on("data", function() {
  // emits all data readings
});

var d = new five.Sensor.Digital(2);
d.on("change", function() {
  // emits on change
});
d.on("data", function() {
  // emits all data readings
});
```

Resseguie, 2015-04-24:
@rwaldron Should we 1) add tests to cover callbacks and 2) maybe throw an error if pinMode not set before trying to do a read? (Similar to what we added for Stepper recently?)

rwaldron, 2015-04-24:
@Resseguie I want to see what's going here first.

```js
var tst = new five.Pin(cfg);
tst.on('data', dataCallback);

// This isn't necessary, and will just duplicate 
// what the previous line did.
tst.io.digitalRead(tst.pin, readCallback1);

// This won't work, because `tst` is an object, 
// not a number. Also, if this had attempted to use 
// the same pin number as the first line, there
// would've been an "occupied pin" warning.
five.Pin.read(tst, readCallback2);
```

Like I said, I want to know more about the specific sensor that was having problematic results before we make an decisions or write any patches.

mMerlin, 2015-04-24:
@rwaldron said:
> @mMerlin thanks again for such a great bug report :)

And I still left a potentially important (or at least useful) detail of the environment out.  I tested this with Arduino boards Uno and Duemilanove.

BTW is there a simple way to get node+johnny-five to report details of the attached board, that would be good to paste into bug reports?

@rwaldron said:
>>    I started by trying to create a digital sensor and could not get the .on('data', callback) to actually execute the callback. So I dropped down to the simpler Pin object and tried again. Still nothing.

> Can you give more detail about the sensor you're using?

Will do as separate comment.  A bit big to insert here.

>>    Below are 2 almost identical scripts.

> A few things I've noticed about both:

> Neither have a call to pinMode
>        This will work with analogRead because analog pins (in Arduino parlance: "A0" and friends) don't need the mode set explicitly, so StandardFirmata does not [...]

I did not even think about pinMode, since mode was being set directly during the creation of the pin object.  Which is where the previous bug report came from when I discover that the mode was not what I just set.
```javascript
var cfg = {
  id: 'Digital test',
  pin: '4',
  type: 'digital',
  range: [0, 1],
  mode: five.Pin.INPUT,
  freq: 1000
};
// ...
var tst = new five.Pin(cfg);
```
I added
```javascript
this.pinMode(cfg.pin, five.Pin.INPUT);
```
just after creating the tst Pin instance.  No change.

>    I wonder why you're using the io methods? Johnny-Five provides a Sensor class, which is an analog sensor component class by default, or can be set to "digital". The class does all the setup work internally.

I got to the raw methods exploring alternatives, when I could not get .on('data') to work.  Either with the Sensor, or with Pin.

@Resseguie said:
> Should we 1) add tests to cover callbacks and 2) maybe throw an error if pinMode not set before trying to do a read? (Similar to what we added for Stepper recently?)

I looked at test/sensor.js and could not see how to add tests for the callbacks.  I need something simpler to work on first.

@rwaldron said:
```javascript
var tst = new five.Pin(cfg);
tst.on('data', dataCallback);

// This isn't necessary, and will just duplicate 
// what the previous line did.
tst.io.digitalRead(tst.pin, readCallback1);

// This won't work, because `tst` is an object, 
// not a number. Also, if this had attempted to use 
// the same pin number as the first line, there
// would've been an "occupied pin" warning.
five.Pin.read(tst, readCallback2);
```
Won't work?  **five.Pin.read(tst.pin, readCallback2);** crashes with *Cannot call method 'undefinedRead' of undefined at Function.Pin.read (pin.js:194)*.  Pin.read (pin.js:189) expects the first argument to be an object.  It references (and sets) pin.mode, then passed pin.addr to pin.io[pin.type + "Read"]

With that explicit set of pin.mode by Pin.read, I expect there should be no need to call .pinMode before setting up the callbacks.  At least if **tst.mode = five.Pin.INPUT** is internally equivalent to **this.pinMode(tst.pin, five.Pin.INPUT)**

Resseguie, 2015-04-24:
> > Should we 1) add tests to cover callbacks

> I looked at test/sensor.js and could not see how to add tests for the callbacks. I need something simpler to work on first.

The simplest example testing callbacks is the one for stepper that was added recently:
https://github.com/rwaldron/johnny-five/blob/master/test/stepper.js#L358

rwaldron, 2015-04-24:
>  Pin.read (pin.js:189) expects the first argument to be an object. It references (and sets) pin.mode, then passed pin.addr to pin.io[pin.type + "Read"]

Whoops, you're right. I completely forgot about that. I also documented it incorrectly. Sorry for the total confusion there. Updated: https://github.com/rwaldron/johnny-five/wiki/Pin#methods

Beyond that, I can't really help figure out why a sensor isn't being read properly unless I know what kind of sensor it is—that's more important than any code examples.

mMerlin, 2015-04-24:
@rwaldron said:
> Beyond that, I can't really help figure out why a sensor isn't being read properly unless I know what kind of sensor it is—that's more important than any code examples. 

Not sure what you mean by 'kind of sensor'.  The (back several steps/layers) context was a simple external digital input.  When I started this, I was looking at something to use for a heartbeat for some plot.ly processing.  Since then, I found Board.loop(), which works just fine for that application.  Still want to be able to 'listen' to custom hardware, that uses a simple digital (status|state) pin.

Below  are my 2 test scripts for digital and analog Sensor class instances.  Analog works fine, digital does not output anything.  With or without the call to pinMode.  For the testing, nothing is physically connected to the Arduino.  Just trying to get the callback to work.

Visually looking at the Arduino board, with analog, even with the tst.on() commented out, the tx led flickers constantly.  johnny-five is 'talking' to the board, filling the samples array at lib/sensor.js:92.  That array is what is actual used for input, then emitted at sensor.js:127.

With the digital sensor, the tx light does not flash.  There does not seem to be anything equivalent to the samples array.  A breakpoint says that the "Read event loop" anonymous function starting at sensor.js:71 is never actually called (for digital sensors).  That has code that looks like it is intended to trigger the event handler callback at sensor.js:84, but that anonymous function is never called.  Given what I saw / explored with the raw digital Pin instances, it appears there is a problem with io.digitalRead().  

I finally found nodemodules/firmata/lib/firmata.js, which defines Board.protoype.analogRead() and .digitalRead.  I have not yet trieded accessing those directly.  From pieces seen with previous exploring, the listener for "digital-read-" + pin is not working

A breakpoint says the code does reach sensor.js:104, but that just exits, doing nothing.  Since (for whatever reason) the event loop code at sensor.js:72 is not reached for digital sensors, nothing is ever emitted.

The (working) analog Sensor test code
```javascript
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Analog test',
  pin: 'A5',
  type: 'analog',
  range: [0, 1023],
  freq: 1000
};

function dataCallback() {
  /* jshint validthis:true */
  console.log('data callback for:', this.id, this.value);
}

function boardIsReady() {
  /* jshint validthis:true */
  var tst = new five.Sensor(cfg);
  // this.pinMode(cfg.pin, five.Pin.INPUT);
  tst.on('data', dataCallback);
}

myBoard.on('ready', boardIsReady);
```

The (failing) digital Sensor test code
```javascript
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Digital test',
  pin: 8,
  type: 'digital',
  range: [0, 1],
  freq: 1000
};

function dataCallback() {
  /* jshint validthis:true */
  console.log('data callback for:', this.id, this.value);
}

function boardIsReady() {
  /* jshint validthis:true */
  var tst = new five.Sensor(cfg);
  // this.pinMode(cfg.pin, five.Pin.INPUT);
  tst.on('data', dataCallback);
}

myBoard.on('ready', boardIsReady);
```

Resseguie, 2015-04-24:
@mMerlin As a side note, I'm quite interested in plot.ly myself, so I'd love to see what all you're doing with that. Feel free to hit me up separately on gitter, twitter, etc.

mMerlin, 2015-04-24:
@Resseguie 
See the brand new repository https://github.com/mMerlin/auto-water
You should (I think) be able to leave comments using issues there.
That project was a restructure / refactor / cleanup of code created from various examples by someone else at WCRS http://www.robotgames.com/
That was the first place I saw johnny-five or plot.ly.  I got involved because the original code was not working, and I am the usual local person to get asked about any programming issues.  Once things were cleaned up, it was easy to see the problem.  Plot.ly needs a heartbeat when events are spread more than a minute apart.

rwaldron, 2015-04-24:
> Not sure what you mean by 'kind of sensor'. 

I was just wanted to understand what you were trying to accomplish. Who knows, maybe you were plugging an analog sensor into a digital port and expecting that to work? I don't know, it's just one of the things I always ask.

> Still want to be able to 'listen' to custom hardware, that uses a simple digital (status|state) pin.

Yes, this shouldn't be a problem. 

I located a bug in `Sensor` class that was trying to short circuit the read callback for sensors that were type=digital. I fixed that issue and confirmed in two ways, using the following code: 

```js
var five = require("johnny-five");
var myBoard = new five.Board();

myBoard.on("ready", function() {
  var tst = new five.Sensor({
    pin: 8,
    type: "digital"
  });

  tst.on("data", function() {
    console.log(this.value);
  });
});
```

- nothing connected; the result is logging 0 forever
- with this circuit: http://johnny-five.io/examples/button/#illustrations-/-photos; the result is logging 0 forever and 1 when I press the button.

That patch is pushed and a new release of johnny-five published. 

----

The `Pin` issue was the result of differences in how StandardFirmata reports analog and digital pins. Analog gets reported for every cycle, but digital does not report unless a value has changed. Johnny-Five does it's best to account for that, however the `Pin` class was doing no such accounting. This has also been addressed. 

I confirmed with the same two circuits described above with this code: 

```js
var five = require("johnny-five");
var myBoard = new five.Board();

myBoard.on("ready", function() {

  var pin = new five.Pin({
    pin: 8,
    mode: five.Pin.INPUT
  });

  pin.on("data", function() {
    console.log(this.value);
  });

  pin.on("high", function() {
    console.log("high!");
  });

  pin.on("low", function() {
    console.log("low!");
  });
});
```

rwaldron, 2015-04-24:
I'll wait to for confirmation before closing.

mMerlin, 2015-04-25:
I ran johnny-five@0.8.61 through my various sample code.  Here are the results:
* analog Sensor works the same, and is correct
* digital sensor is now calling the data event handler, but there is no read result value.  Both the value passed as an argument to the handler, and this.value are null.
* analog Pin works the same, and is correct
* digital Pin is now calling the data event handler, and (with nothing attached to the board) this.value === 0
* tst.io.digitalRead(tst.pin, callback) is not executing the callback function
* five.Pin.read(tst, callback) is not executing the callback function

rwaldron, 2015-04-25:
> digital sensor is now calling the data event handler, but there is no read result value. 

- What do you mean, specifically, by "no read result". The following two programs, when run with nothing attached to the board, have the same output: 

```js
// 750-pin.js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var pin = new five.Pin({
    pin: 8,
    mode: five.Pin.INPUT
  });

  pin.on("data", function() {
    console.log("Pin(mode=INPUT)", this.value);
  });

  this.wait(2000, function() {
    process.exit(0);
  });
});
```

```js
// 750-sensor-digital.js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var pin = new five.Sensor({
    pin: 8,
    type: "digital"
  });

  pin.on("data", function() {
    console.log("Sensor(type=digital)", this.value);
  });

  this.wait(2000, function() {
    process.exit(0);
  });
});
```

Here is the output I recorded: https://gist.github.com/rwaldron/3481fcc67629e14f4a08

> Both the value passed as an argument to the handler, and this.value are null.

- There is a legacy bug, dating back to 3 years ago in which I stupidly put an error argument as the first argument to event handlers—that fix will be introduced in v0.9.0. (This means that the data is available as the second argument, but it's better to use the instance's `this.value`.)
- Can you share the _exact_ code that outputs `null` as the value of `this.value`. 

> - tst.io.digitalRead(tst.pin, callback) is not executing the callback function
> - five.Pin.read(tst, callback) is not executing the callback function

As I said: 

>> The Pin issue was the result of differences in how StandardFirmata reports analog and digital pins. Analog gets reported for every cycle, but digital does not report unless a value has changed.

That means: digitalRead doesn't get a message to invoke its handler unless there is a change in the physical state of the pin.

Please don't include `foo.io.digitalRead(...)` in the testing done here, that's provided by Firmata.js and will continue to _only_ call the callback when there are changes to report—as per the StandardFirmata protocol.

> five.Pin.read(tst, callback) is not executing the callback function

Yep, this bug is the result of my overlooking expected operational behavior parity—sorry about that. I've landed a fix with a set of tests that should prevent that mistake. I've published a new released. Please confirm.

mMerlin, 2015-04-26:
Environment:
$ uname -srvpio
Linux 3.19.4-200.fc21.x86_64 #1 SMP Mon Apr 13 21:43:26 UTC 2015 x86_64 x86_64 GNU/Linux
$ node -v
v0.10.38
$ npm list johnny-five
...
└── johnny-five@0.8.62 
Arduino Duemilanove with standardFirmata

That null result was for Sensor, not Pin.  With dummy loads attached to the board.  It is setup to drive leds on every pin, led + resistor to ground:
```javascript
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Digital test',
  pin: 8,
  type: 'digital',
  range: [0, 1],
  freq: 1000
};

function dataCallback(err, val, more) {
  /* jshint validthis:true */
  console.log('data for:', this.id, 'with value:', this.value,
    'and arguments', err, val, more
    );
}

function boardIsReady() {
  /* jshint validthis:true */
  var tst = new five.Sensor(cfg);
  // this.pinMode(cfg.pin, five.Pin.INPUT);
  tst.on('data', dataCallback);
}

myBoard.on('ready', boardIsReady);
```
The output I get from the above is:
```
data for: Digital test with value: null and arguments null null
```

> Please don't include foo.io.digitalRead(...) in the testing done here, that's provided by Firmata.js and will continue to only call the callback when there are changes to report—as per the StandardFirmata protocol.

OK.  As long as I know what to expect as 'normal' operation.

> Yep, this bug is the result of my overlooking expected operational behavior parity—sorry about that. I've landed a fix with a set of tests that should prevent that mistake. I've published a new released. Please confirm. 

Confirmed.  0.8.62 handles my sample case.  Once I saw your description of the cause, I had a thought about a simple implementation to resolved it.  I browsed through your fix.  Mine is 'simpler', but might not handle everything yours does.  Have a look at this.  I stuck a hard-coded 'fake' in the doRead function, just so I could tell the output apart.  .bind is powerful.
```javascript
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Digital test',
  pin: '2',
  freq: 1000
};
function readCallback(val, more) {
  /* jshint validthis:true */
  console.log('read callback with:', val, more, this.id, this.value);
}
function doRead(cb) {
  /* jshint validthis: true */
  // cb.call(this, null, this.value);
  cb.call(this, 'fake', this.value);
}
five.Pin.read2 = function (pin, callback) {
  pin.board.loop(pin.freq, doRead.bind(pin, callback));
};
myBoard.on('ready', function () {
  /* jshint validthis: true */
  var tst = new five.Pin(cfg);
  five.Pin.read(tst, readCallback);
  five.Pin.read2(tst, readCallback);
});
```
That simply 'emits' the callback, with whatever the current stored value is, on a timed cycle based on the freq property.  It (and I based on your description) assumes that the actual value is updated by other standard processing, when the state actually changes.

mMerlin, 2015-05-07:
Still getting null results from that sample code with johnny-five@0.8.71

Unless the fix is supposed to come as part of 0.9.0 ?

rwaldron, 2015-05-07:
Thanks, can you be more specific? Which example?

mMerlin, 2015-05-07:
Current environment
```bash
$ uname -srvpio
Linux 3.19.5-200.fc21.x86_64 #1 SMP Mon Apr 20 19:51:56 UTC 2015 x86_64 x86_64 GNU/Linux
$ node --version
v0.10.38
$ npm list johnny-five
j5-exploring@0.0.1 /home/phil/development/projects/j5-exploring
└── johnny-five@0.8.71 
```
_d-test.js_
```javascript
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Digital test',
  pin: 8,
  type: 'digital',
  range: [0, 1],
  freq: 1000
};

function dataCallback(err, val) {
  /* jshint validthis:true */
  console.log('data for:', this.id, 'with value:', this.value,
    'and arguments', err, val);
}

function boardIsReady() {
  var tst = new five.Sensor(cfg);
  tst.on('data', dataCallback);
}

myBoard.on('ready', boardIsReady);
```
```bash
$ node d-test.js
1431023194113 Device(s) /dev/ttyUSB0 
1431023194171 Connected /dev/ttyUSB0 
1431023197326 Repl Initialized 
>> data for: Digital test with value: null and arguments null null
data for: Digital test with value: null and arguments null null
data for: Digital test with value: null and arguments null null
1431023203894 Board Closing. 
```

rwaldron, 2015-05-07:
```bash
rwaldron at nova in ~/j5-test
$ npm install johnny-five

> galileo-io@0.8.12 postinstall /Users/rwaldron/j5-test/node_modules/johnny-five/node_modules/galileo-io
> node scripts/postinstall

-
> serialport@1.6.3 install /Users/rwaldron/j5-test/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/Users/rwaldron/j5-test/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.6.3/Release/node-v11-darwin-x64/serialport.node" is installed via remote
johnny-five@0.8.71 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── temporal@0.4.0
├── color-convert@0.5.2
├── es6-shim@0.31.1
├── nanotimer@0.3.1
├── chalk@1.0.0 (escape-string-regexp@1.0.3, ansi-styles@2.0.1, supports-color@1.3.1, strip-ansi@2.0.1, has-ansi@1.0.3)
├── galileo-io@0.8.12 (remapped@0.2.1, es6-promise@2.1.1)
├── array-includes@1.1.0 (define-properties@1.0.1, es-abstract@1.2.1)
├── lodash@3.8.0
├── firmata@0.4.1 (object-assign@1.0.0, browser-serialport@2.0.2, es6-map@0.1.1)
└── serialport@1.6.3 (bindings@1.2.1, sf@0.1.7, async@0.9.0, nan@1.7.0, debug@2.1.3, optimist@0.6.1)
rwaldron at nova in ~/j5-test
$ node 750-new.js
1431027034762 Device(s) /dev/cu.usbmodem1421
1431027034775 Connected /dev/cu.usbmodem1421
1431027038240 Repl Initialized
>> data for: Digital test with value: 0 and arguments null 0
data for: Digital test with value: 0 and arguments null 0
data for: Digital test with value: 0 and arguments null 0
data for: Digital test with value: 0 and arguments null 0
data for: Digital test with value: 0 and arguments null 0

(^C again to quit)
>>
1431027043947 Board Closing.
rwaldron at nova in ~/j5-test
$ cat 750-new.js
'use strict';
var five = require('johnny-five');
var myBoard = new five.Board();
var cfg = {
  id: 'Digital test',
  pin: 8,
  type: 'digital',
  range: [0, 1],
  freq: 1000
};

function dataCallback(err, val) {
  /* jshint validthis:true */
  console.log('data for:', this.id, 'with value:', this.value,
    'and arguments', err, val);
}

function boardIsReady() {
  var tst = new five.Sensor(cfg);
  tst.on('data', dataCallback);
}

myBoard.on('ready', boardIsReady);
```

mMerlin, 2015-05-07:
It seems we have some sort of environment difference.

Tested with 2 different Arduino Duemilanove boards (328), an UNO, and a Nano (328),
all with the same results.

StandardFirmata example sketch loaded from Ardiuno IDE version 1.0.6

First install below used the latest commit from the repository.  Second used
whatever npm thinks is the 'latest'

```
$ rm -rf node_modules/johnny-five/
$ grep "johnny-five" package.json 
  "description": "Explorations of node.js with johnny-five for communications with Arduino boards",
    "johnny-five"
    "johnny-five": "rwaldron/johnny-five"
$ npm install
\
> galileo-io@0.8.13 postinstall /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/galileo-io
> node scripts/postinstall

-
> serialport@1.6.3 install /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.6.3/Release/node-v11-linux-x64/serialport.node" is installed via remote
johnny-five@0.8.71 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── temporal@0.4.0
├── color-convert@0.5.2
├── es6-shim@0.31.1
├── chalk@1.0.0 (escape-string-regexp@1.0.3, ansi-styles@2.0.1, supports-color@1.3.1, strip-ansi@2.0.1, has-ansi@1.0.3)
├── nanotimer@0.3.1
├── array-includes@1.1.0 (define-properties@1.0.1, es-abstract@1.2.1)
├── galileo-io@0.8.13 (remapped@0.2.1, es6-promise@2.1.1)
├── lodash@3.8.0
├── firmata@0.4.1 (object-assign@1.0.0, browser-serialport@2.0.2, es6-map@0.1.1)
└── serialport@1.6.3 (bindings@1.2.1, sf@0.1.7, async@0.9.0, nan@1.7.0, debug@2.1.3, optimist@0.6.1)
$ node d-test
1431035953986 Device(s) /dev/ttyUSB0 
1431035954007 Connected /dev/ttyUSB0 
1431035957142 Repl Initialized 
>> data for: Digital test with value: null and arguments null null
data for: Digital test with value: null and arguments null null
data for: Digital test with value: null and arguments null null
data for: Digital test with value: null and arguments null null
1431035961380 Board Closing.

$ rm -rf node_modules/johnny-five/
$ npm install johnny-five
-
> galileo-io@0.8.13 postinstall /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/galileo-io
> node scripts/postinstall

/
> serialport@1.6.3 install /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.6.3/Release/node-v11-linux-x64/serialport.node" is installed via remote
johnny-five@0.8.71 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── temporal@0.4.0
├── color-convert@0.5.2
├── es6-shim@0.31.1
├── chalk@1.0.0 (escape-string-regexp@1.0.3, ansi-styles@2.0.1, supports-color@1.3.1, strip-ansi@2.0.1, has-ansi@1.0.3)
├── nanotimer@0.3.1
├── array-includes@1.1.0 (define-properties@1.0.1, es-abstract@1.2.1)
├── galileo-io@0.8.13 (remapped@0.2.1, es6-promise@2.1.1)
[... 134 lines cut ...]
  │ │   │ ├── graceful-fs@2.0.3
  │ │   │ ├── inherits@2.0.1
  │ │   │ └── mkdirp@0.3.5
  │ │   ├─┬ fstream-ignore@0.0.7
  │ │   │ ├── inherits@2.0.1
  │ │   │ └─┬ minimatch@0.2.14
  │ │   │   ├── lru-cache@2.5.0
  │ │   │   └── sigmund@1.0.0
  │ │   ├── graceful-fs@1.2.3
  │ │   ├── once@1.1.1
  │ │   ├─┬ readable-stream@1.0.27-1 invalid
  │ │   │ ├── core-util-is@1.0.1
  │ │   │ ├── inherits@2.0.1
  │ │   │ ├── isarray@0.0.1
  │ │   │ └── string_decoder@0.10.25-1 invalid
  │ │   └── uid-number@0.0.3
  │ ├─┬ optimist@0.6.1
  │ │ ├── minimist@0.0.10
  │ │ └── wordwrap@0.0.3
  │ └── sf@0.1.7
  └── temporal@0.4.0

npm ERR! invalid: node-pre-gyp@0.5.19 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp
npm ERR! invalid: readable-stream@1.0.27-1 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream
npm ERR! invalid: string_decoder@0.10.25-1 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream/node_modules/string_decoder

rwaldron, 2015-05-08:
What are those failures at the end? I'll look at the more tomorrow

mMerlin, 2015-05-08:
I did some research on the errors.  I am not a linux expert, but here is what the analysis showed.

I **assumed** that the messages mean there is a problem between the available package versions, and what package manager for the packages says is needed.  Here is a simpler list, that shows the chain with the failing lines.
```
$ npm list string_decoder
j5-exploring@0.0.1 /home/phil/development/projects/j5-exploring
└─┬ johnny-five@0.8.71
  └─┬ serialport@1.6.3
    └─┬ node-pre-gyp@0.5.19 invalid
      └─┬ tar-pack@2.0.0
        └─┬ readable-stream@1.0.27-1 invalid
          └── string_decoder@0.10.25-1  invalid

npm ERR! invalid: node-pre-gyp@0.5.19 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp
npm ERR! invalid: readable-stream@1.0.27-1 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream
npm ERR! invalid: string_decoder@0.10.25-1 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream/node_modules/string_decoder
```
First verify that it is easily reproducible with a fresh install:
```
$ mkdir rawtest
$ cd rawtest/
$ npm install johnny-five

> galileo-io@0.8.13 postinstall /home/phil/development/projects/rawtest/node_modules/johnny-five/node_modules/galileo-io
> node scripts/postinstall

\
> serialport@1.6.3 install /home/phil/development/projects/rawtest/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/home/phil/development/projects/rawtest/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.6.3/Release/node-v11-linux-x64/serialport.node" is installed via remote
johnny-five@0.8.71 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── temporal@0.4.0
├── color-convert@0.5.2
├── es6-shim@0.31.1
├── nanotimer@0.3.1
├── chalk@1.0.0 (ansi-styles@2.0.1, escape-string-regexp@1.0.3, supports-color@1.3.1, has-ansi@1.0.3, strip-ansi@2.0.1)
├── galileo-io@0.8.13 (remapped@0.2.1, es6-promise@2.1.1)
├── array-includes@1.1.0 (define-properties@1.0.1, es-abstract@1.2.1)
├── lodash@3.8.0
├── firmata@0.4.1 (object-assign@1.0.0, browser-serialport@2.0.2, es6-map@0.1.1)
└── serialport@1.6.3 (bindings@1.2.1, sf@0.1.7, async@0.9.0, nan@1.7.0, debug@2.1.3, optimist@0.6.1)

$ npm list string_decoder
/home/phil/development/projects/rawtest
└─┬ johnny-five@0.8.71
  └─┬ serialport@1.6.3
    └─┬ node-pre-gyp@0.5.19 invalid
      └─┬ tar-pack@2.0.0
        └─┬ readable-stream@1.0.27-1 invalid
          └── string_decoder@0.10.25-1  invalid

npm ERR! invalid: node-pre-gyp@0.5.19 /home/phil/development/projects/rawtest/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp
npm ERR! invalid: readable-stream@1.0.27-1 /home/phil/development/projects/rawtest/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream
npm ERR! invalid: string_decoder@0.10.25-1 /home/phil/development/projects/rawtest/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream/node_modules/string_decoder
```

Tracking (down) through the dependancy chain,
* j5-explorer package.json has dependency "johnny-five": "rwaldron/johnny-five"
* node_modules/johnny-five package.json has dependency "serialport": "latest"
 * and optional dependency "serialport": "latest"
 * https://github.com/rwaldron/johnny-five/blob/master/package.json only has optional dependency "serialport": "latest"
  * The local johnny-five package.json has reformatted contributor information, is it auto generated?  Perhaps merging dependencies of dependencies?
* node_modules/johnny-five/node_modules/serialport
  * package.json has
    * dependency "node-pre-gyp": "0.6.x"
    * "bundleDependencies": [ "node-pre-gyp" ]
  * https://github.com/voodootikigod/node-serialport/blob/master/package.json has:
[... 110 lines cut ...]
  "description": "Explorations of node.js with johnny-five for communications with Arduino boards",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "repository": {
    "type": "git",
    "url": "https://github.com/mmerlin/j5-exploring.git"
  },
  "keywords": [
    "node",
    "arduino",
    "johnny-five"
  ],
  "author": "H. Phil Duby",
  "license": "MIT",
  "bugs": {
    "url": "https://github.com/mmerlin/j5-exploring/issues"
  },
  "homepage": "https://github.com/mmerlin/j5-exploring",
  "dependencies": {
    "serialport": "latest"
  }
}
```

mMerlin, 2015-05-08:
Simple local fix to get rid of the npm error messages

```
$ cd node_modules/johnny-five/node_modules/serialport
$ npm list node-pre-gyp
serialport@1.6.3 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport
└── node-pre-gyp@0.5.19  invalid
$ npm update node-pre-gyp
node-pre-gyp@0.6.7 node_modules/node-pre-gyp
$ npm list node-pre-gyp
serialport@1.6.3 /home/phil/development/projects/j5-exploring/node_modules/johnny-five/node_modules/serialport
└── node-pre-gyp@0.6.7
```
cd back to the j5-exploring project folder
```
$ npm list
j5-exploring@0.0.1 /home/phil/development/projects/j5-exploring
└─┬ johnny-five@0.8.71
  ├─┬ array-includes@1.1.0
  │ ├─┬ define-properties@1.0.1
  │ │ ├── foreach@2.0.5
  │ │ └── object-keys@1.0.3
  │ └─┬ es-abstract@1.2.1
  │   ├─┬ es-to-primitive@1.0.0
  │   │ ├── is-date-object@1.0.0
  │   │ └── is-symbol@1.0.1
  │   └── is-callable@1.0.4
  ├─┬ chalk@1.0.0
  │ ├── ansi-styles@2.0.1
  │ ├── escape-string-regexp@1.0.3
  │ ├─┬ has-ansi@1.0.3
  │ │ ├── ansi-regex@1.1.1
  │ │ └── get-stdin@4.0.1
  │ ├─┬ strip-ansi@2.0.1
  │ │ └── ansi-regex@1.1.1
  │ └── supports-color@1.3.1
  ├── color-convert@0.5.2
  ├── descriptor@0.1.0
  ├── ease-component@1.0.0
  ├── es6-shim@0.31.1
  ├─┬ firmata@0.4.1
  │ ├── browser-serialport@2.0.2
  │ ├─┬ es6-map@0.1.1
  │ │ ├── d@0.1.1
  │ │ ├─┬ es5-ext@0.10.7
  │ │ │ └── es6-symbol@2.0.1
  │ │ ├─┬ es6-iterator@0.1.3
  │ │ │ └── es6-symbol@2.0.1
  │ │ ├── es6-set@0.1.1
  │ │ ├── es6-symbol@0.1.1
  │ │ └── event-emitter@0.3.3
  │ └── object-assign@1.0.0
  ├─┬ galileo-io@0.8.13
  │ ├── es6-promise@2.1.1
  │ └─┬ remapped@0.2.1
  │   ├── getobject@0.1.0
  │   └── traverse@0.6.6
  ├── lodash@3.8.0
  ├── nanotimer@0.3.1
  ├─┬ serialport@1.6.3
  │ ├── async@0.9.0
  │ ├── bindings@1.2.1
  │ ├─┬ debug@2.1.3
  │ │ └── ms@0.7.0
  │ ├── nan@1.7.0
  │ ├─┬ node-pre-gyp@0.6.7
  │ │ ├─┬ mkdirp@0.5.0
  │ │ │ └── minimist@0.0.8
  │ │ ├─┬ nopt@3.0.1
  │ │ │ └── abbrev@1.0.5
  │ │ ├─┬ npmlog@1.2.0
  │ │ │ ├── ansi@0.3.0
  │ │ │ ├─┬ are-we-there-yet@1.0.4
  │ │ │ │ ├── delegates@0.1.0
  │ │ │ │ └─┬ readable-stream@1.1.13
  │ │ │ │   ├── core-util-is@1.0.1
  │ │ │ │   ├── inherits@2.0.1
  │ │ │ │   ├── isarray@0.0.1
  │ │ │ │   └── string_decoder@0.10.31
  │ │ │ └─┬ gauge@1.2.0
  │ │ │   ├── has-unicode@1.0.0
  │ │ │   ├─┬ lodash.pad@3.1.0
  │ │ │   │ ├── lodash._basetostring@3.0.0
  │ │ │   │ └─┬ lodash._createpadding@3.6.0
  │ │ │   │   └── lodash.repeat@3.0.0
  │ │ │   ├─┬ lodash.padleft@3.1.1
  │ │ │   │ ├── lodash._basetostring@3.0.0
  │ │ │   │ └─┬ lodash._createpadding@3.6.0
  │ │ │   │   └── lodash.repeat@3.0.0
  │ │ │   └─┬ lodash.padright@3.1.1
  │ │ │     ├── lodash._basetostring@3.0.0
  │ │ │     └─┬ lodash._createpadding@3.6.0
  │ │ │       └── lodash.repeat@3.0.0
  │ │ ├─┬ rc@1.0.1
  │ │ │ ├── deep-extend@0.2.11
  │ │ │ ├── ini@1.3.3
  │ │ │ ├── minimist@0.0.10
  │ │ │ └── strip-json-comments@0.1.3
  │ │ ├─┬ request@2.55.0
  │ │ │ ├── aws-sign2@0.5.0
  │ │ │ ├─┬ bl@0.9.4
  │ │ │ │ └─┬ readable-stream@1.0.33
  │ │ │ │   ├── core-util-is@1.0.1
  │ │ │ │   ├── inherits@2.0.1
  │ │ │ │   ├── isarray@0.0.1
  │ │ │ │   └── string_decoder@0.10.31
  │ │ │ ├── caseless@0.9.0
  │ │ │ ├─┬ combined-stream@0.0.7
  │ │ │ │ └── delayed-stream@0.0.5
  │ │ │ ├── forever-agent@0.6.1
  │ │ │ ├─┬ form-data@0.2.0
  │ │ │ │ └── async@0.9.0
  │ │ │ ├─┬ har-validator@1.7.0
  │ │ │ │ ├── bluebird@2.9.25
  │ │ │ │ ├─┬ chalk@1.0.0
  │ │ │ │ │ ├── ansi-styles@2.0.1
  │ │ │ │ │ ├── escape-string-regexp@1.0.3
  │ │ │ │ │ ├─┬ has-ansi@1.0.3
  │ │ │ │ │ │ ├── ansi-regex@1.1.1
  │ │ │ │ │ │ └── get-stdin@4.0.1
  │ │ │ │ │ ├─┬ strip-ansi@2.0.1
  │ │ │ │ │ │ └── ansi-regex@1.1.1
  │ │ │ │ │ └── supports-color@1.3.1
  │ │ │ │ ├─┬ commander@2.8.1
  │ │ │ │ │ └── graceful-readlink@1.0.1
  │ │ │ │ └─┬ is-my-json-valid@2.10.1
  │ │ │ │   ├── generate-function@2.0.0
  │ │ │ │   ├─┬ generate-object-property@1.1.1
  │ │ │ │   │ └── is-property@1.0.2
  │ │ │ │   ├── jsonpointer@1.1.0
  │ │ │ │   └── xtend@4.0.0
  │ │ │ ├─┬ hawk@2.3.1
  │ │ │ │ ├── boom@2.7.1
  │ │ │ │ ├── cryptiles@2.0.4
  │ │ │ │ ├── hoek@2.13.0
  │ │ │ │ └── sntp@1.0.9
  │ │ │ ├─┬ http-signature@0.10.1
  │ │ │ │ ├── asn1@0.1.11
  │ │ │ │ ├── assert-plus@0.1.5
  │ │ │ │ └── ctype@0.5.3
  │ │ │ ├── isstream@0.1.2
  │ │ │ ├── json-stringify-safe@5.0.0
  │ │ │ ├─┬ mime-types@2.0.10
  │ │ │ │ └── mime-db@1.8.0
  │ │ │ ├── node-uuid@1.4.3
  │ │ │ ├── oauth-sign@0.6.0
  │ │ │ ├── qs@2.4.1
  │ │ │ ├── stringstream@0.0.4
  │ │ │ ├── tough-cookie@1.1.0
  │ │ │ └── tunnel-agent@0.4.0
  │ │ ├─┬ rimraf@2.3.3
  │ │ │ └─┬ glob@4.5.3
  │ │ │   ├─┬ inflight@1.0.4
  │ │ │   │ └── wrappy@1.0.1
  │ │ │   ├── inherits@2.0.1
  │ │ │   ├─┬ minimatch@2.0.7
  │ │ │   │ └─┬ brace-expansion@1.1.0
  │ │ │   │   ├── balanced-match@0.2.0
  │ │ │   │   └── concat-map@0.0.1
  │ │ │   └─┬ once@1.3.1
  │ │ │     └── wrappy@1.0.1
  │ │ ├── semver@4.3.3
  │ │ ├─┬ tar@2.1.0
  │ │ │ ├── block-stream@0.0.7
  │ │ │ ├─┬ fstream@1.0.4
  │ │ │ │ └── graceful-fs@3.0.6
  │ │ │ └── inherits@2.0.1
  │ │ └─┬ tar-pack@2.0.0
  │ │   ├── debug@0.7.4
  │ │   ├─┬ fstream@0.1.31
  │ │   │ ├── graceful-fs@3.0.6
  │ │   │ └── inherits@2.0.1
  │ │   ├─┬ fstream-ignore@0.0.7
  │ │   │ ├── inherits@2.0.1
  │ │   │ └─┬ minimatch@0.2.14
  │ │   │   ├── lru-cache@2.6.2
  │ │   │   └── sigmund@1.0.0
  │ │   ├── graceful-fs@1.2.3
  │ │   ├── once@1.1.1
  │ │   ├─┬ readable-stream@1.0.33
  │ │   │ ├── core-util-is@1.0.1
  │ │   │ ├── inherits@2.0.1
  │ │   │ ├── isarray@0.0.1
  │ │   │ └── string_decoder@0.10.31
  │ │   ├── rimraf@2.2.8
  │ │   ├─┬ tar@0.1.20
  │ │   │ ├── block-stream@0.0.7
  │ │   │ └── inherits@2.0.1
  │ │   └── uid-number@0.0.3
  │ ├─┬ optimist@0.6.1
  │ │ ├── minimist@0.0.10
  │ │ └── wordwrap@0.0.3
  │ └── sf@0.1.7
  └── temporal@0.4.0
```
No more error messages.

Did not change the output of **node d-test** though.  Still all null.

mMerlin, 2015-07-22:
After the npm enviroment update issues were resolved, we never got back to the problem of getting null values from the data event handler for digital sensors.  The npm fix did not actually resolve that for me.

rwaldron 2015/04/24
> I located a bug in Sensor class that was trying to short circuit the read callback for sensors that were type=digital. I fixed that issue and confirmed in two ways, using the following code:
```javascript
var five = require("johnny-five");
var myBoard = new five.Board();
>
myBoard.on("ready", function() {
  var tst = new five.Sensor({
    pin: 8,
    type: "digital"
  });
>
  tst.on("data", function() {
    console.log(this.value);
  });
});
```
> * nothing connected; the result is logging 0 forever
> * with this circuit: http://johnny-five.io/examples/button/#illustrations-/-photos; the result is logging 0 forever and 1 when I press the button.

mMerlin 2015/05/07
> Still getting null results from that sample code with johnny-five@0.8.71
> 
> Unless the fix is supposed to come as part of 0.9.0 ?

I partially replicated your results, and found what I think the real issue is.  Probably down in firmata, though johnny-five could resolve it.  If you change the circuit to have pin 8 pulled low initially, then drive it high with the switch, so that it should start reporting 0, I think you will find it reports null instead.  Push the switch, it reports 1, release, it reports 0, and works correctly from then on.  I suspect that your 'open circuit' was getting a enough noise on startup to trigger a toggle event.

My symptoms say that if the pin is (held) low when the board is initialized, the callback set by calling this.io["digitalRead"] does not get called initially.  Until the first time the pin goes high.  For digital pins, the associated event is only triggered on data changes, and the initial *previous* state is 0, so no event, no callback, and null value when the Sensor data event is triggered at the end of the freq interval.  This could also be from slightly different versions / sources for the "StandardFirmata" sketch.  I thought my version was current.  At least latest that is downloaded as part of the Arduino package for Fedora 21.

I would have expected that, like the analog Sensor change event, the event would always be triggered the first time, when there is no previous value to compare against.  A little trickier for digital / boolean values, but possible.

The trivial fix is, for digital sensors, to set the initial value of "value" to 0 instead of null.  That way, if no low level event is triggered first, the data event will show 0.  Not going to change that until you can replicate and confirm the result and logic.

rwaldron, 2015-11-05:
@mmerlin I'm sorry this took 4 months for me to follow up :( 

I thin the change you proposed is completely reasonable and here's why: all digital IO pins will have been initialized `LOW`, so it follows that `0` is the initial value once that pin's mode becomes input.

closed 2016-05-25 by reconbot