rwaldron/johnny-five #753: properties of state object passed to pin.query(callback)

mMerlin, 2015-04-25:
What are the properties (supposed to be) for the state object passed to the callback for pin.query()?  The documentation at http://johnny-five.io/api/pin/ and https://github.com/rwaldron/johnny-five/wiki/Pin show
```javascript
{ 
  supportedModes: [ 0, 1, 3, 4 ],
  mode: 0,
  value: 0,
  report: 1,
  analogChannel: 127 
}
```
but a samples I get in a test also include state: 0.  Is the documentation out of data?  What do the values of the state property mean?  I have only seen a zero in my limited exploration.

rwaldron, 2015-04-25:
This is a copy of the current in-program-memory representation of a given pin. This information is coordinated between Firmata.js and StandardFirmata. This is only a _copy_ so modifying it doesn't result in any observable changes to the pin (eg. setting value = 1 will bring the pin high). 

>  http://johnny-five.io/api/pin/ and https://github.com/rwaldron/johnny-five/wiki/Pin 

These are the exact same thing ;)

I've updated the docs to include a table that describes each property of the object http://johnny-five.io/api/pin/#api

closed 2015-04-25 by rwaldron

mMerlin, 2015-04-26:
> This is a copy of the current in-program-memory representation of a given pin. This information is coordinated between Firmata.js and StandardFirmata. This is only a copy so modifying it doesn't result in any observable changes to the pin (eg. setting value = 1 will bring the pin high). 

I assume you meant: setting value = 1 will NOT bring the pin high