rwaldron/johnny-five #754: Button: value property

bartmichu, 2015-04-26:
Hi, would you consider adding readonly ```value``` property to indicate a digital state reading? Something like in ```IR.Motion``` class.

I'm mocking about with the lights and reed switches (I use ```Button``` class for that) and it would be really helpful to simply check its current state.

rwaldron, 2015-04-26:
Thanks for the report—this is definitely a bug and yes, absolutely.  All component class instances should have that property, but some are still missing. Do you want to write a patch and some tests?

bartmichu, 2015-04-26:
Don't really know how to do that I'm affraid, I would probably break something in the process :)

bartmichu, 2015-04-28:
It looks like we can also use ```isDown``` property in this case, at least until it gets fixed.

rwaldron, 2015-04-28:
No worries, I just wanted to offer you the chance to fix it if you'd like to. I can take care of it. I'll ping when this is ready

rwaldron, 2015-04-28:
johnny-five@v0.8.64

closed 2015-04-28 by rwaldron