rwaldron/johnny-five #755: RPI / Arduino Nano hang on Device or Firmware Error

valencreative, 2015-04-26:
Hello Guys,

I currently developing a node-express-socket-johnny app for RPi and run into an issue when i tested my app.js on the raspberry pi2. It was developed and worked fine on my Mac with all the same module versions, but the RPi hang on this:

```
1430084459857 Device(s) /dev/ttyUSB0 
1430084459942 Connected /dev/ttyUSB0 
undefined
undefined
undefined
1430084469954 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting 
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (/home/bodobot/node_modules/johnny-five/lib/board.js:388:26)
    at Timer.listOnTimeout (timers.js:110:15)
```

It needs to be look like this btw (from my Mac):

```
1430087707599 Looking for connected device  
1430087722452 Device(s) /dev/cu.usbserial-AL008BQZ 
1430087722461 Connected /dev/cu.usbserial-AL008BQZ 
1430087727121 Repl Initialized 
>> 
```

thanks for the kind solutions

bryan-m-hughes, 2015-04-26:
Can you connect to the Nano from the Raspberry Pi using the Arduino IDE? It may be possible that you don't have the right FTDI drivers installed, and this would be a good way to verify one way or another.

rwaldron, 2015-10-08:
Please reopen if still an issue

closed 2015-10-08 by rwaldron