rwaldron/johnny-five #759: v0.9.0

rwaldron, 2015-04-29:
May 4th?

- [ ] [Remove bogus `null` arguments from emitters. Fixes gh-561 #667
](https://github.com/rwaldron/johnny-five/pull/667)

What else?

scottgonzalez, 2015-04-29:
[Standard Event object](https://github.com/rwaldron/johnny-five/issues/569) and [Migration to a 0-100% value range](https://github.com/rwaldron/johnny-five/issues/370) were recently brought up.

https://github.com/rwaldron/johnny-five/labels/0.9.0 should probably be updated to accurately reflect waht you want in 0.9.x.

Resseguie, 2015-04-29:
Should we get #742 in there too? cc @scottgonzalez 

And probably #737 since it's related to 0-100 ranges. cc @rwaldron 
Should we deprecate `brightness()` in Led AND RGB as part of 0.9? (related to above)

And #756? cc @dtex 

Should we remove Gripper as in #698 and then put it back later once a device/controller model is added? (cc me)

Does #665 need to be addressed before 0.9? cc @boneskull 

And we've got potentially breaking changes in #741 for piezo. Do we want those in before? cc @McDonnellDean

Resseguie, 2015-04-29:
And do we want Motion refactor in 0.9? #687 cc @frxnz

frxnz, 2015-04-29:
Is this the right time to remove deprecated classes like `IR.Distance`, `IR.Proximity`, and `Ping`?

I should also be able to get Motion refactored this weekend if we want to get that in.

Resseguie, 2015-04-29:
If some of those I listed aren't actually breaking, then of course we can wait.

scottgonzalez, 2015-04-30:
> Should we get #742 in there too? cc @scottgonzalez

I'm just waiting for my SparkFun and Adafruit orders to arrive so I have the PCA9685 and BlinkM controllers to work with.

McDonnellDean, 2015-04-30:
@Resseguie @rwaldron I can finish this before the 4th. I was really hopping to get some feedback on my changes before finalising. They aren't breaking though. I kept the old API fully intact so it can be sunsetted gracefully.

rwaldron, 2015-04-30:
I'll be honest, I'm a little overwhelmed by the number of large things we have here. I changed the date to May 11th, but still: some of things can't reasonably be done, and done well, in that time frame.

scottgonzalez, 2015-04-30:
Was 0.9.0 originally planned to be released so soon, or was the date just picked because there was an uptick in discussion and it seemed like there wasn't a lot to do?

rwaldron, 2015-04-30:
> Was 0.9.0 originally planned to be released so soon

I originally wanted to do it in January

> or was the date just picked because there was an uptick in discussion and it seemed like there wasn't a lot to do?

I made it up when I filed this, which I did from my phone at the dog park and didn't really consider all of the issues with 0.9.0 label.

boneskull, 2015-05-01:
If you're following semver, #665 would probably necessitate a major release.

Resseguie, 2015-05-01:
@rwaldron I added 0.9.0 label to all the things mentioned here. Feel free to remove ones that you deem not 0.9. I just figured that would make it easier for everyone to track.

scottgonzalez, 2015-05-01:
> If you're following semver, #665 would probably necessitate a major release.

Nothing ever requires a major release in 0.x versions.

boneskull, 2015-05-01:
@scottgonzalez Ah, yes, of course.  Well, regardless, I have no time to get to that in the immediate future.

closed 2015-11-05 by rwaldron