rwaldron/johnny-five #762: Controller initialization logic

scottgonzalez, 2015-04-30:
I noticed that many components support the same controllers and each component handles the controller initialization directly. For example, the PCA9685 controller is currently supported by:

* esc ([initialization block](https://github.com/rwaldron/johnny-five/blob/310cdd11a4331ff1552a69a49d12826ea5e0b0f0/lib/esc.js#L26-L45))
* servo ([initialization block](https://github.com/rwaldron/johnny-five/blob/310cdd11a4331ff1552a69a49d12826ea5e0b0f0/lib/servo.js#L53-L72))
* led ([initialization block](https://github.com/rwaldron/johnny-five/blob/310cdd11a4331ff1552a69a49d12826ea5e0b0f0/lib/led/led.js#L25-L44))
* motor ([initialization block](https://github.com/rwaldron/johnny-five/blob/310cdd11a4331ff1552a69a49d12826ea5e0b0f0/lib/motor.js#L128-L145))

The initialization blocks are mostly the same, but there are some inconsistencies:

* `esc` and `servo` use `i2cWriteReg()`; `led` uses `i2cWrite()`; and `motor` uses `sendI2CWriteRequest()`.
* `esc`, `servo`, and `led` set `0x70` for the prescalar; `motor` sets `0x3`.
* `led` and `motor` do additional setup during the controller intialization ([`led` logic](https://github.com/rwaldron/johnny-five/blob/310cdd11a4331ff1552a69a49d12826ea5e0b0f0/lib/led/led.js#L46-L56),[`motor` logic](https://github.com/rwaldron/johnny-five/blob/310cdd11a4331ff1552a69a49d12826ea5e0b0f0/lib/motor.js#L147-L150)).

I haven't looked at other controllers yet, but I assume similar behavior exists.

----

I'd recommend creating controller modules that contain the common logic and then utilizing those inside the components modules.

I don't know much about how these controllers work yet, so I'm not sure if the components can be mixed. Is the intent that only one type of component may be used on a specific controller? If so, is this actually a hardware limitation? If not, having any component-specific logic inside the driver conditional seems wrong, as the initialization is now dependent on the order in which the components are created.

scottgonzalez, 2015-04-30:
It looks like `motor` is actually supporting [Adafruit Motor/Stepper/Servo Shield for Arduino v2](https://www.adafruit.com/product/1438), which seems to be built on the PCA9685 chip. I'm basing this on the [Arduino code](https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library/blob/b2ba608b22afe495c2ec6643eafbb7f9c81dc71d/Adafruit_MotorShield.h#L22).

dtex, 2015-04-30:
The Arduino code is good for a quick start, and our initialization process is very similar to their's, but to really understand what's happening with the PCA9685 this document is invaluable: http://www.nxp.com/documents/data_sheet/PCA9685.pdf

Also, I think it's helpful to not think about supporting specific boards or shields. We are supporting the PCA9685 chip which works the same regardless of who makes the shield or breakout board.

This all came up some time ago but didn't go far: https://github.com/rwaldron/johnny-five/pull/439#discussion_r17683431

dtex, 2015-04-30:
Not to say it was a bad idea, just that it was premature.

rwaldron, 2015-04-30:
> led uses i2cWrite(); and motor uses sendI2CWriteRequest()

Yeah, these just needed to be updated (I just pushed an update).

> I'd recommend creating controller modules that contain the common logic and then utilizing those inside the components modules.

See comment here: https://github.com/rwaldron/johnny-five/blob/310cdd11a4331ff1552a69a49d12826ea5e0b0f0/lib/servo.js#L40-L47

That was the plan that @dtex and I discussed some time ago, but we wanted to see what each controller implementation needed before we jumped the gun on an abstraction.

dtex, 2015-07-12:
So the plan now would be to add the PCA9685 to expander.js then update servo and led correct?

scottgonzalez, 2015-07-12:
What updates would ned to land in servo and led?

FWIW, I plan on digging into removing the duplication on my flight to India next week.

dtex, 2015-07-12:
Just to remove the duplication. The controllers in LED and Servo would leverage the new PCA9685 expander, right?

dtex, 2015-07-12:
Oh snap! Rick already added that to expander.js.

dtex, 2015-10-31:
All the classes that have a PCA9685 controller now leverage the PCA9685 Expander

closed 2015-10-31 by dtex