rwaldron/johnny-five #766: Possible memory leak with GP2Y0D805Z0F

frxnz, 2015-05-01:
Using [this device](http://osepp.com/products/sensors-arduino-compatible/osepp-ir-proximity-sensor-module/) over I2C, getting a memory leak warning:

```
(node) warning: possible EventEmitter memory leak detected. 11 I2C-reply-38-0 listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at addListener (events.js:179:15)
    at once (events.js:204:8)
    at Board.sendI2CReadRequest (/Users/notbook/git/johnny-five/node_modules/firmata/lib/firmata.js:921:8)
    at IR.<anonymous> (/Users/notbook/git/johnny-five/lib/ir.js:162:13)
    at wrapper [as _onTimeout] (timers.js:265:14)
    at Timer.listOnTimeout (timers.js:110:15)
```

Running this code:

```js
var five = require("../lib/johnny-five.js");

five.Board().on("ready", function() {

  var ir = new five.IR({
    device: "GP2Y0D805Z0F"
  });

  ir.on("data", function(err, timestamp) {
    console.log("data");
  });

});
```

It's entirely possible this is a wiring issue, since none of the pins are labelled. I've tried it a couple different ways with the same results.

[image]

dtex, 2015-05-01:
These line numbers don't seem to match up. What version of Johnny-Five?

rwaldron, 2015-05-01:
This bug is not likely in Johnny-Five. It pops up when there is something wrong with the wiring causing the component to fail.

frxnz, 2015-05-01:
@rwaldron Do you have one of these? Does the wiring above look right?

fivdi, 2015-05-01:
@frxnz is it because there are no pull-ups? SB1 & SB2 should be shorted according to the [document](http://osepp.com/wp-content/uploads/2012/01/OSEPP-PROX-01_REV1_1.pdf) linked to above.

frxnz, 2015-05-01:
>  is it because there are no pull-ups?

"Only short SB1 and SB2 if sensor is not connected to OSEPP I2C Shield". The pinout on the shield is just Vcc, GND, SDA and SCL.

fivdi, 2015-05-01:
Yes, "Only short SB1 and SB2 if sensor is not connected to OSEPP I2C Shield". In the photo above the proximity sensor is connected directly to the Arduino and not indirectly through an [OSEPP I2C Shield](http://osepp.com/products/shield-arduino-compatible/i%C2%B2c-expansion-shield/). I don't actually have one of the proximity sensors or an OSEPP I2C Shield, but the document leads me to believe that the proximity sensor board has 2 solder bridges called SB1 and SB2 that need to be closed to enable the 2K2 pull-up resistors on the proximity sensor board. Does the proximity sensor board have 2 solder bridges and are they closed? They are not visible in the photo, perhaps they're on the bottom of the board.

frxnz, 2015-05-01:
@fivdi Ah good catch. I assumed it was the I2C shield because I'm bad at reading. There are indeed 2 solder bridges on the back, and they're open. I'll close them tonight and give this another try. Thanks!

closed 2015-05-06 by rwaldron