rwaldron/johnny-five #767: Node-RED or Johnny-Five issue?

bbence84, 2015-05-01:
I cannot seem to start Node-Red anymore (it worked before) if I enable johnny-five in the context:

        jfive:require("johnny-five"),
        j5board:require("johnny-five").Board({ port: "/dev/ttyACM0", repl:false })

This gives the below error upon running node-red (node-red-pi --max-old-space-size=128):

1430517767179 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting
2 May 00:02:47 - [red] Uncaught Exception:
2 May 00:02:47 - Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (/usr/lib/node_modules/johnny-five/lib/board.js:388:26)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)

The connection works fine I guess (also from the node-red startup log):
1430517673680 Connected /dev/ttyACM0

What is curious is that this worked before, but I did an npm update on johnny-five, although I am not 100% sure is this has caused the above issue. But this is the only thin I can think of so far.

Also running johnny-five in a standalone node.js works fine:

pi@raspiot ~/node $ node led.js
1430517918380 Connected /dev/ttyACM0

var five = require("johnny-five");
var board = new five.Board({ port: "/dev/ttyACM0", repl:false });

board.on("ready", function() {

  var led = new five.Led(9);

  led.fadeIn();

  // Toggle the led after 5 seconds (shown in ms)
  this.wait(5000, function() {
    led.fadeOut();
  });
});

Any suggestions why it does not work in node-red?

Also, the npm module node-red-contrib-gpio works too, if I comment out the johnny five init lines. This npm module uses johnny-five.

Thanks!

rwaldron, 2015-05-01:
@monteslu any thoughts?

bbence84, 2015-05-02:
One clarification: the first two lines I mentioned (to init johnny five when starting node red) is taken from this tutorial:
http://nodered.org/docs/hardware/arduino.html

bbence84, 2015-05-02:
Ok, solved this, kinda. Reinstalled the whole Rasperry image, started from sratch. Not sure what was causing this though.

closed 2015-05-02 by bbence84

rwaldron, 2015-05-02:
That's bizarre! I'm glad this is resolved though. Please re-open if you encounter the issue again in the future.

Let's ping @bryan-m-hughes so this is at least on his radar in case we have to revisit.

bryan-m-hughes, 2015-05-03:
Interesting, I'd never heard of node-red before.