rwaldron/johnny-five #768: Board timeout and serialport error handling

scottgonzalez, 2015-05-03:
There's currently [logic to detect timeout errors between `connect` and `ready`](https://github.com/rwaldron/johnny-five/blob/c3af2c4131d8a20bb2e9a00987db9dc18dfe4f34/lib/board.js#L374-L389), but there's no timeout logic for a board that's not connected at all. Is that intentional? It seems like `new five.Board()` with no connected devices should probably result in a timeout as well.

There's also no error handling for `serialport.list()` (see https://github.com/rwaldron/johnny-five/blob/c3af2c4131d8a20bb2e9a00987db9dc18dfe4f34/lib/board.js#L74). This should probably result in an `error` event on the board instance.

I wanted to get some discussion around the various failure scenarios for board initialization before I started any work on this.

rwaldron, 2015-05-03:
> but there's no timeout logic for a board that's not connected at all. Is that intentional?

If there is no board physically attached, `new five.Board()` will wait for you to plug a board in: https://github.com/rwaldron/johnny-five/commit/fb85d50090023a66f968609bb5efcbabd4bedb14 

> There's also no error handling for serialport.list()

Yep, `serialport.list` will never produce an error there—it's probably there for the sake of callback parameter consistency, but it will never be anything but `null`. This has been [true forever](https://github.com/voodootikigod/node-serialport/commit/f44e7e66f0661f4482955c08e24f35c253bc1ddc#diff-4f97e03868bd8b1d8ea209b7a03863abR178). If you think there should be an error event emitted, then a patch will be accepted, but it's effectively a dead code path.

scottgonzalez, 2015-05-03:
> If there is no board physically attached, new five.Board() will wait for you to plug a board in: fb85d50

Ah, and only when there is no board connected do you see the "Looking for connected device" message. Sounds good.

> Yep, serialport.list will never produce an error there—it's probably there for the sake of callback parameter consistency, but it will never be anything but null. This has been true forever. If you [...]

I had spent some time trying to figure out if this was the case. When I saw [serialport's own code checking for the error](https://github.com/voodootikigod/node-serialport/blob/171377b55ad6d114b8f6e6e9874be4ef39f8f128/bin/serialportList.js#L6-L9), I thought it *might* throw errors, but also figured it might just have been written out of habit.

We should probably either just be careful and add the error event, or add a comment so that it's clear that we're intentionally ignoring the error. Do you have a preference?

rwaldron, 2015-05-04:
Comment is good. If it ever gets updated, we'll update to emit the error

closed 2015-05-04 by rwaldron

ericman314, 2015-11-05:
I have this code to initialize a `Board`:

```js
var five = require("johnny-five");
var board = new five.Board();
```

When I run this with nothing plugged in, I encounter this error:

```
/home/pi/hotcocoa/node_modules/johnny-five/lib/board.js:86
      ports = result.filter(function(val) {
                    ^

TypeError: Cannot read property 'filter' of undefined
    at Board.<anonymous> (/home/pi/hotcocoa/node_modules/johnny-five/lib/board.js:86:21)
    at /home/pi/hotcocoa/node_modules/serialport/serialport.js:565:11
    at FSReqWrap.oncomplete (fs.js:82:15)
```

I found this comment in `board.js`, `serialport.list() will never result in an error. On failure, an empty array is returned. (#768)`, which led me here.

I'm not exactly sure what is meant by this, but it seems to me like `result` is undefined rather than an empty array.

As for the cause of this error, the `err` object in the callback contained this information:

```js
{ [Error: ENOENT: no such file or directory, scandir '/dev/serial/by-id']
  errno: -2,
  code: 'ENOENT',
  syscall: 'scandir',
  path: '/dev/serial/by-id' }
```

So `serialport` couldn't return any ports because the directory `/dev/serial` does not exist when there are no connected ports. Mystery solved.

This was not too hard to figure out, but my real concern is that the error message `Cannot read property 'filter' of undefined` isn't immediately helpful in identifying the problem.

**Raspberry Pi version:** Model B Revision 1.0 + ECN0001

**Board:** None

**Version of Johnny-Five:** 0.9.6

**What my expectations are:** A helpful error message, like `No serial devices found`

**Actual outcome:** `TypeError: Cannot read property 'filter' of undefined`

**Steps to reproduce:** Run this code with nothing connected to the raspberry pi:
```js
var five = require("johnny-five");
var board = new five.Board();
```

rwaldron, 2015-11-06:
> I'm not exactly sure what is meant by this, but it seems to me like result is undefined rather than an empty array.

Looks like a breaking change from serialport. That sucks.

rwaldron, 2015-11-06:
According to the node-serialport source, there is still no way that this path will be taken, because the error is always set to an empty string: 

- https://github.com/voodootikigod/node-serialport/blob/master/src/serialport.cpp#L439-L445
- https://github.com/voodootikigod/node-serialport/blob/master/src/serialport.cpp#L456-L476

@beriberikix any experience with this issue?

jcontour, 2017-05-30:
Was there ever a resolution to achieving an error or timeout if no board is connected?

scottgonzalez, 2017-05-30:
If no board is ever connected, it will just wait forever. You'll shouldn't get an error or timeout in that case; the code just assumes you haven't plugged the board in yet. I believe timeouts occur when the board is connected, but the board isn't responding during initialization.

jcontour, 2017-05-30:
Thanks, is there an established way to check if the serial connection exists before instantializing a board? I'm trying to make sure that my code covers the a situation for when a board is not present.

commit bb5bd685 (2015-05-04): Board: Document that serialport.list() never errors. Fixes #768