rwaldron/johnny-five #770: LED.Matrix HT16K33 How to draw updating integers

deevolutionism, 2015-05-03:
I pulled an example from the Johnny-Five documentation on Adafruit's 8x8 LED matrix. I noticed that the functionality currently built in allows you to write static characters as well as create your own shapes using arrays. However, I can't figure out how to draw a variable int to the matrix. I have a project that involves voting and need to be able to pass an int into the draw function. Is there a way to do this? 

Thanks.

rwaldron, 2015-05-04:
Not at the moment, but it wouldn't be hard to add. I will work on this tomorrow and will keep you posted.

deevolutionism, 2015-05-04:
Thanks! That would be awesome! It's going to be part of a wireless binary voting system Im making thats sort of based off of Steve Lamberts "public forum" piece. I am also going to be implementing v1.0 Xbee modules to transmit the collected data to either my laptop or a raspi 2 where a lan server will be hosted.
[image]

rwaldron, 2015-05-04:
That's sharp!

I was looking at the code and saw that all characters are stored as strings, 0-9 are present, and a number will be coerced to a string anyway (I think this needs to be better documented anyway). 

Does the following work?

```js
matrix.draw(indexOfMatrix, 1); //  should display the number 1 on which ever matrix device is at indexOfMatrix
```

deevolutionism, 2015-05-04:
The above works fine, but the problem is that its a static number and needs to be a variable. 

My code stripped down to just the basic button and matrix below:

```javascript
var five = require("johnny-five"),
    bumper1, bumper2,
    board = new five.Board();

var votes1 = 0;
var votes2 = 0; 

board.on("ready", function() {
bumper1 = new five.Button(7);
bumper2 = new five.Button(6);

var matrix1 = new five.Led.Matrix({
    addresses: [0x70],
    controller: "HT16K33",
    rotation: 2,
  });

var matrix2 = new five.Led.Matrix({
    addresses: [0x71],
    controller: "HT16K33",
    rotation: 2,
  });

matrix1.clear();
matrix1.draw(1, votes1);

matrix2.clear();
matrix2.draw(2, votes2);

bumper1.on("hit", function() {
    votes1 += 1;  
    console.log('button 1 pressed'); 
  }); 

bumper2.on("hit", function() {
   votes2 += 1;
   console.log("button 2 pressed");
});
```
However, when i pass the votes1 variable into the draw function, it gives me an error.

rwaldron, 2015-05-04:
Ok, this helps me understand a bit better. Can you paste the error here?

rwaldron, 2015-05-04:
Just a quick note: I'm away from home this week and don't have this hardware with me, but I just ordered 2 of them from adafruit for overnight delivery. Hopefully by tomorrow I can be more help ;)

rwaldron, 2015-05-04:
I was able to get my hands on a shift register & matrix and I discovered a bug, maybe it's the bug you're hitting, maybe not? Hitting the button to count up past 9 will throw an exception, because there is no entry in the character table. [I fixed that](https://github.com/rwaldron/johnny-five/commit/1703a6ae53282a86df56b6819ead44e744150a5a#diff-9f8ec4adada6464c0d7658ef23115071R1095)

deevolutionism, 2015-05-04:
This was the error I got when I used 

```javascript
matrix.draw(votes1);
```

 /Users/deevolution/node_modules/johnny-five/lib/led/ledcontrol.js:262
        character = LedControl.MATRIX_CHARS[chr].slice(0);
TypeError: Cannot read property 'slice' of undefined
    at LedControl.draw (/Users/deevolution/node_modules/johnny-five/lib/led/ledcontrol.js:262:49)
    at Board.<anonymous> (/Users/deevolution/Documents/4th-Semester/Studio Environments/FinalProject/PublicVote/matrix.js:25:10)
    at Board.emit (events.js:104:17)
    at process._tickDomainCallback (node.js:381:11)

I will update johnny-five and see if your changes solves the problem.

rwaldron, 2015-05-04:
Yep, that was the error I just fixed.

rwaldron, 2015-05-04:
published to npm v0.8.70

deevolutionism, 2015-05-04:
AWESOME! that did the trick. Thank you! 

I think I was expecting similar functionality to Adafruit's [LED backpack library] (https://github.com/adafruit/Adafruit-LED-Backpack-Library) which allows you to easily pass variables into  their print() function. Maybe you've played around with their library before, but if not you should definitely check it out. if you could steal some of their functionality and integrate it with johnny-five, I think there is definitely some value there. 

This is from their text example:
```c++
  matrix.setTextSize(1); //set text sizes
  matrix.setTextWrap(false);  // we dont want text to wrap so it scrolls nicely
  matrix.setTextColor(LED_ON);
  for (int8_t x=0; x>=-92; x--) {
    matrix.clear();
    matrix.setCursor(x,0);
    matrix.print("I <3 johnny-five");
    matrix.writeDisplay();
    delay(100);
  }
```

Thanks again.

rwaldron, 2015-05-04:
It's definitely a goal to be able to animate the displayed output, modified font size, etc :)

Thanks again for the report and for your patience!

rwaldron, 2015-05-04:
It just occured to me that I still have 2 ht16k33's on their way via overnight delivery... 0_0

closed 2015-05-04 by rwaldron

deevolutionism, 2015-05-04:
One other (hopefully) quick problem! The matrix runs successfully on Arduino Uno when I2C clock is attached to analog 5 and 12C data is attached to analog 4. Running on the mega with I2C clock attached to digital pin 21 (SCL on Mega) and digital I2C data to digital pin 20 (SDA on Mega) nothing is being displayed on the matrix. I also attempted to hook up clock to analog 5 and data to analog 4 on the mega, but nothing happens. I am able to run one of Adafruits matrix example sketches successfully on the Mega with the same pinout.

rwaldron, 2015-05-04:
That's strange, Johnny-Five doesn't do anything differently for I2C on any of the Arduino boards, but of course I don't have my Mega here with me :| I will try to pick one up this evening from RadioShack. This might be a dumb question, but here goes: the Mega has the latest version of Firmata/StandardFirmata (2.4.3) flashed to it, right?

deevolutionism, 2015-05-04:
D'OH! :laughing: though i had it flashed, guess not. Thanks again, man!

rwaldron, 2015-05-04:
Is it all good now?

deevolutionism, 2015-05-04:
Works great!

[image]

Thanks!

rwaldron, 2015-05-05:
I love this! 

cc @annagerber

rwaldron, 2015-05-05:
@deevolutionism one important thing: the numbers stop at 99 (because I'm not sure if we can fit anything  bigger). Currently if you try to display a number that can't be displayed, it will throw an intentional error exception, but maybe maybe we can change that to a logged warning instead? What do you think?

deevolutionism, 2015-05-05:
I may opt to make those numbers percentages rather than single votes and just laser cut a little percentage symbol to put next to the matrix.

deevolutionism, 2015-05-05:
Alternatively I might utilize this limitation to my advantage. When the number of votes for one side have reached 99, it is the conditional for when the next set of items will be compared and voted upon. Don't worry about making any nit-picky changes. I can work within these constraints just fine. 

Thanks a bunch for your support! Keep up the good work on this library!