rwaldron/johnny-five #779: Missing sensor events on long running script

LegendaryTom, 2015-05-06:
Hi,

I'm running a script that measures the number of electric pulses that are output from a utility meter.  I have an Arduino connected to an electric meter to measure electric consumption in kWh.    For every 0.072 kWh our building consumes, the input to the board will toggle from high to low, or low to high.  This script uses Sensor.within() to find out how many pulses have occurred.  

```javascript
var five = require("johnny-five");
var board = new five.Board({port:"/dev/ttyATH0"});

var pulseCount = 0;

board.on("ready", function()
{
	var pulseSensor = new five.Sensor("A0");

	var alreadyLow = false; //used to keep track of when a new pulse is happening

	/* Input is low */
	pulseSensor.within([0, 100], function(unused, value)
	{
		var pulseTime = new Date();

		if(alreadyLow)
		{
			console.log("[LOW ][%s][Voltage: %s]", pulseTime, value);
			return;
		}

		pulseCount++;  //input switched from high to low
		console.log("[LOW ][%s][Voltage: %s][pulse#: %s]", pulseTime, value, pulseCount);

		alreadyLow = true;
	});

	/* Input is high */
	pulseSensor.within([900, 1023], function(unused, value)
	{
		var pulseTime = new Date();

		if(!alreadyLow)
		{
			console.log("[HIGH][%s][Voltage: %s]", pulseTime, value);
			return;
		}

		pulseCount++;  //input switched from low to high
		console.log("[HIGH][%s][Voltage: %s][pulse#: %s]", pulseTime, value, pulseCount);

		alreadyLow = false;
	});

});

process.on('uncaughtException', function(err) {
	console.log([new Date()], 'Caught exception: ' + err);
	process.exit(1);
});

```

This logs the time and voltage on input pin "A0" about 30-40x per second to the console.  The output looks like this (I shortened it some):

```
root@Arduino:~/evcc# node pulseCounter2.js
1430922367558 Connected /dev/ttyATH0 
1430922372972 Repl Initialized 
>> [HIGH][Wed May 06 2015 14:26:13 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:13 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:13 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:13 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:13 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:13 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:13 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 1023]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0][pulse#: 1]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:14 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:15 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 14:26:15 GMT+0000 (UTC)][Voltage: 0]
[HIGH][Wed May 06 2015 14:26:15 GMT+0000 (UTC)][Voltage: 950][pulse#: 2]
[HIGH][Wed May 06 2015 14:26:15 GMT+0000 (UTC)][Voltage: 991]
[HIGH][Wed May 06 2015 14:26:15 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 14:26:15 GMT+0000 (UTC)][Voltage: 1023]
```

I've been running this script on the board 24/7 for a few weeks now and am I'm running into an issue that occurs only after the script has been running for longer than 12 hours.  What happens is that the the within() event does not seem to be firing for X number of seconds.  

```
[LOW ][Wed May 06 2015 10:57:23 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 10:57:23 GMT+0000 (UTC)][Voltage: 0]
[HIGH][Wed May 06 2015 10:57:23 GMT+0000 (UTC)][Voltage: 1017][pulse#: 78628]
[HIGH][Wed May 06 2015 10:57:23 GMT+0000 (UTC)][Voltage: 976]
[HIGH][Wed May 06 2015 10:57:23 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:24 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:24 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:24 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:24 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:24 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:24 GMT+0000 (UTC)][Voltage: 1023]
[LOW ][Wed May 06 2015 10:57:32 GMT+0000 (UTC)][Voltage: 0][pulse#: 78629]
[LOW ][Wed May 06 2015 10:57:33 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 10:57:33 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 10:57:33 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 10:57:33 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 10:57:33 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Wed May 06 2015 10:57:33 GMT+0000 (UTC)][Voltage: 0]
[HIGH][Wed May 06 2015 10:57:33 GMT+0000 (UTC)][Voltage: 1010][pulse#: 78630]
[HIGH][Wed May 06 2015 10:57:34 GMT+0000 (UTC)][Voltage: 1022]
[HIGH][Wed May 06 2015 10:57:34 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:34 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Wed May 06 2015 10:57:34 GMT+0000 (UTC)][Voltage: 1023]
```

In this log, there are no events for about 8 seconds between 10:57:24 and 10:57:32.  If I stop the script and restart it, it will work fine for 10-12 hours, then start missing pulses again.  My current workaround is to just restart the script every day.  But, it irks me.  Any ideas about what is happening or how I can troubleshoot this?  Or am I just doing it wrong, and should be using hardware interrupts to measure the pulses? 

The setup is an Arduino Yun, with node.js running directly on that board.  Latest Yun image is installed, running off 16G microSD card, with 100MB swap file enabled.  Node is the only additional application that I'm running on the Linux side of the board.  Running johnny-five@0.8.46.  I've got a 2nd Yun connected to the electric meter which is reading that pulses are coming in during that time.

rwaldron, 2015-05-06:
This is really interesting and not something that I've encountered myself yet—but we'll work through it together :)

First thing: my relationship with the Yun has not been a positive one. I've bricked (100% non-responsive) 2 of the 3 that I've purchased, so I had long since given up trying—looks like I need to come back around on that. Can you share the setup steps used to prepare your Yun to run node.js & Johnny-Five, that will help get me up to speed and in turn to be more useful to you :)

mMerlin, 2015-05-06:
One idea would be that some of the timing in firmata relies on a timer/counter that is wrapping around.  When the count reaches maximum value, and wraps to zero, some comparisons start failing.  I am not aware of anything that should be kicking in at 10 plus hours.  I have hit similar when working (directly) on an Arduino when using the micros() values.  That wraps at just over an hour of run time.

mMerlin, 2015-05-07:
Check the version of firmata you are using.  I noticed in the firmata/arduino repository https://github.com/firmata/arduino that a recent (Apr 11, 2015) commit 
https://github.com/firmata/arduino/commit/e2cede35460333ed97c27f271dadbf7349b11c0e
brought StandardFirmataYun up to date with StandardFirmata

LegendaryTom, 2015-05-07:
The Yun is the only board I've worked with, but it does seem to have a lot of "quirks" to work around.  Here are my notes that I use for setting up new Yun's: http://htmlpreview.github.io/?https://gist.github.com/LegendaryTom/d0bf1a384b10e3f8547b/raw/

Fair warning, the very first step voids your warranty and has the potential to brick your Yun.

Here's what's in the notes:
1: Flashing UBOOT on the Yun
2: Configure Network
3: Change Boot Script
4: Expand Yun Disk Space to microSD card
5: Add swap space
6: Installing node.js
7: Installing johnny-five on seperate box and copy to Yun
8: Enable serial connection for firmata
9: Upload Firmata sketch

Thanks for looking into this.

LegendaryTom, 2015-05-07:
@mMerlin I will try out that version of firmata and see how it goes.

rwaldron, 2015-05-07:
@LegendaryTom Thanks for the run down. I will try this as soon as I have a moment available. 

Sort of related: One thing to note: the Nino-IO plugin is designed to work on an Arduino Yun that's running LininoOS. This would change step 7, 8, 9 to: 

7: `npm install johnny-five nino-io`

LegendaryTom, 2015-05-11:
@mMerlin I tried the newer version of standard firmata over the weekend and it seems to still be experiencing the same issue:
```
[LOW ][Mon May 11 2015 20:07:26 GMT+0000 (UTC)][Voltage: 0]
[LOW ][Mon May 11 2015 20:07:26 GMT+0000 (UTC)][Voltage: 0]
[HIGH][Mon May 11 2015 20:07:26 GMT+0000 (UTC)][Voltage: 974][pulse#: 128894]
[HIGH][Mon May 11 2015 20:07:26 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Mon May 11 2015 20:07:26 GMT+0000 (UTC)][Voltage: 1022]
[HIGH][Mon May 11 2015 20:07:55 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Mon May 11 2015 20:07:57 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Mon May 11 2015 20:07:57 GMT+0000 (UTC)][Voltage: 1023]
[HIGH][Mon May 11 2015 20:07:57 GMT+0000 (UTC)][Voltage: 1023]
```

@rwaldon I was aware that there are j5 plugins for many different boards.  But somehow I passed over the fact that there is one for the Yun.  I will definitely try this out.

rwaldron, 2015-05-11:
> the fact that there is one for the Yun. I will definitely try this out.

Just make sure that you're running the right OS (LininoOS). I'm almost scared to recommend it because of the frustration I experienced. Maybe we should just press on with your current set up. 

For now, let's strip away the `Sensor` class: 

```js
var five = require("johnny-five");
var board = new five.Board({port:"/dev/ttyATH0"});

board.on("ready", function() {
  this.io.pinMode(0, this.io.MODES.ANALOG);
  this.io.analogRead(0, function(adc) {
    console.log(Date.now(), adc);
  });
});
```

LegendaryTom, 2015-05-14:
I ran that code from Tue, 12 May 2015 22:45:00 to Wed, 13 May 2015 21:00.  I kept the terminal window open on my desktop and it received just over 4 million lines of output.  So I just did some skimming of the output until I found a place with a gap.

```
1431550691838 1023
1431550691856 1023
1431550691876 1023
1431550691895 1023
1431550691916 1023
1431550691933 1023
1431550694265 1023
1431550694277 1023
1431550694281 1023
1431550694285 1023
1431550694290 1023
1431550694294 1023
1431550694298 1023
1431550694303 1023
```
Throughout most of the log, the difference in time between each line of output was about 20ms.  However, for the 750ms period following the gap, the difference in time between each line of output was about 5ms.

mMerlin, 2015-05-14:
What environment are you running node.js on?  That symptom and description indicate (to me) that something outside of johnny-five and node.js 'took over' the system for a short time, then when it released, the queued firmata analogRead did some catchup.

750 ms / 5 ms is 150 entries.  That gap is 2332 ms.  Plus the 750 ms 'speedup' time is 3082 ms.  At 20ms separation, that would be 154 entries.  That is *VERY* close given all the approximations involved.

I suggest there is some automatic process outside of node that is blocking things for a couple of seconds periodically.  Something in the OS, or another running background program that 'kicks in' for a couple of seconds.

LegendaryTom, 2015-05-15:
@mMerlin That's a good observation.  Based on that count, it looks like no callbacks were missed by ```analogRead```.  They were just delayed while the system was busy.  

Remember in my first post, I'm trying to count the number of transitions from low->high and high->low that occur.  Sometimes after a gap the code counts many pulses.  So the total number of pulses counted basically stays the same.  This happened in the above 750ms.

When is the value ```adc``` captured for the callback ```function(adc)```?  Is there anything asynchronous going on between when the ```adc``` value is first received from firmata and when I'm logging the time using ```console.log(Date.now(), adc);```?  

Here's a graph of what my collected data looks like:
[image]

At the start of the day, there were pulses being missed (not just delayed).  And then there's a dip in the blue line which is when I rebooted the pulseCounter.js application.  After rebooting, the data matches almost exactly.

> What environment are you running node.js on?

Node.js is running directly on the Arduino Yun board.  It's running the OpenWRT distribution of Linux, and my understanding of the Yun is that the OS/hardware on the Linux side of the board is similar to many home WiFi routers.

mMerlin, 2015-05-15:
> When is the value adc captured for the callback function(adc)?

Now you are outside where I can make assumptions.  My high level understanding of the process, is that a Firmata program is run on the Arduino side.  On the linux / node.js / johnny-five side, firmata is also being run.  Commands and responses are being sent back and forth 'as needed'.  Specifically for analog sensors, that involves a continuous stream of data reads.  I suspect, but do now know, that the Arduino board has been told to do continuous reads, and just keep sending that data.  If that is the case, and no packets were dropped during the 'busy time', then (johnny-five) analog reads are just consuming the values as fast as it can, as long as there is new data.  That would mean that after the gap, the reported Date.now() is well after when the value was actually captured.  Getting closer as node.js catches up to the Arduino again over the 750 ms.

That is what I **think** happened.  It matches the symptoms, and what I expect is the best way to get streaming, realtime data out of the Arduino.  I do not know the Firmata protocol, or the commands johnny-five is using though.  Another scenario that fits, is that johnny-five has a low level process that keeps asking the arduino for new analog values.  That is triggering are 'read request' at a constant rate.  During the busy time, those request could have been backup and not sent.  When things started again, the delayed requests were sent to the Arduino at a faster rate, so the readings themselves were delayed.

Have to dig into the firmata and io code to be sure.  A quick browse, starting at node_modules/johnny-five/node_modules/galileo-io/lib/galileo.js finds:
Galileo.prototype.analogRead
calling read, which is using a samplingInterval.

That would seem to imply that the second scenario is more accurate.

reconbot, 2016-05-25:
@mMerlin is spot on. And depending on what else you have going on, on the board, it could be tracked down to a large collection of things. I'm going to close this ticket due to age, but reopen it if you would like to debug this further.

closed 2016-05-25 by reconbot