rwaldron/johnny-five #780: What determines the board that is associated with (internal to) a component?

mMerlin, 2015-05-06:
I created a package to create components based on a hardware 'model' object.  The package exports a function that takes the description objects as parameter, and returns an object containing the created components.

The package requires 'johnny-five', but it does NOT create an instance of a board.  Given my understanding of javascript scoping rules, I was expecting to need to pass the board instance from the the calling code, to be able to associate the created components with the hardware board.  I am finding that is not the case.  The components created in my package have a .board property that matches (===) the Board instance created in the calling code.

While that is convenient, and makes my current package code easier, I would like to understand where my package is getting the board from when it creates the components.  The package should (is intended) be able to handle multiple connected boards, and create appropriate components for each of them.  How do I get the package code to use the 'right' board for the components?

Here is cut down working sample code.

_testCall.js_
```javascript
'use strict';
var five = require('johnny-five');
var j5Load = require('./testPackage.js');
var model = {
  blinker: {
    class: "Led",
    options: { pin: 13 }
  }
};
new five.Board().on('ready', function () {
  this.children = j5Load(model);
  console.log(this === this.children.blinker.board);
  this.children.blinker.on();
});
```
_testPackage.js_
```javascript
'use strict';
var five = require('johnny-five');
function loadComponentsForModel(model) {
  var container = {},
    key = 'blinker';
  container[key] = new five[model[key].class](model[key].options);
  return container;
}
module.exports = loadComponentsForModel;
```
Where is the board used for the component created in testPackage coming from?  How can I override that, to specify a specific board?

The full, under development, package can be seen at https://github.com/mMerlin/johnny-load in lib/johnny-load.js.  It is a fuller implementation than might be expected, because I did not discover the board being defaulted, until I added some (expected to be) failing unit tests.

rwaldron, 2015-05-07:
Take a look at the component plugin template here: https://github.com/rwaldron/johnny-five/blob/master/docs/plugin.md

rwaldron, 2015-05-07:
Sorry, hit "Comment" button too soon!

- That link above is for defining third party component classes. 
- To instruct a component to use a specific board, all classes accept a `board` property in their constructor options, eg

```js
var led = new five.Led({ pin: 13, board: some board instance });
```

mMerlin, 2015-05-07:
I saw a sample of the board property someplace, but it did not look like it was being used.  I tried to force an error by setting it to null.  I guess the low level code that uses that is filtering it out, if the board property is 'falsey', which would ignore null, undefined, 0, and empty strings.

So for my case, I can just supply a board parameter on the call.  If it is supplied, it gets used.  If not, undefined will get ignored, so I do not even need to use conditional logic to add the board property to the options object.

Still 'curious' how the board instance created in testCall.js made into the component created in testPackage.js.  I would have expected the scopes to be very separate.  It must be stored in global scope someplace.

Resseguie, 2015-05-07:
@mMerlin I think the magic you're looking for is here:
https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L690

Components call Board.Component which in turn calls Board.mount. If the board property is set, it uses it. If not, it tries to guess from all currently defined boards.

Resseguie, 2015-05-07:
Example of where Led calls Board.Component:
https://github.com/rwaldron/johnny-five/blob/master/lib/led/led.js#L169-L171

mMerlin, 2015-05-07:
That 'guessing' is not very smart.  It just picks whatever the first board in the array is.  Of course, I can not think of anything to use for reasonable criteria.

That does explain where the defaulting is coming from though.  At least after I update my understanding of node.js packages.  From exploring, a module that is 'required' in more than one place is really only loaded once, so the the scope for that module is accessed from each reference.  I was thinking 'include' functionality, instead of 'load'.

I saw the Board.Component calls, but did not trace them deep enough to find the mount and array.

closed 2015-05-07 by mMerlin

rwaldron, 2015-05-08:
> not very smart

It's worked quite well for over 3 years without producing a single known or reported complaint. The same code is run for every IO-Plugin of every platform that's supported and has never produced a bug (that's been reported). If a program needs to be explicit, it can be, but otherwise I see nothing "smart" about burdening the code with ceremonial boilerplate.

mMerlin, 2015-05-08:
Like I said, I can not think of any available criteria that could be used to make it 'smarter'.  Simple is good too.  The 'not smart' was more directed to the 'guess' reference in the earlier comment.