rwaldron/johnny-five #782: RGB LED does not fully turn off with PCA9685

scottgonzalez, 2015-05-07:
I tried using tow different types of common anode RGB LEDs and neither one fully turned off, they both were very dimly lit. I don't have any common cathode RGB LEDs to test with.

rwaldron, 2015-05-07:
Using your example, I see complete on and complete off. Is this the wrong example to repro? Let me know!

scottgonzalez, 2015-05-08:
Yeah, I have it wired as shown in the new example. I ran `node eg/led-rgb-anode-PCA9685.js`, then from the REPL I ran `led.stop().off()` and this is what I end up with:

[image]

rwaldron, 2015-05-08:
I've tried several different RGB LEDs (diffused & clear) and I can't reproduce. I'll pick it back up tomorrow morning and see if I've missed something

scottgonzalez, 2015-05-24:
This is due to the controller using a special value for fully on and fully off. See https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/8846078c6198314aa610213bcd0ce8a0a05a7bca/Adafruit_PWMServoDriver.cpp#L105-L115

I've verified that using these values works for me. I just ordered some common cathode RGB LEDs to test with, so I'll send a patch as soon as those come (should be here on Tuesday).

closed 2015-06-02 by rwaldron

ryangonzalez133, 2024-09-02:
Have I found any way to actually solve this?

PR #816 (2015-06-02): Led.RGB: Handle special value for fully off common anode LEDs
Fixes #782 

I tested with a common cathode RGB LED as well, but didn't see the problem. The C code does have a special value for both though, so let me know if you'd like me to add the other case as well.
commit c851c141 (2015-06-02): Led.RGB: Handle special value for fully off common anode LEDs

Fixes #782