rwaldron/johnny-five #786: invalid package dependencies reported by npm list

mMerlin, 2015-05-08:
Is it just my environment, or is this a general issue getting npm to install the serialport package as part of a johnny-five dependency?

While trying to discover the differences in our environments for #750, I used
```bash
$ npm list
```
to show what all was installed for the test project folder.  That gives a lot of information, but ended up with:
```
npm ERR! invalid: node-pre-gyp@0.5.19 /home/phil/development/projects/serialtest/node_modules/serialport/node_modules/node-pre-gyp
npm ERR! invalid: readable-stream@1.0.27-1 /home/phil/development/projects/serialtest/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream
npm ERR! invalid: string_decoder@0.10.25-1 /home/phil/development/projects/serialtest/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream/node_modules/string_decoder
```

Doing
```bash
$ npm install johnny-five
```
Does not report any errors, nor does using
```bash
$ npm install
```
with johnny-five listed as a dependency in package.json.  However npm list reports invalid package versions when listing afterward.  To get a short list, use
```bash
$ npm list string_decoder
```
In my test environment, that gives
```
$ npm list string_decoder
water-load@0.0.1 /home/phil/development/projects/water-load
└─┬ johnny-five@0.8.71
  └─┬ serialport@1.6.3
    └─┬ node-pre-gyp@0.5.19 invalid
      └─┬ tar-pack@2.0.0
        └─┬ readable-stream@1.0.27-1 invalid
          └── string_decoder@0.10.25-1  invalid

npm ERR! invalid: node-pre-gyp@0.5.19 /home/phil/development/projects/water-load/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp
npm ERR! invalid: readable-stream@1.0.27-1 /home/phil/development/projects/water-load/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream
npm ERR! invalid: string_decoder@0.10.25-1 /home/phil/development/projects/water-load/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar-pack/node_modules/readable-stream/node_modules/string_decoder
```

The problem seems to be triggered by the
```
"bundleDependencies": [
  "node-pre-gyp"
],
```
entry in the the serialport package.json configuration.  The actual serialport git repository does not actually include any bundled package.  Npm tries to recover from that, but ends up with v0.5.19, while another entry in the same package.json file has a (normal) dependency of
```
"node-pre-gyp": "0.6.x"
```

If that is a general issue, not just something strange in my setup, I have some solutions.  Referencing the serialport git repository directly for the dependency convinces npm to properly locate and install node-pre-gyp@0.6.7 in the project, which gets rid of the error messages in npm list.  That could be done by adding:
```
"serialport": "voodootikigod/node-serialport"
```
as a dependency in a local project, or by changing
```
"serialport": "latest"
```
to
```
"serialport": "voodootikigod/node-serialport"
```
in the johnny-five package.json.

TO test that, I cloned a fresh copy of the johnny-five repository.  Another test confirmed that
```
$ npm install
```
with a johnny-five dependency pointing at the local clone still failed with the same errors.  Changing the single entry in the cloned package.json repository got a repeat of the previous test to work correctly.

Manually doing:
```bash
$ npm install serialport
```
did not work.  That by itself got the same error message when listing.  However,
```bash
$ npm install voodootikigod/node-serialport
```
and
```bash
$ npm install voodootikigod/node-serialport#v1.6.1
```
to pick a specific version **did** work.  Doing that manually before installing johnny-five, either manually, or as a dependency in package.json file, got rid of the npm list error messages for me.

It **appears** the real problem is something to do with the published npm registry.  Bypassing that resolves the problem.

rwaldron, 2015-05-08:
Maybe there's a permissions issue? I've never had/seen this issue :(

mMerlin, 2015-05-09:
Maybe.  Doesn't really look like it though:
```
$ sudo -i
# cd /home/phil/development/projects/
# mkdir test
# cd test
# npm install serialport
\
> serialport@1.6.3 install /home/phil/development/projects/test/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/home/phil/development/projects/test/node_modules/serialport/build/serialport/v1.6.3/Release/node-v11-linux-x64/serialport.node" is installed via remote
serialport@1.6.3 node_modules/serialport
├── bindings@1.2.1
├── sf@0.1.7
├── async@0.9.0
├── nan@1.7.0
├── debug@2.1.3 (ms@0.7.0)
└── optimist@0.6.1 (wordwrap@0.0.3, minimist@0.0.10)
# npm list node-pre-gyp
/home/phil/development/projects/test
└─┬ serialport@1.6.3
  └── node-pre-gyp@0.5.19  invalid

npm ERR! invalid: node-pre-gyp@0.5.19 /home/phil/development/projects/test/node_modules/serialport/node_modules/node-pre-gyp
```

rwaldron, 2015-05-09:
@voodootikigod I need help with this

mMerlin, 2015-05-09:
I just dropped https://github.com/voodootikigod/node-serialport/issues/523 for this in the node-serialport issues, linked back here.

If you have not seen this, what version of node-pre-gyp is shown with npm list node-pre-gyp after npm install johnny-five?  What is your base environment?  I am using a laptop with Fedora 21 64 bit.  Recent rebuild, so there is not a lot of 'extras' on there yet.

voodootikigod, 2015-05-09:
I would suggest closing this here - it is 1) not a j5 issue and 2) not even a serialport issue. Its npm not abiding by the package specification.

rwaldron, 2015-05-09:
Closing. Thanks everyone

closed 2015-05-09 by rwaldron

mMerlin, 2015-05-11:
Assuming this was caused by npm, it appears that the latest update for the registry fixed the problem.  A simple 
```bash
npm install serialport
# or
npm install johnny-five
```
now (properly) install/use node-pre-gyp@0.6.7
Comparing the before and after (npm package registry) generated node_modules/serialport/package.json files, the used version of npm changed from 2.5.1 to 2.8.3.  The version in my environment was 2.7.6, which (after discovering above) I just updated to 2.9.1