rwaldron/johnny-five #787: this.io.pulsein undefined when using proximity

Darkein, 2015-05-09:
Hi,
i'm a new j5 user, and i really love this library.

unfortunately i encounter an error when i'm tring to use the proximity api.
I'm on windows 8, using node11 x86. I'm triing to play with the hc-sr04 sensor on an arduino mega with custom firmata.

>> [...]\node_modules\johnny-five\lib\proximity.js:175

          this.io.pulseIn(settings, function(microseconds) {
                  ^
TypeError: undefined is not a function
    at Proximity.<anonymous> (C:\Users\Pierre\Desktop\nodebot\node_modules\johnny-five\lib\proximity.js:175:19)
    at wrapper [as _onTimeout] (timers.js:265:14)
    at Timer.listOnTimeout (timers.js:110:15)

Thanks for your help!

frxnz, 2015-05-09:
It looks like you may have mismatched versions of Johnny-Five and firmata. `pulseIn` was recently changed to `pingRead` in https://github.com/jgautier/firmata/commit/bfc1e1ea1286e2e0f4025f67c87ce42aeb344dbf

Updating Johnny-Five should resolve the issue, `Proximity` was updated in v0.8.74. Do you know what version you're running?

rwaldron, 2015-05-09:
Sorry about that, I had accidentally published Firmata with the pulseIn=>pingRead changes. 

This is fixed with: 

- https://github.com/rwaldron/johnny-five/commit/c5de5cdacb6e3b0b83c9efbbe3b9113e056a5204
- https://github.com/jgautier/firmata/commit/bfc1e1ea1286e2e0f4025f67c87ce42aeb344dbf

- johnny-five@0.8.74
- firmata@0.5.0

All you need to do is `npm uninstall johnny-five && npm install johnny-five`, then make sure you have the latest PingFirmata from here: http://johnny-five.io/api/proximity/#pingfirmata

rwaldron, 2015-05-09:
@frxnz naw, it was my fault. I had accidentally `npm publish`ed Firmata.js :|

frxnz, 2015-05-09:
@rwaldron Aha, just noticed 0.8.74 came after this issue. I thought `pingRead` had landed in J5 yesterday. Carry on.

rwaldron, 2015-05-09:
You were totally right, there was a mismatch—but it was my fault :D

Darkein, 2015-05-10:
Thank you for correcting the problem so fast! I'll try again this afternoon.

rwaldron, 2015-05-10:
Awesome, and sorry about the bug. Once you confirm that this works correctly please close the issue. Thanks again!

Darkein, 2015-05-10:
Works for me, thanks!

closed 2015-05-10 by Darkein