rwaldron/johnny-five #791: Unable to control ESCs on Intel Edison

Abv204, 2015-05-14:
Hi, I am working on a project using Intel Edison and johnny-five to control a tricopter. The thing is the board is connected and initialized and then I get a type error something like this
 TypeError: Object #<Galileo> has no method 'servoConfig'
    at ESC.Controllers.DEFAULT.initialize.value (/home/edison/node_modules/johnny-five/lib/esc.js:72:17)
    at new ESC (/home/edison/node_modules/johnny-five/lib/esc.js:242:8)
    at Board.<anonymous> (/home/edison/Project codes/Flight.js:7:13)
    at Board.emit (events.js:92:17)
    at process._tickCallback (node.js:448:13)
    at Function.Module.runMain (module.js:499:11)
    at startup (node.js:119:16)
    at node.js:935:3
Could anyone help with this?
I am using ubilinux on my Intel Edison.

rwaldron, 2015-05-14:
Thanks for the report, I'll get this fixed immediately

rwaldron, 2015-05-14:
Please update Galileo-IO (or Edison-IO) to v0.8.14 and confirm that the issue is resolved—thanks!

Abv204, 2015-05-14:
Failed at the galileo-io@0.8.14 postinstall script 'node scripts/postinstall'.
npm ERR! This is most likely a problem with the galileo-io package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/postinstall

This is the error I get when trying to update.

rwaldron, 2015-05-14:
I'm not sure why it would error like that, especially if you'd previously installed it with no issues. I'll take a closer look and report back what I find.

rwaldron, 2015-05-14:
Please run: 

```
opkg info libmraa0
```

And post the output—thanks!

rwaldron, 2015-05-14:
I think I know what this is. If you've flashed your board with the 2.1 image, Intel broke the native bindings (missing some header file). You'll need to flash with 2.0 and reinstall johnny-five and galileo-io. I'm sorry that's such a pain in the ass—if I worked at Intel, I could do something about it, but I don't, so I can't. Here's a link to the page that has 2.0 download: https://software.intel.com/en-us/iot/hardware/edison/downloads

danyocom, 2015-05-15:
What header files are missing?

Abv204, 2015-05-15:
I'll switch back to the old image and let you know if it works. 
Thank you.

Abv204, 2015-05-17:
I re-flashed the edison to the 2.0 image and re-installed johnny-five and galileo-io. The modules work fine but I get an error saying, 
illegal arguments for construction of _exports_PWM

Abv204, 2015-05-17:
I have tried changing the PWM swizzers on the board and still nothing and also apparently the board has only four PWM pins despite there being six printed on it.

rwaldron, 2015-05-17:
> I have tried changing the PWM swizzers on the board and still nothing and also apparently the board has only four PWM pins despite there being six printed on it.

That's correct. The Edison can only handle 4 PWMs at a time, so the native bindings have hard-coded a set of allowed PWM pins internally. There is nothing this project can do to change that. This means programs may only use:

- [3](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c#L1164-L1165)
- [5](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c#L1179-L1180)
- [6](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c#L1188-L1189)
- [9](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c#L1209-L1210)

Abv204, 2015-05-18:
Is there a way to run the esc-keypress code without initailizing Repl, so that the motors switch on once the program runs?

michael-joseph-payne, 2015-05-18:
Hi rwaldron - there are several Intel folks who have seen this thread and would like to lock down the problem.  Can you describe what header files are impacting this use case?

rwaldron, 2015-05-18:
@Abv204 

> Is there a way to run the esc-keypress code without initailizing Repl, so that the motors switch on once the program runs?

The "esc-keypress" code is just a fun demo ;) It sounds like you want something simpler? Try this: 

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var a = new five.ESC(3);
  var b = new five.ESC(5);
  var c = new five.ESC(6);
  var d = new five.ESC(9);
  
  a.speed(25);  
  b.speed(25);
  c.speed(25);
  d.speed(25);
});
```

Or more simply...

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  var escs = new five.ESC.Array([3, 5, 6, 9]);
  escs.speed(25);
});
```

And when you're ready to run the program silently, add `repl: false, debug: false`: 

```diff
  var five = require("johnny-five");
  var Edison = require("galileo-io");
  var board = new five.Board({
    io: new Edison(), 
+    repl: false,
+    debug: false
  });

  board.on("ready", function() {
    var escs = new five.ESC.Array([3, 5, 6, 9]);
    escs.speed(25);
  });
```

--------

@michael-joseph-payne

> Hi rwaldron - there are several Intel folks who have seen this thread and would like to lock down the problem. Can you describe what header files are impacting this use case?

I only know what I learned at the NYC Intel IoT Roadshow event. I was a judge at the event and was I was told that any teams that wanted to use the mraa JavaScript or Python bindings had to roll back to the 2.0 image because `mraa` would not build on the 2.1 image. Sorry, but I don't have a spare Edison that I can trash just to get the file name.

cc @ashishdatta @IoTman

michael-joseph-payne, 2015-05-18:
@rwaldron - no problem, that's enough to get us started checking a few things

rwaldron, 2015-05-18:
@michael-joseph-payne thanks for the attention to this :)

Abv204, 2015-05-19:
I wanted to implement the keypress module into the project, I got the escs working with the addition to the code you mentioned.
Thanks for all the help :-).

closed 2015-05-19 by Abv204

rwaldron, 2015-05-20:
That's great! I look forward to seeing your work when you're ready :)