rwaldron/johnny-five #793: validating analog pin number

mMerlin, 2015-05-14:
I am writing some unit tests for a johnny-five project.  The project includes a user modifiable configuration file with details like which pin a sensor is connected to.  I want the unit test to sanity check the properties exported by the configuration.

Is there an simple way to check that an entry is a valid analog pin number, in a board agnostic way?

Looking at some of the existing test code, I can see using MockFirmata to create a dummy board, then use that to attempt to create a Sensor instance, and check the result.  Is there a simpler way?  Is there a way to check that is not dependant on a specific board?   When the unit tests are being run, the controller board is not known.

Looking at some johnny-five library code, it appears that checking for a string with first character 'A' or 'I', followed by a number would be close.  What would be the maximum analog pin number for supported boards?

rwaldron, 2015-05-14:
`MockFirmata` is "board agnostic" as any IO Plugin, but all IO Plugins must implement instance properties called `pins` and `analogPins`: https://github.com/rwaldron/io-plugins#minimum-plugin-class-requirements 

I recommend using https://github.com/achingbrain/board-io to create testable "board" objects.

mMerlin, 2015-05-15:
That looks a bit 'involved' to get working.  I've added a link reference to this thread in the unit test comments.  For now (until I run into a valid case that fails the test, or invalid case that pass), I'm just going to use a regex test:
```
/^[AI]\d$/.test(value)
```
That should handle up to 10 analog pins.  Since the physical board is not know when running the test, even the full suggested implementation would have problems deciding what the limits should be.

closed 2015-05-15 by mMerlin

rwaldron, 2015-05-15:
> That looks a bit 'involved' to get working. 

It's really not, I promise :/

I'm not really sure what I can do, because I don't really understand the problem case that you're trying to solve—and trust me, I want to help :)

The physical board will report its digital and analog pins when Johnny-Five connects to it, so we mock result of that capabilities report with the pin array stubs.

mMerlin, 2015-05-15:
I guess I left some 'detail's of the context out.  That's what happens when you already know enough about the environment to be dangerous, but not enough to know what might be important, while looking for details in a specific area.

The project is designed to be generic / reusable, so no particular environment assumptions are *safe*.  The data that drives some of the test cases comes from a configuration file that the end-user WOULD modify, to adjust things like the specific pin numbers things are attached to.  Even for the initial round, the code will be running on totally different computers (both the johnny-five project, and the Arduino board) than the current development and testing is being done on.  There might not even be a board connected at the time that grunt is run to do the tests.  The test cases have to either be explicitly configured, or assume **everything** about what the actual board and run time environment would look like.  

I looked at some of what the mock was doing when adding test cases for the Pin class.  From what I could see, it simulated approximately the capabilities of an UNO class Arduino board, based on the available pins.  Going forward, an enhancement would be to (on mock creation) tell it what specific board, or range of pins to create.  IE: provide the mock with an options structure what would set the capabilities that would be reported back.

Currently more important to me, is figuring out a way to create test cases that need to respond to and generate events.  Where functions are publicly visible, events can be simulated by calling the handlers directly with appropriate binding and arguments.  The package code I am working on (writing) though uses initializer methods that add listeners referencing package private event handlers.  I am not seeing a way to get hold of the entry points.  I should be able to directly emit appropriate test events, but then I need to catch the results.  Which are not the return value of the function.  So far, I have not found samples close enough to what is needed to wrap my head around a structure that works with the testing framework.

From looking over the johnny-five test code, it looks like the sinon package will be useful, but that is another that I do now know yet.

I need to finish the current round of refactoring, to get close to runnable code again, then I can push it up to github, and you could see the whole structure.  The current repo version has a structure significantly different than the one I am writing test cases for.  Really, I am writing test cases while trying to create a structure for more tests.

I am a programmer with a lot of years of experience in larger systems, with knowledge of a lot of programming languages.  I am not familiar with some of the structures needed to work with the lower levels of event driven systems, especially for testing.  I can do event handlers and emitters, but do know know how to get the 'hooks' in to be able test from the outside.  I have lots of experience testing from the inside, by adding debug code directly to the applications, and using tools to trace / step through.  Not so much with external testing frameworks.  I can get the code to work, but am having difficulty setting up some external regression testing where the code to be tested can not be simply called with test parameters, and the results checked.  I know lots about programming, but am missing some environment specific concepts, so I do not know the right question to ask.  Or even that there is a question.  I've worked with client side javascript before, but nodejs, npm, node-unit are all NEW.  So is that structure of event driven applications.  I've worked with browser based events, and years ago a bit with VB, so the concepts (I think) I understand, but implementation details are vague at best.

I have also worked directly with Arduino, enough to teach some programming / computer science with it for a local hobbyist robotics group.  My background includes general electronics knowledge as well.  That is enough to generally follow what the johnny-five components are doing.  I've got some of the highest and lowest levels of information, but am missing a layer in the low/middle.

mMerlin, 2015-07-22:
@ rwaldron 
If you have the time, and are interested, here are links to the repositories for the project that started the thread.  That was done awhile ago.  I learned enough to get reasonable coverage for the unit test.  Finally thought to dig out this thread, and post the links

https://github.com/mMerlin/johnny-load
https://github.com/mMerlin/auto-water

The live test showed it to be working well.  At least until someone physically removed the moisture sensor :(

rwaldron, 2015-07-22:
> At least until someone physically removed the moisture sensor

Humans can't be trusted :)

These projects are super cool—thanks for following up :) I love seeing this stuff—it really makes my day