rwaldron/johnny-five #795: Device or Firmware Error

ottofrisch, 2015-05-15:
Hi,
I've got the same issue using Arduino Mega 2560 on Win7 x64 and I've correctly uploaded StandardFirmata on the board via Arduino IDE.

I'm testing johnny-five+serialport libs. Everything's ok with something like this:
<pre>board = new j5.Board( { port: 'COM3' } )</pre>
(output)
<pre>1426735984300 Connected COM4
1426735984310 Repl Initialized</pre>
But with this:
<pre>board = new j5.Board ( { port: new SerialPort ( 'COM4', { baudrate: 9600 }, false ) } )</pre>
I got this message:
<pre>1426735984300 SerialPort COM4
1426735984310 Connected COM4
'on open' done <-- this is a console.log inside port.on ( 'open', ... )
'open' done <-- this is a console.log inside port.open ( ... )
1426735994311 Device or Firmware Error A timeout occurred while connecting to the Board. 
Please check that you've properly flashed the board with the correct firmware. 
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (/root/Documents/Arduino/test/node_modules/johnny-five/lib/board.js:405:26)
    at Timer.listOnTimeout (timers.js:110:15)</pre>

Any suggestion?

soundanalogous, 2015-05-15:
The baud rate must be 57600.

rwaldron, 2015-05-15:
Thanks @soundanalogous

closed 2015-05-15 by rwaldron