rwaldron/johnny-five #796: Sensor Frequency

vmadman, 2015-05-18:
I am attempting to sample a sensor at several thousand hertz (up to 40 kHz), however, the "freq" parameter, given that it is in ms, seems to be restrictive of reads to 1kHz.

Is there a known workaround to increase sampling beyond 1kHz?

Similarly, in my tests with `freq: 1`, I only saw a resolution of 64-65 hz.. or about 15ms.  Because it was suspiciously consistent I am assuming that it has something to do with the limitations of the serial transfer?  (Sorry, I'm very new to this)

Any advice would be appreciated.

rwaldron, 2015-05-18:
> I am attempting to sample a sensor at several thousand hertz (up to 40 kHz), however, the "freq" parameter, given that it is in ms

Sorry about this poorly named parameter. I have a plan for correcting this in a future version.

> Is there a known workaround to increase sampling beyond 1kHz?

No, and that's an unfortunate drawback to using JavaScript.

> Similarly, in my tests with freq: 1, I only saw a resolution of 64-65 hz.. or about 15ms. Because it was suspiciously consistent I am assuming that it has something to do with the limitations of the serial transfer.. or firmata maybe? 

You are exactly correct. There are two mechanisms in play here: 

- StandardFirmata's sample interval limit: 10ms
- JavaScript runtime capabilities, which is at best 1kHz (if the program is doing only _very_ minimal work each execution turn). 
  - Practically speaking, 200Hz is doable for most applications that require some amount of processing

If a sensor is initialized with a < 10ms interval (`freq`, again I apologize for this dumb naming), Johnny-Five will just report the most recent reading value that it has, as tightly as it can. 

> (Sorry, I'm very new to this)

No apologies necessary

vmadman, 2015-05-19:
Ok, I've begun on getting a bit more creative on the C side.. so wish me luck, lol.

There might be better names than `freq` .. but its not a horrible name.  I immediately knew what its purpose was and its proper, I think.

"Check that sensor!" ... "How frequently?" ... "Every 100 milliseconds!"

Works for me :)

rwaldron, 2015-05-19:
Wow, that's literally how I came to name it—no kidding!

vmadman, 2015-05-19:
Although "frequency", at least in electronics, is often used to refer to a particular attribute of a wave.. we shouldn't forget that it was derived from existing language.  Frequency actually refers to "how often".

> *n.* the rate at which something occurs or is repeated over a particular period of time or in a given sample.

So, if people give you a hard time about the variable name... they just don't know English :)  I understood your meaning right away and I think you chose the right word.  It's intuitive :)

scottgonzalez, 2015-06-02:
Perhaps we can create an FAQ and include information like this.

I think this issue can be closed though. I don't think there are any action items; it was just a question which has already been answered, right?

rwaldron, 2015-06-02:
If the resolution is to make a faq, then this can close when that's made ;)

scottgonzalez, 2015-06-14:
@rwaldron I've created #824 as a more general FAQ tracking issue. I've captured the relevant information from here, so it's probably ok to close this now.

rwaldron, 2015-11-05:
Thanks @scottgonzalez!

closed 2015-11-05 by rwaldron