rwaldron/johnny-five #797: Installation failure on Linux

190n, 2015-05-19:
I'm trying to install Johnny Five to use with my Arduino Uno on Linux (Ubuntu 14.10). The installation fails. I'm using io.js 2.0.1 and npm 2.9.1.

Here is the output:
```
npm WARN package.json @ No description
npm WARN package.json @ No repository field.
npm WARN package.json @ No README data
 
> galileo-io@0.8.14 postinstall /home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/galileo-io
> node scripts/postinstall

|
> serialport@1.7.1 install /home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport
> node-pre-gyp install --fallback-to-build

gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: 404 status code downloading tarball
gyp ERR! stack     at Request.<anonymous> (/usr/lib/node_modules/node-gyp/lib/install.js:246:14)
gyp ERR! stack     at emitOne (events.js:82:20)
gyp ERR! stack     at Request.emit (events.js:166:7)
gyp ERR! stack     at Request.onRequestResponse (/usr/lib/node_modules/node-gyp/node_modules/request/request.js:1156:10)
gyp ERR! stack     at emitOne (events.js:77:13)
gyp ERR! stack     at ClientRequest.emit (events.js:166:7)
gyp ERR! stack     at HTTPParser.parserOnIncomingClient (_http_client.js:414:21)
gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:88:23)
gyp ERR! stack     at Socket.socketOnData (_http_client.js:304:20)
gyp ERR! stack     at emitOne (events.js:77:13)
gyp ERR! System Linux 3.16.0-33-generic
gyp ERR! command "/usr/bin/iojs" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.1/Release/node-v44-linux-x64/serialport.node" "--module_name=serialport" "--module_path=/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.1/Release/node-v44-linux-x64"
gyp ERR! cwd /home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport
gyp ERR! node -v v2.0.1
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/iojs /usr/lib/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.1/Release/node-v44-linux-x64/serialport.node --module_name=serialport --module_path=/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.1/Release/node-v44-linux-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:169:7)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:953:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1020:5)
node-pre-gyp ERR! System Linux 3.16.0-33-generic
node-pre-gyp ERR! command "/usr/bin/iojs" "/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport
node-pre-gyp ERR! node -v v2.0.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/iojs /usr/lib/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.1/Release/node-v44-linux-x64/serialport.node --module_name=serialport --module_path=/home/ben/code/arduino/johnny-five/blink/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.7.1/Release/node-v44-linux-x64' (1)
npm WARN optional dep failed, continuing serialport@1.7.1
johnny-five@0.8.74 node_modules/johnny-five
├── ease-component@1.0.0
├── descriptor@0.1.0
├── temporal@0.4.0
├── color-convert@0.5.2
├── es6-shim@0.31.2
├── nanotimer@0.3.1
├── galileo-io@0.8.14 (remapped@0.2.1, es6-promise@2.1.1)
├── lodash@3.8.0
├── array-includes@1.1.0 (define-properties@1.0.1, es-abstract@1.2.1)
├── firmata@0.5.1 (object-assign@1.0.0, browser-serialport@2.0.2, es6-map@0.1.1)
└── chalk@1.0.0 (escape-string-regexp@1.0.3, ansi-styles@2.0.1, supports-color@1.3.1, strip-ansi@2.0.1, has-ansi@1.0.3)
```

rwaldron, 2015-05-19:
The failure is caused by serialport 

cc @voodootikigod

voodootikigod, 2015-05-19:
Use node.

190n, 2015-05-20:
Okay, it works now.

Do you know why it doesn't work on io.js?

closed 2015-05-20 by 190n

bryan-m-hughes, 2015-05-20:
io.js 2 introduced some breaking changes for native modules

rwaldron, 2015-05-20:
Ugh. Good to know.

voodootikigod, 2015-05-20:
yea... that... its a fun time!

*Chris Williams*

@voodootikigod <http://twitter.com/voodootikigod> | GitHub
<http://github.com/voodootikigod>

*The things I make that you should check out:*
SaferAging <http://www.saferaging.com/> | JSConf <http://jsconf.com/> |
RobotsConf <http://robotsconf.com/> | RobotsWeekly
<http://robotsweekly.com/>

Help me end the negativity on the internet, share this
<http://jsconf.eu/2011/an_end_to_negativity.html>.