rwaldron/johnny-five #800: Proximity - HC-SR04 not working

YesPapa, 2015-05-22:
Hi,

I have an issue when I try to use a proximity - HC-SR04 sensor when I try to run the johnny-five's default script. 

Here it is the issue:
(node) warning: possible EventEmitter memory leak detected. 11 ping-read-2 listeners added. Use emitter.setMaxListeners() to increase limit.

I cannot see where is the problem.

Perhaps somebody can help me.

Thank you.

scottgonzalez, 2015-05-22:
This usually happens when the wiring is not correct or the physical component isn't working.

rwaldron, 2015-05-22:
@YesPapa In addition to @scottgonzalez's tip, please make sure that you're using PingFirmata on the actual board: http://johnny-five.io/api/proximity/#pingfirmata

Please close when you've confirmed that the issue is resolved—thanks!

YesPapa, 2015-05-22:
Nope. I'm using StandardFirmata. So I must have a try with PingFirmata.

I come as soon as I have test it.

Thank you @rwaldron and @scottgonzalez

YesPapa, 2015-05-23:
Thank you very much. It is working like a charm. And sorry not to see the right page on the website.

rwaldron, 2015-05-23:
That's great news! 

No apologies necessary ;) How can we make that information easier to find or more obvious?

YesPapa, 2015-05-23:
Perhaps you can put a warning at the top of the page "Warning specific firmata is needed: PingFirmata".

In a better english ;-)

I think, it is the same thing for the "component classes". I would be better to put it at the top. when I first read it, I did not pay attention to it. Especially when the page is long.

You make a great job !

scottgonzalez, 2015-06-02:
It'd probably be good to include that information in the examples as well. Though I think if we're going to work on this, we should open a new issue to track the docs update and close this one.

rwaldron, 2015-06-02:
>  Though I think if we're going to work on this, we should open a new issue to track the docs update and close this one.

Seems good to me

YesPapa, 2015-06-03:
Yep. It can be closed.

Thank you.

rwaldron, 2015-06-03:
Thanks @YesPapa !

rwaldron, 2015-06-03:
derp

closed 2015-06-03 by rwaldron