rwaldron/johnny-five #801: lcd on ready function not working

rudemex, 2015-05-23:
Hi, the function on ready not working.

i have this code

var five = require("johnny-five"), lcd, board;

board = new five.Board({
  port: "/dev/cu.usbmodemfa141"
});

board.on("ready", function() {
	console.log("Placa lista.");
        
        lcd = new five.LCD({
		pins: [7, 8, 9, 10, 11, 12],
		rows: 4,
		cols: 20,
		backlight: 10,
	});

	lcd.on("ready", function(){
		console.log("LCD its ready");
		lcd.home().print("HELLO WORLD!"});
	});
});

console.log("\nEsperando a que inicialice el dispositivo...");

fivdi, 2015-05-23:
In newer versions of Johnny-Five, the LCD class doesn't emit a `ready` event. AFAIK it did in the past. There are some usage examples [here](https://github.com/rwaldron/johnny-five/wiki/LCD).

rwaldron, 2015-05-23:
@fivdi yep, the ready event has been eliminated.

rudemex, 2015-05-23:
ok, but in the examples it is used. (sorry, my english is not very good)

http://johnny-five.io/examples/lcd/

fivdi, 2015-05-23:
Hmm, it looks like the documentation needs to be updated. The [usage](https://github.com/rwaldron/johnny-five/wiki/LCD#usage) section of the wiki appears to be up to date. @rudemex is your LCD working now?

rudemex, 2015-05-23:
jajaja yes, they should update the documentation and examples. If the LCD is working, only he had this doubt. Thanks.

rwaldron, 2015-05-23:
Yep, needs updating. Sorry about that!

rwaldron, 2015-05-23:
I've updated all the examples—thanks for the heads up! http://johnny-five.io/examples/lcd/

closed 2015-05-23 by rwaldron