rwaldron/johnny-five #802: Microphone

rudemex, 2015-05-23:
There are some example of how to implement this shield ?

fivdi, 2015-05-23:
A microphone with an analog output? If so, see [here](https://github.com/rwaldron/johnny-five/issues/264).

rwaldron, 2015-05-23:
@rudemex secondarily to @fivdi's advice: what shield?

rudemex, 2015-05-23:
i have this shield http://articulo.mercadolibre.com.ar/MLA-552413873-modulo-sensor-de-sonido-microfono-regulable-arduino-_JM

rwaldron, 2015-05-23:
Ok, great, then exactly what @fivdi suggested should work just fine :)

rudemex, 2015-05-23:
great, There is a library of voice recognition ? I used the program bitvoicer but with Arduino code.

thanks

fivdi, 2015-05-23:
The [YL-56 microphone](http://www.wayengineer.com/sound-sensor-module-p-1051.html#.VWEIDqHyM_s) has a digital output.

rwaldron, 2015-05-23:
No, there is no voice recognition—you'd have to pass the value readings from the sensor to whatever processing library is available. You might try porting μSpeech https://github.com/arjo129/uSpeech

rudemex, 2015-05-23:
yes, i need recognition words, used this librery. 

Example
http://www.instructables.com/id/Speech-Recognition-with-BitVoicer-and-Arduino/

rwaldron, 2015-05-23:
https://github.com/rwaldron/johnny-five/commit/5579d12c61842671bac9709a7b02b61d31dc9ba2

rudemex, 2015-05-23:
thanks @rwaldron, @fivdi

rwaldron, 2015-05-23:
@rudemex no problem :)

rwaldron, 2015-05-23:
Also: http://johnny-five.io/examples/microphone/

rudemex, 2015-05-23:
as we say in Argentina...

Gracias Idolos! jajajaja

fivdi, 2015-05-24:
If the microphone being uses is the [YL-56](http://www.wayengineer.com/sound-sensor-module-p-1051.html#.VWEUQaHyM_t), which appears to be the case, a [digital-sensor](http://johnny-five.io/api/sensor/#digital-sensor) can also be used.

rwaldron, 2015-05-24:
@fivdi good eye on that one—I wonder if there is a fritzing for that component? I'm mobile so searching isn't in the cards for me

fivdi, 2015-05-24:
I can't find a fritzing for that specific component.

rudemex, 2015-05-24:
@fivdi This component has the same connection, working very good.

L -------------------------------> R
OUT(A0) - GND - VCC(3.3v)

fivdi, 2015-05-24:
So you have it working :) That's good news.
And the output from the microphone is analog, not digital?

rudemex, 2015-05-24:
i dont now jajajaja its working :P

https://instagram.com/p/3E95s1u41z/?taken-by=rudemex

rwaldron, 2015-05-25:
Wow, that's awesome! Will share the code for this?

rudemex, 2015-05-25:
yes, of course. I have to migrate code to js. but the orginal code is this. http://www.instructables.com/id/Speech-Recognition-with-BitVoicer-and-Arduino/

a software (http://www.bitsophia.com/BitVoicer.aspx only S.O. WIN) is needed to set the words to recognize and execute the functions.

I hope to migrate , you can do pretty amazing things

scottgonzalez, 2015-06-02:
Since the examples are live on the site, is there anything left to do before closing this issue?

rwaldron, 2015-06-02:
I don't think so, seems closable

closed 2015-06-02 by rwaldron