rwaldron/johnny-five #804: Component RFID-RC522

rudemex, 2015-05-24:
Hi, i have this shield http://articulo.mercadolibre.com.ar/MLA-546532441-modulo-rfid-rc522-1356mhz-tarjeta-llavero-tag-arduino-_JM#redirectedFromParent

There are some example to read and write?

Thanks

fivdi, 2015-05-25:
The contactless reader/writer looks like it has an NXP chip, but the picture is blurred and the numbers on the chip are not really visible. Most (all?) contactless reader/writer chips from NXP support I2C, SPI, and UART. On this board, only the SPI signals appear to be available. 

On some boards like the [Adafruit PN532 RFID/NFC Breakout and Shield](https://learn.adafruit.com/adafruit-pn532-rfid-nfc?view=all) I2C, SPI, and UART are available.

Johnny-Five doesn't support SPI currently, so I'm afraid you're out of luck this time.

rudemex, 2015-05-25:
buuu :(  so bad jajaja thanks!

christill, 2015-11-03:
So can you use johnny-five with https://learn.adafruit.com/adafruit-pn532-rfid-nfc?view=all ?

rudemex, 2015-11-03:
yes, i have two nfc, Adafruit PN532 and RFID522

christill, 2015-11-03:
@rudemex How does that work? I can't find any docs on it?

rudemex, 2015-11-03:
sorry, not , in the end I could not make it work from J5 , install the serialport and work with what he read me the port. https://instagram.com/p/8Mqdzju44t/?taken-by=rudemex

christill, 2015-11-03:
@rudemex Ok, cool, Am I able to use johnny five along side this on the same arduino uno? I want to be able to scan and NFC card then turn on an LED attached to the same board, using J5 for controlling the LED. Thanks

rudemex, 2015-11-03:
@christill I'm working on a project with NFC , unfortunately , you'll have q send / receive data through the serial port and handle everything from Arduino code with J5 I could not make it work.

christill, 2015-11-03:
@rudemex. OK, I'm confused now. The RFID is an arduino shield, so connected to the arduino. Can I use J5 and serialport on the same board?

rudemex, 2015-11-03:
sorry , my English is not very good .

video for example , do not use J5 , but if you use nodejs . what I do is, Arduino code read the ID tag , and nodejs is reading what gets the serial port , and from nodejs do what I have to do and return to send data through the serial port , so they receive the arduino , arduino code and get from that data sent and do what I do. you will not be able to do everything from J5 , and that's what you got to carry on the atmega the standarfirmdata to work J5 and this is not updated yet to work for i2c , but above what they said.

if you like , I can go through a git the source of that project.

rwaldron, 2015-11-03:
We just landed support in the io protocol for serial/uart, so expect new component support soon.

rudemex, 2015-11-03:
@rwaldron great!

christill, 2015-11-03:
@rwaldron awesome! How soon? :)

rwaldron, 2015-11-04:
Once @soundanalogous publishes a release of [Firmata](https://github.com/firmata/arduino), then Johnny-Five can start planning new component classes using the serial APIs. 

My present thinking is to create a generic  `Receiver` class (and of course `Transmitter`), which can be used to "receive" data and deliver it via data events. That would then be used to implement RFID/NFC component classes. The best way to get started is to look at all of the available RFID/NFC boards, shields, breakouts, etc and create a table (spreadsheet) that illustrates the capabilities of each to find the most common intersections. From that, we can work together to design APIs that make the most common use cases as easy as possible.

kmoe, 2016-02-09:
@rudemex it would be awesome to see the code for your project!

rudemex, 2016-02-09:
@kmoe yes, the base of the project is divided into two parts

ARDUINO CODE

`/*
 * ----------------------------------------------------------------------------
 * https://github.com/miguelbalboa/rfid
 * Typical pin layout used:
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
 *             Reader/PCD   Uno           Mega      Nano v3    Leonardo/Micro   Pro Micro
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin
 * -----------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
 * SPI SS      SDA(SS)      10            53        D10        10               10
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15
 */

#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN 10
#define RST_PIN 9
MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance.
byte cardPresent;

void setup() {
  Serial.begin(9600); // Initialize serial communications with the PC
  SPI.begin();      // Init SPI bus
  mfrc522.PCD_Init(); // Init MFRC522 card
  //Serial.println("Ready to read!");
}

void loop(){
  
  // Look for new cards
  if (mfrc522.PICC_IsNewCardPresent() && mfrc522.PICC_ReadCardSerial()){
    
    //Serial.print("Card UID:");
    for (byte i = 0; i < mfrc522.uid.size; i++) {
      //Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
      Serial.print(mfrc522.uid.uidByte[i], HEX);
    } 
    Serial.println();
    delay(1500);
  }
}`

NODEJS CODE - configure var serialPOrt and the baudrate

`var serialport = require("serialport");
var SerialPort = serialport.SerialPort;

var serialPort = new SerialPort("/dev/tty.usbmodemFA131", {
  baudrate: 9600,
  parser: serialport.parsers.readline("\n")
});

serialPort.on("open", function () {
  console.log('Serial Port Open');
  console.log('=================');
  serialPort.on('data', function(data) {
    console.log('Card UID: ',data);
  });
});
`

georgeterbush, 2016-02-20:
@rwaldron has there been any more development in Johnny Five to implement RFID/NFC component classes using the serial APIs to directly support cards like the Adafruit PN532 and RFID522?

gedeagas, 2016-09-06:
Waiting for official support

closed 2016-09-06 by rudemex

krushnaR, 2017-02-06:
Any updates guys?

matejicekme, 2017-02-22:
Waiting for the too :) !

mdtahmid, 2019-10-23:
> @kmoe yes, the base of the project is divided into two parts
>
> ARDUINO CODE
>
> /*
>
> * https://github.com/miguelbalboa/rfid
> * Typical pin layout used:
>
> * ```
>           MFRC522      Arduino [...]

Hi mate, do you mind creating 

> @kmoe yes, the base of the project is divided into two parts
>
> ARDUINO CODE
>
> /*
>
> * https://github.com/miguelbalboa/rfid
> * Typical pin layout used:
>
> * ```
>           MFRC522      Arduino [...]

Hi mate, mind creating a repository of the code? I saw your Instagram video and thought it was awesome. Appreciate it!

rudemex, 2019-10-24:
Hello @mdtahmid, how are you?
Thank you very much for your comment. Unfortunately at the moment I'm a bit away from Arduino ='( but I have plans to do something big with the things I was learning in this time, and no doubt, one of my favorite technologies is the NFC, which I use in some functional projects for a lot of people.
As soon as I'm done for a while, I'm going to create a repository about my Arduino projects, even if they're base concepts and startup projects.

Greetings Atte.

awong1900, 2020-01-14:
wait for.  And i want contributive.