rwaldron/johnny-five #805: Ubuntu Core

bartmichu, 2015-05-25:
Hi, has anyone tried j5 with that new Ubuntu Core thingy?

rwaldron, 2015-05-25:
Not that I know of, but I also don't know much about it :)

bartmichu, 2015-05-26:
It's being positioned as IoT operating system:
http://www.ubuntu.com/things

Apparently it works on BBB and RPi. I'm just trying it out in a virtual machine and I'm not even sure if Node.js is supported :/

rwaldron, 2015-05-26:
Cool! Keep posting your progress here :)

cc @julianduque @beriberikix @bryan-m-hughes

bryan-m-hughes, 2015-05-26:
I haven't messed with it myself yet, but I've heard that it's kinda unstable on the RPi 2 at the moment. There is also https://github.com/bryan-m-hughes/raspi-io/issues/24.

I'm really curious to follow it, though, I think it could be a pretty major boon in terms of the RPi ecosystem.

beriberikix, 2015-05-26:
Definitely keep us posted! It looks like the RPi build is [not official](http://developer.ubuntu.com/en/snappy/start/#snappy-raspi2) yet but has some legs. Might need to get me an RPI2 now :)

nebrius, 2015-11-05:
I still haven't played with it yet. I've had a few people write in curious about other OS'es, but I feel like Arch support is more requested than Ubuntu Core. It's an admittedly very small sample size.

reconbot, 2016-05-25:
I'm going to close due to age, "snappy" as it's now called (I think?) still has very light adoption and support isn't really a j5 issue.

closed 2016-05-25 by reconbot

sant123, 2016-12-10:
It seems now there's support for. https://github.com/nebrius/raspi-i2c/pull/7

nebrius, 2016-12-12:
FWIW Ubuntu Core _should work_ with raspi-io, but it's not _supported_. The only OS that is officially supported by raspi-io still is Raspbian and derivatives from it. While things probably work on Ubuntu Core, I don't test against it during development so it could break at any time.