rwaldron/johnny-five #809: Consider Octavian for piezo frequency management

Resseguie, 2015-05-29:
As presented at JSConf2015
https://github.com/stevekinney/octavian

By @stevekinney

stevekinney, 2015-05-29:
I'm willing to work on making whatever changes/features you all need to make your lives easier. Let me know.

mcdonnelldean, 2015-06-03:
@Resseguie @stevekinney Sorry! I completely missed this. 

So I think Octavian would be an awesome fit and is more or less ready to use, the main issue is RTTF support and how it relates to our note input.

@stevekinney I've put some notes specifically for you here: https://github.com/rwaldron/johnny-five/pull/741 Under the heading Octavian. If we can kick off the process on what makes sense for you to have and what makes sense for us to keep we can get cracking.

stevekinney, 2015-06-05:
Awesome! I’m teaching an 8-hour class on JavaScript this weekend and I’ll be ready to turn my full attention to this.

mcdonnelldean, 2015-06-05:
@stevekinney Sweet. I'm on GMT time but if you need anything let me know. Over the weekend I will get a few more tests and that in so that there isn't too much extra effort on your end. Octavian should actually slot in handy enough once the changes are done.

Let me know when you get free and we can touch base :D

mcdonnelldean, 2015-07-14:
@stevekinney Did you ever get a chance to look into some of this? If not this weekend I was going to see what I could pull from here to Octavian so we can finish off the Piezo stuff?

stevekinney, 2015-07-23:
@mcdonnelldean I have been traveling and had not gotten a change, but it has definitely not fallen off my radar.

closed 2018-08-28 by stale[bot]