rwaldron/johnny-five #822: Display LCD 20x4 not working

rudemex, 2015-06-11:
Hi , update the library j5 as conduct a testing with 20x4 lcd display weird characters began to show .

https://fbcdn-sphotos-h-a.akamaihd.net/hphotos-ak-xta1/v/t34.0-12/11277490_10153450122213623_1777188040_n.jpg?oh=4ea88f3746cc2fcf0ff6675986b0f215&oe=557B18D5&__gda__=1434126453_3f0437c6b7c394e6e3c38e62157cce79

rwaldron, 2015-06-11:
Thanks for filing!

Please provide the following: 

- the exact model of device you're using
- the exact code that produced the non-sense characters
- a clearer picture of the entire wiring

rudemex, 2015-06-11:
model 2004A driver is , but indistinct, and it worked without defining, this is an old photo of when running the code https://instagram.com/p/3B6R4rO43F/?taken-by=rudemex this is code.

var five = require("johnny-five"), lcd, board;

board = new five.Board({
  port: "/dev/cu.usbmodemfa141"
});

board.on("ready", function() {
	console.log("Placa lista.");

	lcd = new five.LCD({
		pins: [7, 8, 9, 10, 11, 12],
		rows: 4,
		cols: 20,
		backlight: 13
	});

  	lcd.useChar("heart");
	lcd.cursor(1, 5).print("I :heart: WOONKY", { dontProcessSpecials: false });

});

console.log("\nEsperando a que inicialice el dispositivo...");

rwaldron, 2015-06-11:
Thanks! I will look into this a little tonight, but definitely tomorrow :)

rwaldron, 2015-06-11:
Let me try to understand: 

- This worked correctly before?
- Something happened and the result was non-sense characters?

rudemex, 2015-06-11:
yes

rudemex, 2015-06-12:
WORKING https://fbcdn-sphotos-h-a.akamaihd.net/hphotos-ak-xft1/v/t34.0-12/11430261_10153452166223623_543543947_n.jpg?oh=e23a164785de2cc2acf0aafcb964d0c5&oe=557C3C34&__gda__=1434210950_c0054ae06c2dd68042593b0f1993bad6 , reinstall the library , and connect the cables to the breadboard again(http://www.ladyada.net/wiki/tutorials/learn/lcd/charlcd.html), before ide cable used a pc , I think the problem was that cable.

Thanks.

rwaldron, 2015-06-12:
That's great news! Can you close the issue? :D

rudemex, 2015-06-12:
yes!! :D

closed 2015-06-12 by rudemex