rwaldron/johnny-five #824: Create an FAQ

scottgonzalez, 2015-06-14:
Let's collect some items that should go into an FAQ, then create one.

### Why is my sensor input limited to 50-200Hz?

(from https://github.com/rwaldron/johnny-five/issues/796#issuecomment-103080959):

* StandardFirmata's sample interval limit: 10ms
* JavaScript runtime capabilities, which is at best 1kHz (if the program is doing only very minimal work each execution turn).
  * Practically speaking, 200Hz is doable for most applications that require some amount of processing

### Why do I get random data on pins 0 and 1?

Pins 0 and 1 are used for transmitting and sending serial data. Johnny-Five communicates from the host machine to the microcontroller over the serial port, which will cause interference with any other components connected to those pins.

### I've used all the pins on my microcontroller, but I want to connect more components. What can I do?

Use a shift register.

mMerlin, 2015-06-28:
> Use a shift register

or other form of port expander.

reconbot, 2016-05-25:
@scottgonzalez if you still want to make this happen (and it's good info) I'd open this on johnny-five.io's repo https://github.com/j5js/johnny-five.io/issues

The website kind of changed the game.

closed 2016-05-25 by reconbot