rwaldron/johnny-five #827: test/pin.js fails if test/led.js doesn't run first

dtex, 2015-06-17:
I doubt the former is supposed to be dependent on the latter so probably a bug.

(note led.js isn't in the output. I moved it out of the test folder)

```bash
Testing joystick.js..............OK
Testing lcd.js...................OK
Testing options.js.OK
Testing piezo.js...............OK
Testing pin.js.....................F.
>> PinShape - defaultPinShape
>> Message: pin(19) address
>> Error: 5 == 19
>> at Object.exports.PinShape.defaultPinShape (test/pin.js:511:10)
>> at Object.exports.PinShape.setUp (test/pin.js:481:5)

Testing ping.js....OK
Testing pir.js.....OK
```

rwaldron, 2015-06-18:
```
$ grunt nodeunit:just:pin
Running "nodeunit:just:pin" (nodeunit:just) task

Running "nodeunit:tests" (nodeunit) task
Testing pin.js.......................OK
>> 134 assertions passed (315ms)

Done, without errors.
```

dtex, 2015-06-18:
board.js breaks it, led.js fixes it:

```bash
Running "nodeunit:tests" (nodeunit) task
Testing pin.js.......................OK
>> 134 assertions passed (133ms
```
```bash
Running "nodeunit:tests" (nodeunit) task
Testing board.js.................................OK
Testing pin.js.....................F.
>> PinShape - defaultPinShape
>> Message: pin(19) address
>> Error: 5 == 19
>> at Object.exports.PinShape.defaultPinShape (test/pin.js:511:10)
>> at Object.exports.PinShape.setUp (test/pin.js:481:5)
```

```bash
Running "nodeunit:tests" (nodeunit) task
Testing board.js.................................OK
Testing led.js...............................................OK
Testing pin.js.......................OK
>> 463 assertions passed (279ms)
```

dtex, 2015-06-18:
Specifically, the ```Board.purge()``` in led.js fixes it. Haven't figured out what in board.js breaks it.

rwaldron, 2015-06-18:
Ugh. Those tests, "PinShape", etc. need to test what happens to non-object arguments. So we can't specify a `board` and all the garbage that entails

rwaldron, 2015-06-18:
The setUp for those tests doesn't even make sense to me right now

rwaldron, 2015-06-18:
I think we should back this out 02d58172433d83bdf93bc52da72be4b18c3d2311

dtex, 2015-06-18:
Fine by me but then I don't understand it really. Should I submit a PR or is there some git wizardry to back out a merged PR? We should also ping the the author to see if he can fix it properly instead of us just axing it without telling him.

rwaldron, 2015-06-18:
I'm not sure what to do, those tests are valuable, so maybe we don't back them out, but I don't have time today to figure out what it is that I missed when they were first landed.

dtex, 2015-06-18:
Let me see if I can get the author involved...

Paging @mMerlin The tests you submitted are valuable to us but they are causing a problem. Can you please check out this thread from the top and advise?

rwaldron, 2015-06-18:
It looks like the `setUp` just needs to replace

```j
var io = new MockFirmata();

io.emit("ready");
```

with

```js
newBoard()
```

And call `Board.purge()` in `tearDown`

dtex, 2015-06-18:
I'll make that change and test it out.

rwaldron, 2015-06-18:
Did you figure it out? I found the source.

dtex, 2015-06-18:
Well, not yet but it looks like we had a leaking board instance at the top of board.js (got injected into the boards array but never used/removed). Removing the wreaked havoc on some other tests which probably need newBoard().

dtex, 2015-06-18:
Never mind. It was only leaking because I hadn't replaced some of the other new Board() calls in baord.js with newBoard() I see you've already PR'd the fix. @nebrius thinks you are actually a set of twins because you get stuff done so fast.

rwaldron, 2015-06-18:
:dancers:

closed 2015-06-18 by rwaldron

mMerlin, 2015-06-19:
Late to the conversation, but it looks like everything is resolved now.  If I understand correctly, the fix was to add teardown to remove the test board being used, so that it did not interact with other tests.

rwaldron, 2015-06-19:
@mmerlin yep, I think that sorted it out

Resseguie, 2015-06-19:
Do we need to document that somewhere on the wiki for future contributor reference as best practice?

rwaldron, 2015-06-19:
@Resseguie I think this was really my fault—there were bad tests in the file and those were copied (because that makes the most sense).