rwaldron/johnny-five #829: move:complete is emitted by servoWrite when degrees === last.degrees

dtex, 2015-06-18:
```
var servo = new five.Servo(11); // Servo is at 90 by default
servo.to(180); // move:complete is not emitted. Expected.
servo.to(180); // move:complete is emitted. Not expected.
```

Also with the upcoming move to animation() for servo.to() we get:
```
var servo = new five.Servo(11); // Servo is at 90 by default
servo.to(90, 1000); // move:complete is emitted twice. Not expected.
```
I believe that if we are not going to emit move:complete when calling servo.to without a duration, then it should not be emitted just because degrees === last.degrees. In the event that an animation is called, animation() will run the callback that emits the event after the requested period of time has passed.

I think we should remove:
https://github.com/rwaldron/johnny-five/blob/master/lib/servo.js#L29
and
https://github.com/rwaldron/johnny-five/blob/master/lib/servo.js#L105

rwaldron, 2015-06-18:
I'm ok with this

dtex, 2015-06-18:
Will include this change with my animation() wrapping servo.to  PR. That's just waiting on #828 which has a fix pending (thank you for that BTW).

rwaldron, 2015-06-18:
No problem, but we still need to address that breakage :\

dtex, 2015-06-18:
Which breakage do you mean? #827 won't affect this. Or did you mean this is a breaking change because it alters previous behavior?

rwaldron, 2015-06-18:
The breakage I pinged you on [here](https://github.com/rwaldron/johnny-five/commit/2fd7f9caccd5d404f597453c8d153c4cc48ef51c#commitcomment-11754585) and [here](https://github.com/rwaldron/johnny-five/pull/833#issuecomment-113266582)

Resseguie, 2015-06-28:
Hmm, I somehow didn't see that ping either. I'll try to have a look and see if I have any useful input.

dtex, 2015-07-12:
Included in #858

dtex, 2015-12-11:
Forgot to close this.

closed 2015-12-11 by dtex