rwaldron/johnny-five #831: PCF8574 does not work as input

rodrigoadachi, 2015-06-18:
<code>
  board.on("ready", function() {
    var virtual = new five.Board.Virtual(
      new five.Expander({ controller: "PCF8574",  address: 0x20 })
    );
    var button = new five.Button({
      board: virtual,
      pin: 7, 
    });
    button.on("press", function() {
      console.log('button on');
    });
</code>

Did tests with outputs and everything works ok, however when declare inputs with outputs and inputs do not work only.

rwaldron, 2015-06-18:
Thanks for the report, I will take a look at this today

rwaldron, 2015-06-18:
Can you post a picture of your exact wiring? Thanks!

rwaldron, 2015-06-18:
@rodrigoadachi can you try this with the johnny-five @ master?

rodrigoadachi, 2015-06-18:
[image]

rodrigoadachi, 2015-06-18:
Como assim johnny-five @ master?

rodrigoadachi, 2015-06-18:
var   , five = require("johnny-five");

  var board = new five.Board({ port: "/dev/ttyUSB0" });

  board.on("ready", function() {
    var virtual = new five.Board.Virtual(
      new five.Expander({
        controller: "PCF8575",
        address: 0x20
      })
    );

    var led = new five.Led({
      board: virtual,
      pin: 0, 
    });

    var button = new five.Button({
      board: virtual,
      pin: 7, 
    });

  button.on("press", function() {
    console.log('pulsador1 on');
    led.on();
  });

  button.on("release", function() {
    console.log('pulsador1 off');
    led.off();
  });

});

rwaldron, 2015-06-18:
I just pushed the latest changes to npm: johnny-five@0.8.80

I confirmed the bug you experienced, but will also try your code above

rwaldron, 2015-06-18:
The code you have above uses `PCF8575`, but this was filed about `PCF8574` and that's the chip in the diagram—these are not interchangeable. Which one should I test with?

rodrigoadachi, 2015-06-18:
Ohh, sorry, 'cause I tried using the IC PCF8575 PCF8574, but the fact is the PCF8574.

rwaldron, 2015-06-18:
Ok, thanks. I'm still seeing the issue with your code example (updated to use `PCF8574`) so I'll continue working on this.

rodrigoadachi, 2015-06-18:
Okay, I thank you for your attention.

rwaldron, 2015-06-18:
Please try your code with johnny-five@0.8.81

rodrigoadachi, 2015-06-18:
It did not work, I tested with this code, and only expanderP0 goes to 1, where it should be 0, but if I comment on the entries, exits work.

var board = new five.Board({ port: "/dev/ttyUSB0" });
  
  board.on("ready", function() {

    var virtual = new five.Board.Virtual(
      new five.Expander({
        controller: "PCF8574",
        address: 0x20
      })
    );

    var expanderP0     	= new five.Led({ board: virtual, pin: 0 });
    var expanderP1     	= new five.Led({ board: virtual, pin: 1 });
    var expanderP2      = new five.Led({ board: virtual, pin: 2 });
    var expanderP3      = new five.Led({ board: virtual, pin: 3 });

    var expanderP4   	= new five.Button({ board: virtual, pin: 4 });
    var expanderP5  	= new five.Button({ board: virtual, pin: 5 });
    var expanderP6   	= new five.Button({ board: virtual, pin: 6 });
    var expanderP7   	= new five.Button({ board: virtual, pin: 7 });

    expanderP0.off();
    expanderP1.off();
    expanderP2.on();
    expanderP3.on();

    expanderP4.on("press", function() {
      console.log('expanderP4:');
  	});
    expanderP5.on("press", function() {
      console.log('expanderP5:');
  	});
    expanderP6.on("press", function() {
      console.log('expanderP6:');
  	});
    expanderP7.on("press", function() {
      console.log('expanderP7:');
  	});

});

rwaldron, 2015-06-18:
When I run your code, the result is: 

- expanderP0 is off
- expanderP1 is off
- expanderP2 is on
- expanderP3 is on

Can you take a photo of your hardware?

rodrigoadachi, 2015-06-18:
I'll try

rwaldron, 2015-06-18:
Thanks, it will be very helpful

rodrigoadachi, 2015-06-18:
I can not access it, installed above the ceiling, it controls my enlightenment, is running with a stand-alone application. But I try to take a picture tomorrow.

rwaldron, 2015-06-18:
I'm confused—what is installed in the ceiling?

rodrigoadachi, 2015-06-18:
Sorry for the translation, My card is installed above the ceiling, it is connected in a cubieboard remotely access.

rwaldron, 2015-06-18:
No worries, I'm patient, I just want to make sure that I understand you :)

rodrigoadachi, 2015-06-18:
OK thank you.

rodrigoadachi, 2015-06-22:
This is the famous board.

I did some testing today and nothing worked, not even the out.
Only when commenting on the statements of the entries that the outputs return to work

[image]

board.on("ready", function() {
    var virtual = new five.Board.Virtual(
      new five.Expander({
        controller: "PCF8574",
        address: 0x20
      })
    );
   // out
    var expanderP0     	= new five.Led({ board: virtual, pin: 0 });
    var expanderP1     	= new five.Led({ board: virtual, pin: 1 });
    var expanderP2      = new five.Led({ board: virtual, pin: 2 });
    var expanderP3      = new five.Led({ board: virtual, pin: 3 });
    var led 			= new five.Led.RGB({ pins: { red: 6, green: 5, blue: 9 } });
    // in
    var expanderP4   	= new five.Button({ board: virtual, isPullup: true, pin: 4 });
    var expanderP5  	= new five.Button({ board: virtual, isPullup: true, pin: 5 });
    var expanderP6   	= new five.Button({ board: virtual, isPullup: true, pin: 6 });
    var expanderP7   	= new five.Button({ board: virtual, isPullup: true, pin: 7 });
   // set out
   expanderP0.off();
   expanderP1.off();
   expanderP2.off();
   expanderP3.off();
   led.off();
   // get in   
   expanderP4.on("press", function() {
     console.log('expanderP4:');
   });
   expanderP5.on("press", function() {
      		console.log('expanderP5:');
    	});
	expanderP6.on("press", function() {
      		console.log('expanderP6:');
    	});
	expanderP7.on("press", function() {
      		console.log('expanderP7:');
    	});
});

rodrigoadachi, 2015-06-22:
This is the motherboard kkk

[image]

rwaldron, 2015-06-22:
> Only when commenting on the statements of the entries that the outputs return to work

Can you paste the exact code you're referring to?

In the first photo, there looks like 6 ground and signal pairs (3 each on either side), is that correct? Where are the buttons connected?

> ```js
var expanderP4 = new five.Button({ board: virtual, isPullup: true, pin: 4 });
var expanderP5 = new five.Button({ board: virtual, isPullup: true, pin: 5 });
var expanderP6 = new five.Button({ board: virtual, isPullup: true, pin: 6 });
var expanderP7 = new five.Button({ board: virtual, isPullup: true, pin: 7 });
```

Why are you using the `isPullup` flag? 

> This is the motherboard

Unfortunately, this is too blurry for me to learn anything from it :(

- Is that an arduino nano? 
- What is the device that's connecting the processor to the host machine?
- Can you list the hardware there?

rodrigoadachi, 2015-06-23:
Only works when the OUTPUTS comment on the statements of INPUTS.
this:

//var expanderP4   	= new five.Button({ board: expanderX21, pin: 4 });
//var expanderP5  	= new five.Button({ board: expanderX21, pin: 5 });
//var expanderP6   	= new five.Button({ board: expanderX21, pin: 6 });
//var expanderP7   	= new five.Button({ board: expanderX21, pin: 7 });

Use optos 4N35 at the entrances, because of this the inputs are pull-up

is board below has a Pro Micro Arduino connected to the computer via RS232
[image]

And this, possesses one PCF8574, 4 relays and 4 solids in optos 4N35 pull-up
[image]

The assembly is ugly, but it works only using Arduino kkk

Ah, sorry for the delay in responding

rwaldron, 2015-06-23:
This is soooo helpful—thank you! I will work on this more as soon as I can :)

rwaldron, 2015-06-23:
> but it works only using Arduino kkk

This is valuable. I will make sure to use exactly the same board you have above :)

rodrigoadachi, 2015-06-23:
ok, thank you, kkkkk

rodrigoadachi, 2015-07-01:
Hello, any news?

rwaldron, 2015-07-02:
I have been unable to reproduce the issues that you've described above. Let's ping @dtex @Resseguie @BrianGenisio @frxnz @ajfisher perhaps one of them can attempt as well

rodrigoadachi, 2015-07-02:
I added another PCF8574, address 0x21, one 0x20 pcf to exits and other pcf 0x21 for entries, and so does

closed 2018-08-28 by stale[bot]

ryrych, 2023-09-04:
@rwaldron This issue has been closed, but I'm wondering if it is possible to connect buttons via an expander. Is there any diagram that shows how to make buttons work as input when connected via PCF8574?
Here's the JS (excerpt)

```js
board.on('ready', () => {
      connected = true;
      virtual = new five.Board.Virtual(new five.Expander({ controller: 'PCF8574', address: 0x20 }));

      const button = new five.Button({
        board: virtual,
        pin: 1,
        holdtime: 1000,
        isPulldown: true,
      });

      button.on('up', function () {
        console.log('hello');
      });
      resolve({ board: board });
    });
  });
```

I'm using ESP8266 WIFI.

ryrych, 2023-09-05:
@rwaldron what I'm going to achieve is to connect two buttons in Nodemcu ESP8266. If this is possible somehow differently, please let me know. I'm not attached to expander solution.