rwaldron/johnny-five #832: Device or firmware error

gtzogana, 2015-06-18:
Hi, I'm trying to talk to an Arduino Uno using a HC-06 Bluetooth module. I've uploaded the StandardFirmata sketch, and I'm able to see the device as a serial port after pairing (I'm able to `screen` /dev/tty.HC-06-DevB), but when I run a basic `board.on` script, I get the following error:

`Device or Firmware Error A timeout occurred while connecting to the Board. `

rwaldron, 2015-06-18:
I'm not sure if the same instructions will work, but here are the docs for the HC-05: https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-HC-05-Bluetooth-Serial-Port-Module

gtzogana, 2015-06-18:
No luck with those -- I'm not able to get the full configuration messages as shown, even by changing the baud rate. Is that command meant to change the baud rate of the Bluetooth module?

rwaldron, 2015-06-18:
I'll need to ping other with more bluetooth experience

@ajfisher @monteslu

ajfisher, 2015-06-19:
@gtzogana BT can be a notoriously challenging thing to debug.

So things to try are this:

Make sure that the HC-06 is paired as a serial device. In OSX this should be just connecting it and making sure it's working

Ensure that the BT module is running at 57600 BAUD. Often they aren't by default so you'll need to configure that with a sketch to tell it to operate at that speed. If you absolutely know it's not running at that speed then change the `firmata.begin(57600)` line in the firmata sketch to be the same baud rate your BT module is running at.

You can check if your machine is talking to the arduino via BT by doing the following:

* set up arduino with firmata and connected to BT module
* connect mac to the BT module and get your BT serial port (something like `/dev/tty.HC-06BT`) 
* Make sure you have permissions to talk to the BT serial port `ls -la /dev/tty.HC-06BT` - if not change the udev rules or add yourself to the group it is presented as.
* Now use screen (or some other serial terminal) to talk to the arduino over BT:

```
$ screen /dev/tty.HC06BT 57600
```

* Now hit reset on the arduino - if everything is working properly you should see some text appear similar to this `StandardFirmata.ino` as Firmata will send it's filename to the serial port on boot. 

If you get that then you should be running fine with Johnny Five. IF you aren't then debug your Arduino /  BT / OSX connection because that's where the issue will be.

Other things you can try is doing something like running a sketch that pumps out a counter to serial and then try the same thing I described above so you can tinker with connections etc until you start seeing numbers appearing in your serial terminal. Once you get that going you should be good on the rest.

gtzogana, 2015-06-19:
Thank you both very much for your help! Changing the baud rate in the Firmata sketch solved the problem.

rwaldron, 2015-06-19:
@gtzogana awesome!!

closed 2015-06-19 by rwaldron

ajfisher, 2015-06-19:
Great news. @rwaldron I'll spend some time updating that wiki entry to be
more detailed like the WiFi one.