rwaldron/johnny-five #844: The sensor change event is not properly mixing filtering and the threshold

mMerlin, 2015-06-30:
PR #843 contains additions to test/sensor.js that detected the problem, and lib/sensor.js that fixes it.

The sensor change event is not properly mixing filtering and the threshold.  When checking if the latest calculated median value should be used for a change event, it (median) is checked to see if it is not between last +/- the threshold.  However, the 'last' variable contains the second most recent raw value from {xxx}Read.  Not the median value report by the previous change event.  last needs to be updated (only) when a new change event is emitted.  Not every time a new {xxx}Read value is
acquired.

The existing (old) code works (only) for digital sensors, that do not use the median filtering logic.

Last also needs to be initialized to a value that will cause the first calculated median value to always be outside the change threshold.

Digital sensors do not actually use the contents of the samples array, but the array is still filled in.  But it is never emptied.  It appears that using digital sensors will leak memory.

rwaldron, 2015-07-01:
Those are great bugs and the fixes solid as well. I've landed #843—is this issue resolved and closable?

mMerlin, 2015-07-01:
Yes this can be closed (doing that here).  I spotted these bugs while adding to the sensor tests in preparation for getting back to the sensor enhancements briefly discussed in #788 May 12.  I wanted more tests for regression before I started with any major changes.

I think there will be some more small changes before I get there.

Should the event frequency be dynamically changeable, for sensor or other device classes? The .freq property seems directly settable, but from reading the code, it does not appear that setInterval is updated when it changes.  That is next on the list of tests to add.  It looks to me like .freq needs to be changed to getter/setter (.defineProperties), and the setter needs to call clearInterval, then setInterval again.  This looks like a general problem.  A search (grep) for setInterval finds references in 15 files, but only esc, piezo, and servo call clearInterval.  This might not be such a *small* update to implement.  Would need to start by looking at several of the usage cases, and trying to devise a common / generic solution.  Drop in mixins, or maybe add to fn.

closed 2015-07-01 by mMerlin