rwaldron/johnny-five #845: Possibility to implement a FingerPrint Reader/Scanner

icaroscherma, 2015-07-01:
Hello, I was wondering if it is already possible to use a FingerPrint Reader like [AdaFruit Fingerprint Scanner](https://www.sparkfun.com/products/13007) through Johhny-Five or if I still need to implement the interface, I found the library [FPS_GT511C3](/sparkfun/Fingerprint_Scanner-TTL) which uses TTL Serial, but I'm not sure how to proceed. About the boards, I will probably use an Arduino (Uno Rev3) which uses ATMega328p or RaspBerry Pi GPIO Ports (if possible).

Ps.: Probable valid tags for this "issue": __Feature__, __Enhancement__, __Question__.
Ps.: I followed the guideline but I'm still not sure if this is a __Feature Request__ or __Hardware Support__ .

Thanks for reading until here. =)

rwaldron, 2015-07-01:
Until _very_ recently, this wouldn't have been possible because there was no UART/Serial capabilities in the Firmata protocol. [That's changing](https://github.com/firmata/arduino/pull/205)! 

At this point, I have a growing survey of platform capabilities: https://docs.google.com/spreadsheets/d/1mxlwwmyqHiAPU3NEpJXSWdpbgZjFaHbUEBAi-74SUU0/edit?usp=sharing in the next few days I plan to [draft an API for the io-plugin spec](https://github.com/rwaldron/io-plugins/issues/8). Once we have this, then I think your request is totally possible :)

icaroscherma, 2015-07-02:
Hmmm, interesting. 

Since will have changes into the Firmata Protocol, I will need to download a newer version (of the code or Arduino IDE containing the newer version) and upload it to the ATMega328p, right?

And about the API for the IO-Plugin you mean for the Fingerprint itself, like `` var fp = new FingerPrint("5");`` or "just" for the IO-Plugin? Because I think I can take a look into what is required/used in FingerPrint Scanners/Readers and help with the interface...

rwaldron, 2015-07-03:
The IO-plugin spec is just an aside. The Fingerprint reader will just be implemented like any other sensor

alphashuro, 2016-03-25:
Any updates on this?

gpincheiraa, 2016-07-05:
@alphashuro +1

Xilver, 2016-12-02:
Is this still in the pipeline?

dtex, 2016-12-04:
Serial landed in Firmata about a year ago and J5 classes have already been built upon it (GPS was the first). All this needs is a motivated contributor.

Awk34, 2016-12-12:
I just bought one of these: https://www.sparkfun.com/products/11792

Hopefully I'll get a chance to J5 it!

bneiluj, 2017-05-21:
Is someone working on this issue ?

lesliechw, 2017-06-06:
Can this API be use with Finger Print module? http://johnny-five.io/api/sensor/

Just wanna make sure I'm heading the right way, please advice. Thank you.

sauloco, 2017-06-14:
I need to know more about this too! So I comment to get updates... soon, I hope.
Thanks for the hard work!

bneiluj, 2017-06-16:
I'm working on updating the lib to work on `Fingerprint Scanner - TTL (GT-511C3)`

dtex, 2018-01-08:
@rwaldron Do you think this should be the start of a new class (Biometric) should it be a plug-in?

dtex, 2018-01-11:
Hi @icaroscherma , @alphashuro , @gpincheiraa , @xilver , @awk34 , @bneiluj , @sauloco and @bneiluj ,

Wow, there sure are a lot of people interested in fingerprint readers for Johnny-Five! I hope we can find a contributor to work on this soon... One of y'all perhaps?

In the meantime, we are moving this request.

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for fingerprint scanners.

closed 2018-01-11 by dtex

jasonjurotich, 2018-09-21:
Any news on this? Thanks.

dtex, 2018-09-22:
Hi @jasonjurotich, Nobody has stepped up on this feature yet but there sure are a lot of people who want it. When someone does knock this out they are going to make a lot of people very happy. 

I notice your Github profile says that you are "just starting out with Javascript". Contributing to Johnny-Five is a great way to build your skills. You've got a supportive group of contributors and a repo-owner who just happens to be on TC-39 and maintains idiomatic.js. I know working on J5 has helped me grow my JS skills quite a bit.

If you're interested in taking this on I am happy to discuss that the best pattern for this feature might be.

umahatokula, 2018-12-28:
> Hi @jasonjurotich, Nobody has stepped up on this feature yet but there sure are a lot of people who want it. When someone does knock this out they are going to make a lot of people very happy.
>
> I [...]

Hi, i am also very new to JS but i am willing to give this a try. Just let me know what is required of me

dtex, 2018-12-28:
I'm leaning away from suggesting that this should be part of a new class. "Biometric" is too diverse (very little shared code with other biometric devices) and "Fingerprint" is too focused. I would lean toward making this a Johnny-Five plugin.

Here is the pattern: http://johnny-five.io/examples/plugin/

matigarowel, 2022-01-08:
Hi @dtex I would like to work with Fingerprint Module below product:
* `Fingerprint sensor <https://www.adafruit.com/product/751>`_ (Product ID: 751)
* `Panel Mount Fingerprint sensor <https://www.adafruit.com/product/4651>`_ (Product ID: 4651)