rwaldron/johnny-five #851: Problem: One Arduino, 4 motors and 2 servos

tsofist, 2015-07-05:
I'm using Arduino Nano (although the effect is repeated on the Arduino Pro Micro) for the control of four motors and two servo motors.
The motors are controlled by the driver on the basis of L298N.
Without initialization servo drives work fine: you can change speed and direction.
When initializing at least one servo one of the motors starts to behave strangely: does not respond to reverse or change the speed, but the servo work well.
Used pins:
    motors: 2-3, 4-5, 6-7, 8-9
    servos: 10, 11

I use a direct connection with a computer via USB on Windows (and trying on Raspbian)

Initialization:
```javascript
	var flMotor, frMotor, blMotor, brMotor,
		vServo, hServo;
	vServo = new five.Servo({ pin: 11, range: [0, 160], startAt: 70 });
	hServo = new five.Servo({ pin: 10, range: [0, 180], startAt: 60 });
	flMotor = new five.Motor({ pins: [6, 7], invertPWM: true });
	frMotor = new five.Motor({ pins: [3, 2], invertPWM: true });
	blMotor = new five.Motor({ pins: [9, 8], invertPWM: true });
	brMotor = new five.Motor({ pins: [5, 4], invertPWM: true });
```

I would be grateful for any help!

dtex, 2015-07-05:
See this comment:

https://github.com/rwaldron/johnny-five/issues/309#issuecomment-37071805v

I suspect this is the cause of your issue as well. Can you try switching the pins on blMotor so that you are not using pin 10?

tsofist, 2015-07-05:
Thank you much, dtex!
Now it's all working!!
It solved my issue and I think it should be described in the documentation.
Initialization:
```javascript
var flMotor, frMotor, blMotor, brMotor,
    vServo, hServo;
vServo = new five.Servo({ pin: 10, range: [0, 160], startAt: 70 });
hServo = new five.Servo({ pin: 9, range: [0, 180], startAt: 60 });
flMotor = new five.Motor({ pins: [6, 7], invertPWM: true });
frMotor = new five.Motor({ pins: [3, 2], invertPWM: true });
blMotor = new five.Motor({ pins: [11, 12], invertPWM: true });
brMotor = new five.Motor({ pins: [5, 4], invertPWM: true });
```

closed 2015-07-05 by tsofist

dtex, 2015-07-05:
I don't disagree but the rub is not all boards are the same. I wonder if we could build a guard object into board.pins.js that can be used to lookup pins with disabled PWM when servos are in use?

What do you think @rwaldron ?

rwaldron, 2015-07-06:
I like the idea of providing useful "failure" to the user, but I'm not sure how this would work in an IO-Plugin agnostic way

dtex, 2015-07-06:
It seems to come up once a quarter or so. I'll noodle on it for a while and see if I can think of something that's simple and effective.