rwaldron/johnny-five #853: sensor threshold and limit boundary handling differences

mMerlin, 2015-07-07:
Should range checking be made a little more consistent for (at least) sensor devices?  Currently, the range check done for the change (and aliased) events excludes the exact boundary case, so with threshold set (defaulted) to 1, a previous value of 512, and a new median of 513, no change event is emitted.  However, with limit set to [511, 513], limit and limit:upper events ARE emitted.  Exactly hitting the specified threshold value does not trigger a change event, but exactly hitting the specified limit value does trigger a limit event.  I suspect those would be better treating the boundary condition the same way.  Which one is 'right' would need to be determined.

When I was extending the tests for change events, I noticed that the default threshold of "1" would exclude changes of 0.5 and 1.  I do not know if that was deliberate or not.  Was the default threshold intended to act that way?  If the default threshold is supposed to trigger change events for "any" change, the default needs to be less than 0.5, given the automatic (default) filtering provided by the median function.  Changing the threshold logic to include the boundary cases as outside the range would at least include a full unit change in the default change events, which I will guess was the original intent, before median filtering was added.

As a minimum, a little better documentation would be helpful about how those act, possibly with a description of the intent.

I can do any needed changes, if I have a guide to what is correct.

rwaldron, 2015-07-07:
What do you recommend for correcting the cases that you've presented?

mMerlin, 2015-07-07:
My take, but without a lot of background using johnny-five.

Simplest seems to be to change the threshold test to include the boundaries.  IE: in sensor.js, change:
```javascript
if (median < low || median > high) {
```
to:
```javascript
if (median <= low || median >= high) {
```
In addition, if the intent is for the default threshold to emit change events for any change in value, change the default from 1 to 0.5.

mMerlin, 2015-07-07:
With that, code expecting the existing default  would need to explicitly set the freq to something like 1.25 to get back to the same functionality.

I picked 1.25, because that can be represented exactly in binary.  Any value greater than 1.0 and less than 1.5 would work.

rwaldron, 2015-07-07:
> `if (median <= low || median >= high) {`

@BrianGenisio @dtex @Resseguie thoughts?

> With that, code expecting the existing default would need to explicitly set the freq to something like 1.25 to get back to the same functionality.

Did you mean threshold?

dtex, 2015-07-07:
I'm at Sea-Tac getting ready to drive up to Cascadia but will take a look and give it some thought tonight

rwaldron, 2015-07-07:
Cool thanks—and good luck! Have fun!

mMerlin, 2015-07-07:
```
Did you mean threshold?
```
Yep.  Working with both, and mixed them in the message.

dtex, 2015-07-08:
> ```if (median <= low || median >= high) {```

Given the default threshold this makes perfect sense. I also feel the change to existing behavior is so small it will be inconsequential and need not be considered "breaking". It may even improve the experience for some users who were wondering why the event wasn't emitted on first change.

mMerlin, 2015-07-24:
Change included in 0d5887b1c949d8d33d8b6b083361ec4507799d41 and landed with a79f2471c5c775009d1c9752b1542cb0a36e1e72

closed 2015-07-24 by mMerlin