rwaldron/johnny-five #854: changing the frequency of events for an existing device instance

mMerlin, 2015-07-07:
Mentioned in a comment for PR #844, and confirmed by more recent additions to test/sensor.js, changing the freq property does not change the interval between data (or other) events.  A quick check of other component code seems to show that same problem everywhere.  It also showed several different default freq values.
* 20 : Pin
* 25 : Barometer, Motion, Pir, Proximity, Sensor, Temperature
* 50 : IR.Reflect, IR.Proximity (when instance of IR)
* 100 : Magnetometer, Sonar, Temperature (specific driver), Wii.Nunchuck, Wii.Classic
* 500 : IMU, IR, Wii

The main query here is, what is the intended / appropriate functionality when changing the freq property of an existing device instance?

Looking over the code to collect information about the existing functionality, I noticed a few other things:

expander.js uses state.frequency and opts.frequency versus freq property used by most other components.  Should some alignment (or aliasing) be done to get that to match?

compass.js has a comment showing creation with a freq value of 50, and another comment that says the default is 66 ms.  I do not see any actual references to the freq property in the code.  It appears that the comments are cut and pasted from the code for whatever component was used as the base for creating the Compass component.  joystick.js is similar.  A comment, but no apparent property.
Magnetometer is an instance of Compass, and passes a specific value for the freq property, but I do not see where that gets used.  The code that emits (compass) data and change events does not seem to use it.  Those are sent directly by the callback function passed through initialize to io.i2cRead.  No throttling is being done.  Does compass need a structure similar to sensor, to accumulate readings, then send only a filtered value?

rwaldron, 2015-07-07:
> changing the freq property does not change the interval between data (or other) events. A quick check of other component code seems to show that same problem everywhere.

It wasn't meant to—the property should be readonly and value storage moved to internal state. To update the interval times would require messy interaction with the internal intervals. 

> 20 : Pin

This is an attempt to matche the default sampling interval of StandardFirmata, but should just be updated to match the "Barometer, Proximity, Sensor, Temperature"

> Barometer, Motion, Pir, Proximity, Sensor, Temperature

- Barometer, Proximity, Sensor, Temperature... arbitrarily this might as well be the default. 
- Motion, Pir are deprecated and will no longer be updated for any reason.

> 100 : Magnetometer, Sonar, Temperature (specific driver), Wii.Nunchuck, Wii.Classic

-  Magnetometer should be updated to match "Barometer, Proximity, Sensor, Temperature"
- Sonar is deprecated and will no longer be updated for any reason. Magnetometer should  
- Temperature (specific driver) might have constraints? I'll take a look
- Wii.Nunchuck, Wii.Classic... eventually these are going to be removed and put into a separate repo/module. I say just make them match the other sensor components

>  : IR.Reflect, IR.Proximity (when instance of IR)

Both deprecated and will no longer be updated for any reason. 

Deprecated things will be removed for 1.0

> expander.js uses state.frequency and opts.frequency versus freq property used by most other components. Should some alignment (or aliasing) be done to get that to match?

I've mentioned before... somewhere... that "freq" was a bad name for that property, it should be "period". The "frequency" property seen in expander.js is actually for setting the frequency in Hz for that controller. 

> compass.js has a comment showing creation with a freq value of 50, and another comment that says the default is 66 ms. I do not see any actual references to the freq property in the code. It appears [...]

It's just copy/paste junk in joystick, because there is no utility to having such a limitation on joystick, so let's just delete it. In compass, that's sounds like a bug, I will confirm as soon as I have time to take a look. 

> Magnetometer is an instance of Compass, and passes a specific value for the freq property, but I do not see where that gets used. The code that emits (compass) data and change events does not seem [...]

Yeah, I think that's my fault :| It should be like Temperature or Accelerometer

mMerlin, 2015-07-07:
>>  changing the freq property does not change the interval between data (or other) events. A quick check of other component code seems to show that same problem everywhere.

> It wasn't meant to—the property should be readonly and value storage moved to internal state. To update the interval times would require messy interaction with the internal intervals. 

I could do readonly no problem.  Messy?  Maybe.  I think I have the structure needed to implement it.  It did not look that bad to me.  Concept outline would move freq to internal storage as for readonly, plus add a setter function.  The setter would execute a clearInterval(savedId), then savedId=setInterval().  The callback for setInterval could be moved to a named function (outside of the constructor function) instead of anonymous (to avoid creating multiple copies of the function).  Moving the median calculation was already in the plans, to later implement changing the filtering function.  Doing the interval processing function as well is not much more.

mMerlin, 2015-07-09:
> I've mentioned before... somewhere... that "freq" was a bad name for that property, it should be "period".

I've seen that before.  Only got mentioned here because both freq and frequency were being used.  As you say, frequency was appropriate where it was used, being 1/period for that context.

Instead of period, perhaps "interval" would be a good choice to replace the freq property.  That is what the standard javascript uses: setInterval, clearInterval

rwaldron, 2015-07-10:
`interval` works for me. Let's settle on that for eventual 1.0 :)

rwaldron, 2015-11-05:
This is in! https://github.com/rwaldron/johnny-five/commit/0d5887b1c949d8d33d8b6b083361ec4507799d41

closed 2015-11-05 by rwaldron