rwaldron/johnny-five #856: Test issues

dtex, 2015-07-08:
There is more than one so I'm just going to add them to this issue as I find them.

**Issue 1:**
If the readyWithRepl test doesn't run: https://github.com/rwaldron/johnny-five/blob/master/test/board.js#L136-L154 then the repl.js test fails.

rwaldron, 2015-07-08:
repl.js is missing a call to `Board.purge()` and `Repl.isBlocked` should be reset to whatever it was before the test.

dtex, 2015-07-08:
I'll fix that. Also, the "explicit" test: https://github.com/rwaldron/johnny-five/blob/master/test/board.js#L47-L64 times out exactly like ioHasError did https://github.com/rwaldron/johnny-five/blob/master/test/board.js#L93-L114

dtex, 2015-07-08:
Should Board.purge() also be done in the teardown of test/board.js? I know calling it in repl.js wil fix this, but I would think tests should clean up after themselves.

rwaldron, 2015-07-08:
Yes absolutely

dtex, 2015-07-08:
Hmm, Board.purge was already in the teardown. I assumed it cleared Repl.isBlocked when called, but I see now that would not be the case. In fact, I don't see how to unblock the repl with passing ```repl: true``` in a new board opts.

rwaldron, 2015-07-08:
Add `Repl.isBlocked = false` in the setup?

dtex, 2015-07-08:
I'm confused. It doesn't look like Repl.isBlocked actually does anything. 

Now I see that if both ```readyWithNoRepl``` and ```readyWithRepl``` run, the test does in fact clean up after itself because the second one passes repl:true in the opts so I should probably just leave that alone.

I added the missing Board.purge() in my upcoming PR.

Still have that timeout on the explicit test. Should I comment that one out too?

rwaldron, 2015-07-08:
While tracking down the remaining issue with `explicit`, I found a mountain of leaky tests in ESC.

dtex, 2015-07-12:
Closed with #862

closed 2015-07-12 by dtex

rwaldron, 2015-07-13:
Looks like I eliminated the need for it and then forgot that I did that 0_o